espressif/esp-idf #700: FreeRTOS: esp_log_early_timestamp() uses wrong divider, or CPU_CLK_FREQ_ROM defined wrongly; and esp_log_timestamp() seems incorrect

HubbyGitter, 2017-06-14:
Neither could I believe, nor could timestamping the log messages confirm, that my firmware needs 1300 ms to start up. Quite to the contrary, even @ 80 MHz, it clearly starts in < 1000 ms. Also, the startup time is shown as identical for 80 MHz and 240 MHz, which is suspicious.

The `ESP_LOG*()` macros are based on a call to `esp_log_timestamp()` to add a timestamp, which is supposed to return a value representing milliseconds as per `esp_log.h`.

While the task scheduler is not yet started, this call results in a call to `esp_log_early_timestamp()`, which divides `xthal_get_ccount()` by `(CPU_CLK_FREQ_ROM / 1000)`. `CPU_CLK_FREQ_ROM` is defined as `APB_CLK_FREQ_ROM`, which is `( 26*1000000 )`.

It seems to me that before the scheduler  startup, the timestamps are relying on the assumption that the system clock is 26 MHz?
<hr>

Once the scheduler is started, `esp_log_timestamp()` stores the result of `esp_log_early_timestamp()` as a reference upon the first call, and uses `xTaskGetTickCount()` as an offset from there. If the first call to `esp_log_timestamp()` happens N milliseconds after the scheduler is started, all timestamps will show an additional N milliseconds. The solution to this while keeping the simplistic approach would be to subtract `xTaskGetTickCount()` from `esp_log_early_timestamp()` to compute the reference. In order for this to be successful, the value returned by `esp_log_early_timestamp()` must not be too small.

kentaylor, 2017-07-19:
As reported by @HubbyGitter the log message time stamps are wrong. I can add some detail. 

As stated by @HubbyGitter the error is because the [esp_log_early_timestamp()](https://github.com/espressif/esp-idf/blob/master/components/log/log.c#L297) function uses a value for [CPU_CLK_FREQ_ROM](https://github.com/espressif/esp-idf/blob/d998bfc7d62c9ef17a25ea2375e699152f36cb80/components/soc/esp32/include/soc/soc.h#L262) of 26MHz incorrectly. If I turn off logging during boot up then report time just before deep sleeping and subtract this from the time reported immediately after waking I get a difference in boot time from the boot time as reported by the ESP_LOG time stamps of 567 milliseconds. If I have logging on during boot up then both the boot up time and ESP_LOG time stamp error gets larger. This shows that  ESP_LOG time stamps become more erroneous as boot time increases.

If I change CPU_CLK_FREQ_ROM from 26 Mhz to 80 Mhz then the discrepancy between ESP_LOG time stamps and cycle time reduces from 567 milliseconds to 18 millseconds, a vast improvement. 80 MHz is the clock value used after the scheduler [is started](https://github.com/espressif/esp-idf/blob/master/components/log/log.c#L306). This indicates that the 80 MHz clock is started before the scheduler, probably early in the second stage boot. I would guess the 26MHz is correct in the first stage boot when the default internal PLL clock is used (see 3.2.2 Clock Source in [Technical Reference Manual](https://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf)). 

Currently, logic is used to adjust ESP_LOG time for varying CPU frequencies but there is a REF_TICK clock that maintains a constant frequency regardless of CPU frequency (see [3.2.4.2 REF_TICK Source](https://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf)). If this clock is used for the ESP_LOG time stamps the the error should be minimized without requiring logic in the boot code.

kentaylor, 2017-07-20:
@igrr [has explained](https://esp32.com/viewtopic.php?f=13&t=2475#p11657)  why REF_TICK can not be used for ESP_LOG time stamps.

commit 9d9e7719 (2017-09-26): fix(global, log): fix `esp_log(_early)_timestamp` readings after startup by correct the CCOUNT register when switching CPU clock.

TW#13332, Closes #700
closed 2017-09-26 by igrr