espressif/esp-idf #702: esp_wifi_set_protocol throws ESP_ERR_INVALID_ARG

al3x-huang, 2017-06-15:
### Issue
```
void app_main() {
    ...
    wifi_interface_t ifx = ESP_WIFI_IF_STA;
    ESP_ERROR_CHECK(esp_wifi_set_config(ifx, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_set_protocol(ifx, WIFI_PROTOCOL_11G));
    ESP_ERROR_CHECK(esp_wifi_start());
    ...
}
```

### Expected 
Expected no errors, but getting:
```
ESP_ERROR_CHECK failed: esp_err_t 0x102 at 0x40110309
```
referencing that with `esp_err_t`, it is `ESP_ERR_INVALID_ARG`. However, I believe I am using the API call correctly, as using another bitmap works. 

### Notes
* Same error with `WIFI_PROTOCOL_11N` but `WIFI_PROTOCOL_11B` and `WIFI_PROTOCOL_LR` do not throw any errors with the same code above. 

* Getting the protocol bitmap is also a little strange:
    ```    
    uint8_t protocol_bitmap;
    esp_wifi_get_protocol(ESP_IF_WIFI_STA, &protocol_bitmap);
    ```
    produces `0b1111` for `protocol_bitmap`, which I'm not sure how to interpret. I thought it would be `0b0001` if `WIFI_PROTOCOL_11B` is set.

negativekelvin, 2017-06-16:
https://github.com/espressif/esp-idf/blob/99d5959d8a319c26b5b907d26a4a5993a2775594/components/esp32/include/esp_wifi.h#L411

al3x-huang, 2017-06-16:
What is 802.11bg and 802.11bgn? Those aren't even standards as far as I know (and yes, I have Googled them). 

I read that comment and thought they just made a typo as there are exactly 3 modes they support and 3 protocols that they listed in the spec.

igrr, 2017-06-16:
The comment means that only the following options are valid:
- WIFI_PROTOCOL_11B
- WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G
- WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N
(i.e. currently you can't request N only or G only)

And also WIFI_PROTOCOL_LR which isn't mentioned in the comment. Header documentation needs to be updated to reflect that WIFI_PROTOCOL_LR is supported as well.

al3x-huang, 2017-06-16:
Thank you. Two more questions:

1. What would something like WIFI_PROTOCOL11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N do? Would the ESP32 use one of these protocols or just 802.11B?

2. Where can I find the definition for something like `esp_wifi_set_protocol`? I've been combing the entire repo and haven't been able to find the source.  

Regarding the original problem: I don't think the documentation is clear enough. You should explicitly state the effect of using a parameter like WIFI_PROTOCOL11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N rather than saying 802.11bgn is allowed. Unless we get to see the source, we don't really know what the underlying library is doing.

igrr, 2017-06-16:
Ok, fair enough, we'll update the function documentation to make it more clear.

If you set protocol mask to `WIFI_PROTOCOL11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N` (that's what 802.11bgn was supposed to mean — i.e. b mode, g mode, and n mode) then ESP will use one of these modes, depending on AP capabilities and other factors.

If you set protocol mask to `WIFI_PROTOCOL11B`, then the ESP will only use b mode, and not any other mode.

You don't get to see the source of these functions because they are in the WiFi stack, which isn't open source.

al3x-huang, 2017-06-16:
Okay, good to hear. I guess it's just hard to understand why your API doesn't support selecting a specific protocol when it can already use one of any of those modes without seeing how it's implemented. Unless at this stage of ESP-IDF, it only defaults to B mode all the time; then, I can understand better why we can't specifically request G or N more.

SoftwareSamurai, 2017-10-09:
Apologies for replying to such an old thread, but I too would like to be able to set/fix the phy protocol to, say 802.11g OFDM 54 Mbps. (I'm conducting R&D regarding a custom wifi data packet protocol, and having this level of control in promiscuous mode would be great.) Is there a reason this can't be done?

jack0c, 2018-01-03:
Not sure what you want. 
11b mode means, ESP32 can only send 1M,2M,5.5M,11M packets, the peer device can only send 1M,2M,5.5M,11M packets to ESP32.
11bg mode means, ESP32 can only send 1M~11M,6M~54M packets, the peer device can only send  1M~11M,6M~54M packets to ESP32.
Your 11g mode means,  ESP32 can only send 6M~54M packets, the peer device can only send  6M~54M packets to ESP32. In this case, ESP32 can not connect to many APs.

negativekelvin, 2018-01-03:
@jack0c most AP can turn off b rates so esp32 as AP should be able to exclude b rates/clients.  Some people want to choose specific rates for research.

Also have you seen this https://esp32.com/viewtopic.php?f=14&t=4080

jack0c, 2018-01-14:
Plan to complete this in idf4.0, now we are developing idf3.1.

chegewara, 2018-01-14:
Maybe you can first release idf 3.0 before you start developing next versions? We stuck with old arduino library because idf3.0 is pre-release for weeks.

jack0c, 2018-01-14:
@chegewara In ideal case, you are right. But release will sometimes blocked by some issues.

negativekelvin, 2018-01-14:
@jack0c wifi features and optimizations keep getting kicked down the road is a big problem. battery powered products and other innovations are stopped dead by these issues.  Either need to put more developers on wifi team or open source more of the stack.

jack0c, 2018-01-15:
@negativekelvin Sorry this is a big feature, need long time to complete. If  you are blocked on this, we can give a temporary version to support this. @liuzfesp , Please help.

liuzfesp, 2018-04-03:
@SoftwareSamurai in order to set fix rate, you can call the WiFi hidden API:
/**
 * @brief     Set fixed rate
 *
 * @attention This API can be used only when AMPDU is disabled, otherwise the behavior is undefined.
 * @attention Normally this API is used in some special cases, such as debug, certification etc
 * @attention If the fixed rate is fixed LR rate, make sure the LR mode is enabled
 *
 * @param  wifi_if : wifi interface id
 * @param  en_fix_rate : fixed rate is enabled or not
 * @param  rate : rate to be configured, valid only when en_fix_rate is true
 *
 * @return
 *    - ESP_OK: succeed
 *    - ESP_ERR_WIFI_NOT_INIT: WiFi is not initialized by esp_wifi_init
 *    - ESP_ERR_WIFI_NOT_START: WiFi is not started by esp_wifi_start
 *    - ESP_ERR_WIFI_ARG: invalid argument
 */
esp_err_t esp_wifi_internal_set_fix_rate(wifi_interface_t ifx, bool en_fix_rate, wifi_rate_t rate);

For fix rate value, please refer to,  "0=1L" means the value 0 indicates 1Mbits long preamble, "7=11S" means 7 indicates 11Mbits short preamble, "26=2S21.6" means 26 indicates MCS2 Short Preamble 21.6Mbits etc:

0 =1L      1 =2L      2 =5L        3 =11L     4 =NA      5 =2S      6 =5S      7 =11S
8 =48L    9 =24L      10=12L  11=6L 12=54      13=36L      14=18L      15=9L
16=0L6.5   17=1L13    18=2L19.5  19=3L26 20=4L39    21=5L52    22=6L58.5  23=7L65
24=0S7.2   25=1S14.4  26=2S21.6  27=3S28.9 28=4S43.3  29=5S57.8  30=6S65    31=7S72.2

liuzfesp, 2020-07-13:
Same as https://github.com/espressif/esp-idf/issues/5061, close this old one and keep the latest one opened.

closed 2020-07-14 by Alvin1Zhang