espressif/esp-idf #703: uart_write_byte_with_break() throws error on 0-size data

benpeoples, 2017-06-16:
The theory (per [this forum post](https://www.esp32.com/viewtopic.php?f=13&t=1476&p=7087&hilit=uart+break#p7087) ) is that you can use `uart_write_bytes_with_break()` and no data to trigger a fixed-size break.  

However, doing this triggers this line:

`UART_CHECK((size > 0), "uart size error", (-1));`

and `uart_tx_all()` also has a `size == 0` check:

` if(size == 0) {
        return 0;
    }
`

So it is unclear if this is intended behavior.   Regardless, it would be nice to be able to trigger a break directly (or prefix data with BREAK instead of suffix).   I'm currently kludging it with `uart_set_line_inverse(uart_num, UART_INVERSE_TXD);` but this is less than ideal.

costaud, 2017-06-28:
The hardware do not support send tx break directly, it can only send break signal after sending some bytes.

benpeoples, 2017-06-30:
We're doing an implementation of ANSI E1.11 DMX-512, which uses a BREAK to signal the start of a packet.   Signal is 250kbit/sec (or 4 us per bit). 

I need a ~100us (or 25 bit) BREAK, followed by at least 12 us of MARK, and then 513 bytes of data.

I see a few methods to achieve this:

1.   `uart_set_line_inverse(uart_num, UART_INVERSE_TXD);` -- Which works along with some hard delays, but keeps the task running instead of idling (although, only for 100us).   We do immediately go into a forced yield right now when we mark, so our 12us MARK is a little long.
2.  Switch the port to a GPIO that's output low.    This works fairly well on an Atmel AVR, but I don't know about switching back and forth between peripherals here.   And again we're busy-waiting.
3. Switch the baud rate around.   Setting the baud rate down to 80kbps and pushing a 0x00 out the port would generate a 100us BREAK and a 25us MARK (2 stop bits... a single stop bit would be 12.5).   

Because of the `uart_param_config()` function, it seems like 3 might be the best bet?   My hope is to just not busy wait through the BREAK.

Does the UART spew random signals while clock switching?

dpgeorge, 2017-06-30:
> Does the UART spew random signals while clock switching?

In some of my tests trying to switch TX/RX GPIO and/or UART configuration parameters I saw spurious outputs on the pins and sometimes the UART RX would receive spurious characters.  But I never got to the bottom of it (it could have been just my misuse of the IDF API) so would be interested to hear other people's experience regarding this.

benpeoples, 2017-06-30:
I'll dig into it when I'm back in the office next week...

benpeoples, 2017-07-10:
OK, dug into this:  ESP32 cleanly supports changing the baud rate.  Not only that, it'll happily change the baud rate in the middle of sending a bit: [image](http://imgur.com/a/ZHJDy)

The technique does not seem to be any better than just flipping the line:

      for(;;)
      {
        uart_wait_tx_done(uart_num,1000);
        uart_set_line_inverse(uart_num, UART_INVERSE_TXD);
        ets_delay_us(92);
        uart_set_line_inverse(uart_num, UART_INVERSE_DISABLE);
        ets_delay_us(12);
        uart_write_bytes(uart_num, (const char*) data, 10);
      }

vs

      for(;;)
      {
        uart_wait_tx_done(uart_num,1000);
        uart_param_config(uart_num, &uart_config_break);
        uart_tx_chars(uart_num, (const char*) data, 1);
        uart_wait_tx_done(uart_num,1);
        uart_param_config(uart_num, &uart_config);
        uart_write_bytes(uart_num, (const char*) data, 10);
      }

Either way, I end up losing task focus, and incurring at least a millisecond instead of a ~12us MAB.  

This does not appear to work (UART stays in baud = 250000 mode), I think because it's interrupt driven?

      portMUX_TYPE myMutex = portMUX_INITIALIZER_UNLOCKED;
      for(;;)
      {
        uart_wait_tx_done(uart_num,1000);
        taskENTER_CRITICAL(&myMutex);
        uart_param_config(uart_num, &uart_config_break);
        uart_tx_chars(uart_num, (const char*) data, 1);
        ets_delay_us(200);
        uart_param_config(uart_num, &uart_config);
        uart_write_bytes(uart_num, (const char*) data, 10);
        taskEXIT_CRITICAL(&myMutex);
      }

This works, but there's still an excessively large (~500-900us) MAB:

      portMUX_TYPE myMutex = portMUX_INITIALIZER_UNLOCKED;
      for(;;)
      {
        uart_wait_tx_done(uart_num,1000);
        taskENTER_CRITICAL(&myMutex);
        uart_set_line_inverse(uart_num, UART_INVERSE_TXD);
        ets_delay_us(92);
        uart_set_line_inverse(uart_num, UART_INVERSE_DISABLE);
        ets_delay_us(12);
        uart_write_bytes(uart_num, (const char*) data, 10);
        taskEXIT_CRITICAL(&myMutex);
      }

moving the `taskEXIT_CRITICAL` above the uart_write_bytes has similar effect.

The problem I basically have now is that I don't seem to be able to get no task switch between the end of the BREAK and the start of the new data writing.    Is there some way to "pause" the UART FIFO?  E.g., I could change the baud rate between the first and second byte?  

It may end up being OK, worst case we lose about a millisecond (going from about 44Hz to 43Hz), but the inefficiency is bugging me.

alien-explosive, 2017-08-18:
We have struggled with this issue as well. Please see [UART 1 ms timeout following empty TX FIFO #917](https://github.com/espressif/esp-idf/issues/917). I suspect it may be a hardware issue.

A workaround is to use the baud rate change method to generate the break, load the FIFO with subsequent data during the break transmission, and change the baud rate back during the MAB. We have implemented this and can successfully send break followed by data, without the 1 millisecond timeout.

We have not yet gone any further, however it appears that the unwanted timeouts can be avoided by never allowing the UART Tx to enter an idle state. It might be that the problem can better be described as a 1 millisecond minimum idle time.

alien-explosive, 2017-08-18:
Note that according to the ANSI_E1-20 inter-slot time up to 2.0 ms is acceptable, but maximum MAB time for transmitters is 88 uS.

benpeoples, 2017-08-18:
That's a good point.   E1.11 puts the maximum MAB at "< 1 second", so for "regular" NSC data, receivers should handle it without a problem.  

Getting the timing right for E1.20 is mainly about not being disruptive so half-duplex actually works reasonably.    Hopefully we can get a solution to the problem.  Otherwise I'm back to an external SPI UART...

alien-explosive, 2017-08-18:
After some more checking I can say that my workaround as proposed in the earlier message solves the problem. We have implemented a standards-compliant DMX transmitter. I didn't try RDM yet, but I don't see any reason why it should not be possible either using the same method. 

Note that we are not using any of the API functions, just directly writing the registers. It would be even better if we could use the DMA, but cannot find any documentation about that at all, just a cryptic reference to a non-existent chapter in the technical manual.

benpeoples, 2017-08-18:
Excellent.  

Considering that the esp-idf doesn't use DMA for `uart_fill_fifo`, I'm not sure DMA is implemented for the UART... =)

How are you detecting that the UART has finished transmitting the BREAK byte?  Just counting cycles?

alien-explosive, 2017-08-18:
Yes, we just count cycles. Theoretically it would also be possible to read the UART state register, but it is more efficient just to set a timer interrupt.

According to page 233 of the technical reference manual:

> 10.3.7 UART DMA
> For information on the UART DMA, please refer to Chapter BUS DMA.

There is no such chapter.

Also at the end of the UART controller chapter the registers for the USB controller are described. But again there is no documentation of an onboard USB controller to be found anywhere. Documentation definitely appears to be work in progress.

igrr, 2017-08-19:
There's definitely no USB controller in there. The thing called UHCI is, in fact, the UART DMA block. We are working on a driver for UART DMA.

alien-explosive, 2017-08-23:
Issue https://github.com/espressif/esp-idf/issues/917 should probably be merged with this issue.

It turns out that the idle time between frames is configurable, and set by default to the maximum. (Why??) It can be set to 0 by means of the following code, thus removing the unwanted 1 ms timeouts.

`UART[uart_num]->idle_conf.tx_idle_num = 0;`

costaud, 2017-10-18:
We will set default tx_idle_num to 0 in driver and provide API to set this value. Will be merged to master branch soon.

closed 2017-10-31 by FayeY