espressif/esp-idf #704: gpio_dev_t

chegewara, 2017-06-16:
Hi.
Im totally newbie and im trying to adapt this library https://github.com/espressif/esp32-nesemu/blob/master/components/nofrendo-esp32/spi_lcd.c , but im getting error from gpio_struct.h.

        error: 'gpio_dev_t GPIO', declared using anonymous type, is used but never defined [-fpermissive] extern gpio_dev_t GPIO;

        F:/esp-idf/components/soc/esp32/include/soc/gpio_struct.h:202:3: note: 'typedef volatile struct<anonymous> gpio_dev_t' does not refer to the unqualified type, so it is not used for linkage } gpio_dev_t;

Any advice or help if its not a bug. Thanks

FHFS, 2017-06-16:
If you include that header you get that error?

try adding that flag [-fpermissive](https://stackoverflow.com/questions/8843818/what-does-the-fpermissive-flag-do). Also read [this](https://stackoverflow.com/questions/16383099/non-local-variable-uses-anonymous-type-warning-in-c)

chegewara, 2017-06-16:
Yes, im getting this error when i include this header in c++ .cpp file. Temporary solution ive found was to change this header from:
`   typedef volatile struct {`
to
`typedef volatile struct GPIO {`

FHFS, 2017-06-16:
try
```
extern "C" {
    #include "soc/gpio_struct.h"
}
```

chegewara, 2017-06-16:
orks. Thanks.

closed 2017-06-16 by chegewara