espressif/esp-idf #705: How to set static DNS servers?

pjjasinski, 2017-06-16:
Hello
I've found that:
tcpip_adapter_set_ip_info(tcpip_adapter_if_t tcpip_if, tcpip_adapter_ip_info_t *ip_info)
do this
...
#if LWIP_DNS /* don't build if not configured for use in lwipopts.h */
        u8_t numdns = 0;
        for (numdns = 0; numdns < DNS_MAX_SERVERS; numdns ++) {
            dns_setserver(numdns, NULL);
        }
#endif

So i'am doing this after set static ip,gw:
		ip_addr_t dnsserver;
		inet_pton(AF_INET, "8.8.8.8", &dnsserver);
		dns_setserver(0, &dnsserver);
		inet_pton(AF_INET, "8.8.4.4", &dnsserver);
		dns_setserver(1, &dnsserver);
but DNS is not working

When use DHCP DNS working perfect.

FHFS, 2017-06-16:
try
```
#include "lwip/dns.h"

ip_addr_t dnsserver;
IP_ADDR4( &dnsserver, "8.8.8.8");
dns_setserver(0, &dnsserver);
```

FayeY, 2017-06-19:
Hi @pjjasinski , is this problem solved now?

pjjasinski, 2017-06-19:
Ok
When i was setting dns right after tcpip_adapter_set_ip_info()  they not working but i did:

esp_err_t event_handler(void *ctx, system_event_t *event)
{
	ip_addr_t dnsserver;
	switch(event->event_id)
	{
	case SYSTEM_EVENT_STA_START:
		break;
	case SYSTEM_EVENT_STA_GOT_IP:
		ESP_LOGI(TAG, "SYSTEM_EVENT_STA_GOT_IP");
		inet_pton(AF_INET, "8.8.8.8", &dnsserver);
		dns_setserver(0, &dnsserver);
		inet_pton(AF_INET, "8.8.4.4", &dnsserver);
		dns_setserver(1, &dnsserver);
		break;
	case SYSTEM_EVENT_STA_DISCONNECTED:
		break;
	case SYSTEM_EVENT_ETH_GOT_IP:
		ESP_LOGI(TAG, "SYSTEM_EVENT_ETH_GOT_IP");
		inet_pton(AF_INET, "8.8.8.8", &dnsserver);
		dns_setserver(0, &dnsserver);
		inet_pton(AF_INET, "8.8.4.4", &dnsserver);
		dns_setserver(1, &dnsserver);
		break;
	default:
		break;
	}
	return ESP_OK;
}
This solution is working fine.

There is no good example how to set static IP,DNS with WIFI and ETHERNET.

closed 2017-10-23 by igrr

sosaucily, 2020-01-28:
For me @FHFS 's code needed to be tweaked just a bit. Thanks for the help!

```
#include "lwip/dns.h"

ip_addr_t dnsserver;
IP_ADDR4( &dnsserver, 8, 8, 8, 8);
dns_setserver(0, &dnsserver);
```