espressif/esp-idf #707: WiFi scan interferes with station connect

rojer, 2017-06-17:
It seems that if scan is initiated when station is connecting, finishing scan gets station connection process stuck.
I have a simple test, which gets station connecting, then, a secon later, initiates a non-blocking scan with the following config:
```c
    wifi_scan_config_t scan_cfg = {
      .ssid = NULL,
      .bssid = NULL,
      .channel = 0,
      .show_hidden = false,
      .scan_type = WIFI_SCAN_TYPE_ACTIVE,
      .scan_time = {
        .active = {
          .min = 10,
          .max = 50,
        },
      }
    };
```
If the scan finishes at just the wrong moment (and it happens more often than not), station fails to connect - it just gets stuck.
Here's log from successful run - station connects and scan completes:
```
mgos_wifi_setup_sta  WiFi STA: Connecting to Over_9000_Internets
...
mgos_init            Init done, RAM: 257968 total, 184036 free, 181744 min free
I (3946) wifi: n:8 0, o:1 0, ap:255 255, sta:8 0, prof:1
I (4796) wifi: state: init -> auth (b0)
I (4806) wifi: state: auth -> assoc (0)
I (4826) wifi: state: assoc -> run (10)
esp32_wifi_ev        WiFi scan done: 0 2
event_handler        event: 1
wifi_scan_cb         123 ssids 0x3ffb3a50
wifi_scan_cb         123 Over_9000_Internets
wifi_scan_cb         123 Mongoose_DA803A
I (5846) wifi: connected with Over_9000_Internets, channel 8
mgos_wifi_on_change_ Wifi: connected
I (5936) event: ip: 192.168.11.33, mask: 255.255.255.0, gw: 192.168.11.1
mgos_wifi_on_change_ WiFi: ready, IP 192.168.11.33, DNS 192.168.11.1
```

however, if scan finishes earlier - at the `auth (b0)` stage, the station never connects:

```
mgos_wifi_setup_sta  WiFi STA: Connecting to Over_9000_Internets
...
mgos_init            Init done, RAM: 257968 total, 184036 free, 181744 min free
I (3946) wifi: n:8 0, o:1 0, ap:255 255, sta:8 0, prof:1
I (4786) wifi: state: init -> auth (b0)
esp32_wifi_ev        WiFi scan done: 0 2
event_handler        event: 1
wifi_scan_cb         123 ssids 0x3ffb3a50
wifi_scan_cb         123 Over_9000_Internets
wifi_scan_cb         123 Mongoose_DA803A
```
there are no further messages from wifi, STA connection never completes.

rojer, 2017-06-17:
btw, i see similar behavior on ESP8266

FayeY, 2017-07-05:
Thanks for your reporting. This issue will be solved in the ESP-IDF V3.0.

rojer, 2017-10-30:
i no longer see this

closed 2017-10-30 by rojer