espressif/esp-idf #709: `btm_api.h` linkage issue (Bluedroid BTM)

JLospinoso, 2017-06-18:
Having a linkage issue with Bluedroid BTM:

```c
#include <btm_api.h>
// ...
  if (BTM_IsDeviceUp()) {
    ESP_LOGI("BT", "BTM says device is up!");
  } else {
    ESP_LOGE("BT", "BTM says device is down :-(");
  }
// ...
```

Error message:
```
libmain.a(bluetooth_scanner.o):(.literal._ZNK16BluetoothScanner4scanEv+0xc): undefined reference to `BTM_IsDeviceUp()'
```

Here's my `sdkconfig`:

```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"

#
# Bootloader config
#
# CONFIG_LOG_BOOTLOADER_LEVEL_NONE is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_ERROR is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_WARN is not set
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
# CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE is not set
CONFIG_LOG_BOOTLOADER_LEVEL=3

#
# Security features
#
# CONFIG_SECURE_BOOT_ENABLED is not set
# CONFIG_FLASH_ENCRYPTION_ENABLED is not set

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="COM4"
CONFIG_ESPTOOLPY_BAUD_115200B=y
# CONFIG_ESPTOOLPY_BAUD_230400B is not set
# CONFIG_ESPTOOLPY_BAUD_921600B is not set
# CONFIG_ESPTOOLPY_BAUD_2MB is not set
# CONFIG_ESPTOOLPY_BAUD_OTHER is not set
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
# CONFIG_FLASHMODE_QIO is not set
# CONFIG_FLASHMODE_QOUT is not set
CONFIG_FLASHMODE_DIO=y
# CONFIG_FLASHMODE_DOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE="dio"
# CONFIG_ESPTOOLPY_FLASHFREQ_80M is not set
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
# CONFIG_ESPTOOLPY_FLASHFREQ_26M is not set
# CONFIG_ESPTOOLPY_FLASHFREQ_20M is not set
[... 223 lines cut ...]
# CONFIG_MBEDTLS_DEBUG is not set
CONFIG_MBEDTLS_HARDWARE_AES=y
CONFIG_MBEDTLS_HARDWARE_MPI=y
CONFIG_MBEDTLS_MPI_USE_INTERRUPT=y
CONFIG_MBEDTLS_HARDWARE_SHA=y
CONFIG_MBEDTLS_HAVE_TIME=y
# CONFIG_MBEDTLS_HAVE_TIME_DATE is not set

#
# OpenSSL
#
# CONFIG_OPENSSL_DEBUG is not set
CONFIG_OPENSSL_ASSERT_DO_NOTHING=y
# CONFIG_OPENSSL_ASSERT_EXIT is not set

#
# SPI Flash driver
#
# CONFIG_SPI_FLASH_ENABLE_COUNTERS is not set
CONFIG_SPI_FLASH_ROM_DRIVER_PATCH=y
```

I'm interested in performing a Bluetooth Classic scan for Bluetooth Device Addresses, and it looks like I may find my way stitching together some functions in [the `btm_api.h` header](https://github.com/espressif/esp-idf/blob/master/components/bt/bluedroid/stack/include/btm_api.h).

Is there an easier way? It looks fairly easy to subscribe to BLE advertisements, but I cannot find working examples of scanning for Bluetooth Classic devices. Alternately, why can I not link in the Bluedroid components?

JLospinoso, 2017-06-18:
This function is of particular interest (from `btm_api.h`):

```c
/*******************************************************************************
**
** Function         BTM_StartInquiry
**
** Description      This function is called to start an inquiry.
**
** Parameters:      p_inqparms - pointer to the inquiry information
**                      mode - GENERAL or LIMITED inquiry
**                      duration - length in 1.28 sec intervals (If '0', the inquiry is CANCELLED)
**                      max_resps - maximum amount of devices to search for before ending the inquiry
**                      filter_cond_type - BTM_CLR_INQUIRY_FILTER, BTM_FILTER_COND_DEVICE_CLASS, or
**                                         BTM_FILTER_COND_BD_ADDR
**                      filter_cond - value for the filter (based on filter_cond_type)
**
**                  p_results_cb   - Pointer to the callback routine which gets called
**                                upon receipt of an inquiry result. If this field is
**                                NULL, the application is not notified.
**
**                  p_cmpl_cb   - Pointer to the callback routine which gets called
**                                upon completion.  If this field is NULL, the
**                                application is not notified when completed.
** Returns          tBTM_STATUS
**                  BTM_CMD_STARTED if successfully initiated
**                  BTM_BUSY if already in progress
**                  BTM_ILLEGAL_VALUE if parameter(s) are out of range
**                  BTM_NO_RESOURCES if could not allocate resources to start the command
**                  BTM_WRONG_MODE if the device is not up.
**
*******************************************************************************/
//extern
tBTM_STATUS  BTM_StartInquiry (tBTM_INQ_PARMS *p_inqparms,
                               tBTM_INQ_RESULTS_CB *p_results_cb,
                               tBTM_CMPL_CB *p_cmpl_cb);

```

nkolban, 2017-06-23:
By any chance, are you trying to call these functions from a C++ application?

JLospinoso, 2017-06-23:
@nkolban why, yes! I anticipate that name mangling is the source of my problem here since you asked the question?

Thanks for the book, by the way :)

TianHao-Espressif, 2017-07-27:
@JLospinoso , sorry for delay. In design, application shouldn't include BTM_xxx.h or BTA_xxx.h. Application should only include esp_xxx.h of bluedroid.
The missing thing is that some of the demo keep the '#include "BTA_xxx.h"', it is incorrect. In the newest master, we will remove these "#include" line.

closed 2017-11-21 by FayeY