espressif/esp-idf #710: UART with TX buffer loses first 2 transmitted character

dpgeorge, 2017-06-19:
It seems that when the UART is configured to use a TX buffer, the first call to uart_write_bytes will drop the first 2 characters.

I managed to reproduce this issue using the provided example "uart_echo".  Against commit 81f98c0a772864cbbaa2f270f7a7f6039c16f4c8 (which was latest master when I was debugging) if you apply the following patch then it should show the problem:
```diff
--- a/examples/peripherals/uart_echo/main/uart_echo_example_main.c
+++ b/examples/peripherals/uart_echo/main/uart_echo_example_main.c
@@ -46,16 +46,17 @@ static void echo_task()
         .data_bits = UART_DATA_8_BITS,
         .parity = UART_PARITY_DISABLE,
         .stop_bits = UART_STOP_BITS_1,
-        .flow_ctrl = UART_HW_FLOWCTRL_CTS_RTS,
-        .rx_flow_ctrl_thresh = 122,
+        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
+        .rx_flow_ctrl_thresh = 0,
     };
     //Configure UART1 parameters
     uart_param_config(uart_num, &uart_config);
     //Set UART1 pins(TX: IO4, RX: I05, RTS: IO18, CTS: IO19)
-    uart_set_pin(uart_num, ECHO_TEST_TXD, ECHO_TEST_RXD, ECHO_TEST_RTS, ECHO_TEST_CTS);
+    uart_set_pin(uart_num, ECHO_TEST_TXD, ECHO_TEST_RXD, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
     //Install UART driver (we don't need an event queue here)
     //In this example we don't even use a buffer for sending data.
-    uart_driver_install(uart_num, BUF_SIZE * 2, 0, 0, NULL, 0);
+    uart_driver_install(uart_num, BUF_SIZE * 2, 32, 0, NULL, 0);
+    uart_write_bytes(uart_num, "START", 5);
 
     uint8_t* data = (uint8_t*) malloc(BUF_SIZE);
     while(1) {
```
The output from the UART is "ART", and is missing the "ST" part.

costaud, 2017-06-27:
I think it is a tx-buffer-length issue, the tx buffer length should be greater than 128, maybe at least 64. We will add length check in code.

dpgeorge, 2017-06-28:
Thanks for the response.  I checked with a buffer of 64 and it now works (the first 2 chars are not lost).

It seems strange though that such a large buffer is needed for it to work.  Every byte of RAM counts...

igrr, 2017-07-23:
Error check added in 40fd9db.

closed 2017-07-23 by igrr