espressif/esp-idf #712: After updating ESP32-IDF through github make clean, make all etc... has stopped to work

Gryffoon, 2017-06-19:
After updating ESP32-IDF through github make clean, make all etc... has stopped to work !!

when I do make clean 
make[1]: *** /home/gryffoon/workspace_esp32/ModularSoft/build/libsodium: Aucun fichier ou dossier de ce type. Arrêt. (*)
/opt/esp-idf/make/project.mk:386: recipe for target 'libsodium-clean' failed
make: *** [libsodium-clean] Error 2
(*) means no such file or directory found. 

make all: 
make: *** No rule to make target '/home/gryffoon/workspace_esp32/ModularSoft/build/include/config/auto.conf', needed by '/home/gryffoon/workspace_esp32/ModularSoft/build/bootloader/bootloader.bin'. Stop.

make menuconfig: 
Error opening terminal: unknown.
make: *** No rule to make target 'menuconfig', needed by '/home/gryffoon/workspace_esp32/ModularSoft/build/include/config/auto.conf'. Arrêt.

I've already made this command "git submodule update --init --recursive" in /opt/esp-idf/

rillhu, 2017-07-03:
It look the ESP-IDF update commad is 
`git submodule update --init`

Please see [http://esp-idf.readthedocs.io/en/latest/get-started/index.html#get-esp-idf](url)

And I guess above command should be run after `git pull`.

Navis-Raven, 2017-07-03:
404 page not found, your link does not work

Navis-Raven, 2017-07-03:
I already did this command and still not work. but one or two days after, (after rebooting computer, working on other things), curiously the make command worked again, WITHOUT any intervention it worked again after one day. 

This is so strange.

Navis-Raven, 2017-07-03:
For info I use git trough Eclipse I don't know if this command can be included automatically into Eclipse.

ikalchev, 2017-10-06:
I just got the same problem:

```
IK-Pro:hello_world$ make menuconfig

cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD   -c -o mconf.o mconf.c
flex -L -P zconf -o zconf.lex.c zconf.l
zconf.l:255: warning, -s option given but default rule can be matched
make[1]: *** [zconf.lex.c] Error 1
bison -t -l -p zconf -o zconf.tab.c zconf.y
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD   -c -o zconf.tab.o zconf.tab.c
In file included from zconf.tab.c:2560:
./confdata.c:113:23: error: use of undeclared identifier 'PATH_MAX'
        static char fullname[PATH_MAX+1];
```
Other make commands are failing as well. Then I tried with a clean clone and got the same problem.

This is macOS High Sierra, the config was done following [the esp get started](https://esp-idf.readthedocs.io/en/latest/get-started/index.html#).

This used to work for a long time a month or so ago.

lixianyu, 2017-11-07:
How about this issue now?

n0p, 2018-01-22:
I got this same problem when trying to compile my project under different releases (by doing git checkout release/vx.x )

I downgraded IDF to 2.1 and now I can't compile under v3.0 or master
```
litetatil ➜  PoC git:(master) ✗  LANG=C make 
make[1]: Entering directory '/home/n0p/src/esp32/esp-idf/tools/kconfig'
make[1]: *** No rule to make target '/usr/include/xlocale.h', needed by 'mconf.o'.  Stop.
make[1]: Leaving directory '/home/n0p/src/esp32/esp-idf/tools/kconfig'
make: *** No rule to make target '/home/n0p/src/esp32/esp-idf/tools/kconfig/conf', needed by '/home/n0p/src/esp32/projectfirm/PoC/build/include/config/auto.conf'.  Stop.
```

igrr, 2018-01-22:
@n0p The fix for this issue has been added between 2.1 and 3.0, in commit https://github.com/espressif/esp-idf/commit/a361a00405c954a8f694a25a07634d1eb919aa2f. 

In 2.1 and 2.1.1, please go to $IDF_PATH/tools/kconfig and do `make clean` **and** `find . -name '*.d' | xargs rm` to remove the old dependency files.

ahmedawais, 2018-05-10:
thanks  now my code is working

yuguangyu, 2018-08-15:
macos:  #make
make[1]: *** No rule to make target `/usr/include/stdc-predef.h', needed by `mconf.o'.  Stop.
make: *** No rule to make target `/Users/guangyu/esp/esp-idf/examples/get-started/hello_world/build/include/config/auto.conf', needed by `/Users/guangyu/esp/esp-idf/examples/get-started/hello_world/build/partitions_singleapp.bin'.  Stop.

kluverp, 2018-08-28:
Maybe this is totally unrelated, but I had some problems as well when updating to new IDF.
Cause of error was a file i named 'log' in the root of my project. (I logged some stuff there from the serial output). 

Anyway, maybe there are some files in your root causing issues because of their names? Also I notices that removing the 'build' folder entirely from your project might help in some cases.

Navis-Raven, 2018-08-29:
@igrr it seems that this issue persists anyway since there is still activity.

projectgus, 2018-09-03:
Unfortunately are some circumstances where a system upgrade can cause mconf builds to fail, based on system header files moving around. And sometimes they fail before `make clean` is allowed to run and fix the situation, meaning that the build gets stuck.

If `make clean` does not complete because of an error, run this command:

```
cd $IDF_PATH
find ./tools/kconfig -name '*.d' -delete
```

(Check the current working directory is the IDF_PATH directory and the full command line for `find` is correct as shown here before running `find`, as it can delete the wrong files if there is a mistake in  the working directory or the command's arguments.)
 
The forthcoming CMake-based build system properly resolves this problem.

william-ferguson-au, 2019-01-27:
This still occurs in esp-idf-3.2.1

Replicate by doing clean build from Eclipse.
Once you have cleaned from eclipse you cannot do 'make all' until you have first run 'make menuconfig'.

MightyPork, 2019-02-01:
I just got it too:

```none
Note: You are using Python 3.7.2. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from /home/ondra/esp/esp-idf/requirements.txt are satisfied.
Running make in directory /home/ondra/esp/fb-node/build
Executing "make -j 6 menuconfig"...
[  0%] Performing build step for 'mconf-idf'
make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[4]: *** No rule to make target '/home/ondra/esp/esp-idf/tools/kconfig/zconf.tab.c', needed by 'zconf.tab.o'.  Stop.
make[3]: *** [CMakeFiles/mconf-idf.dir/build.make:125: mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build] Error 2
make[2]: *** [CMakeFiles/Makefile2:202: CMakeFiles/mconf-idf.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:717: esp-idf/CMakeFiles/menuconfig.dir/rule] Error 2
make: *** [Makefile:307: menuconfig] Error 2
make failed with exit code 2
```
tried bisecting without luck but eventually fixed by idf clean, idf fullclean in the project directory and getting a clean idf clone

Navis-Raven, 2019-02-23:
@igrr  @projectgus 
I think you should re open this issue, 

It seems that the bug still occurs for some users instead of the last corrections

projectgus, 2019-02-25:
@william-ferguson-au 
> Replicate by doing clean build from Eclipse.
>Once you have cleaned from eclipse you cannot do 'make all' until you have first run 'make menuconfig'.

Sorry for the slow response. I was able to reproduce the initial problem with IDF v3.2-beta1. A clean build where the sdkconfig file is missing comes up with this error:

```
make: *** No rule to make target 'menuconfig', needed by '/path/to/sdkconfig'.  Stop.
```

If the sdkconfig file exists (even if it's empty), this does not happen. Will look into this again.

@MightyPork I think this issue is different to the issues reported above.

```
make[4]: *** No rule to make target '/home/ondra/esp/esp-idf/tools/kconfig/zconf.tab.c', needed by 'zconf.tab.o'.  Stop.
``
The other issues were either related to `menuconfig` target to dependencies on header files that were generated in `.d` files by the build system, but this target [has a rule in the Makefile](https://github.com/espressif/esp-idf/blob/master/tools/kconfig/Makefile#L334) so something else is going on here.

You can try deleting `zconf.lex.c` if it exists (in either the `build/` directory or the `tools/kconfig` directory, and then run `make` again.

If it still happens, please open a new issue with the details. Feel free to link back to this one.

sravanth005, 2019-03-04:
admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp-who/examples/single_chip/camera_web_server$ make all
Toolchain path: /home/admin-unizen/esp32/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
The following Python requirements are not satisfied:
cryptography>=2.1.4
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/admin-unizen/esp/esp-idf/requirements.txt" for resolving the issue.
/home/admin-unizen/esp/esp-idf/make/project.mk:500: recipe for target 'check_python_dependencies' failed
make: *** [check_python_dependencies] Error 1

please help me out to clear this error as iam a fresher i dont know to correct if anything is to be installed means just send me the link or commands for ubuntu. certainly i work in UBUNTU 16.04 LTS.
so accordingly send me the commands please.

MightyPork, 2019-03-04:
@sravanth005 it sounds like your
> Toolchain version is not supported

and

> Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
> Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/admin-unizen/esp/esp-idf/requirements.txt" for resolving the issue.

it's right there in your error
anyway it's a different issue than what's discussed here

sravanth005, 2019-03-04:
@MightyPork 
thanks for your replay,
I went through ESP-IDF programming guide and tried the steps but its not happening.I have pasted the commands below can any one help me out to get out from this error.

~/esp$ sudo apt-get install gcc git wget make libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-cryptography python-future python-pyparsing
[sudo] password for admin-unizen: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libncurses5-dev' instead of 'libncurses-dev'
bison is already the newest version (2:3.0.4.dfsg-1).
flex is already the newest version (2.6.0-11).
gcc is already the newest version (4:5.3.1-1ubuntu1).
libncurses5-dev is already the newest version (6.0+20160213-1ubuntu1).
make is already the newest version (4.1-6).
python-serial is already the newest version (3.0.1-1).
python-setuptools is already the newest version (20.7.0-1).
gperf is already the newest version (3.0.4-2).
python-future is already the newest version (0.15.2-1).
git is already the newest version (1:2.7.4-0ubuntu1.6).
python is already the newest version (2.7.12-1~16.04).
python-cryptography is already the newest version (1.2.3-1ubuntu0.2).
python-pyparsing is already the newest version (2.0.3+dfsg1-1ubuntu0.1).
wget is already the newest version (1.17.1-1ubuntu1.4).
python-pip is already the newest version (8.1.1-2ubuntu0.4).
0 upgraded, 0 newly installed, 0 to remove and 22 not upgraded.

tangqingcai, 2019-03-26:
i met the same issue;
after my "make clean" command,then it can not "make menuconfig" successful;
TANQ@ASHANOBTANQ02 MINGW32 ~/esp/esp-idf/examples/get-started/hello_world
$ make menuconfig
make[1]: Entering directory '/home/TANQ/esp/esp-idf/tools/kconfig'
flex -L -Pzconf -ozconf.lex.c /home/TANQ/esp/esp-idf/tools/kconfig/zconf.l
make[1]: *** [Makefile:337: zconf.lex.c] Error 1
make[1]: Leaving directory '/home/TANQ/esp/esp-idf/tools/kconfig'
make: *** No rule to make target '/home/TANQ/esp/esp-idf/tools/kconfig/conf-idf', needed by '/home/TANQ/esp/esp-idf/tools/kconfig/mconf-idf'.  Stop.

TANQ@ASHANOBTANQ02 MINGW32 ~/esp/esp-idf/examples/get-started/hello_world
$ make menuconfig
make[1]: Entering directory '/home/TANQ/esp/esp-idf/tools/kconfig'
/usr/bin/gcc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MMD  /home/TANQ/esp/esp-idf/tools/kconfig/zconf.tab.c -o zconf.tab.o
In file included from /home/TANQ/esp/esp-idf/tools/kconfig/zconf.tab.c:2450:0:
/home/TANQ/esp/esp-idf/tools/kconfig/confdata.c: In function ‘conf_get_default_confname’:
/home/TANQ/esp/esp-idf/tools/kconfig/confdata.c:113:23: error: ‘PATH_MAX’ undeclared (first use in this function)
  static char fullname[PATH_MAX+1];
                       ^~~~~~~~
/home/TANQ/esp/esp-idf/tools/kconfig/confdata.c:113:23: note: each undeclared identifier is reported only once for each function it appears in
/home/TANQ/esp/esp-idf/tools/kconfig/confdata.c: In function ‘conf_write’:
/home/TANQ/esp/esp-idf/tools/kconfig/confdata.c:741:15: error: ‘PATH_MAX’ undeclared (first use in this function)
  char dirname[PATH_MAX+1], tmpname[PATH_MAX+1], newname[PATH_MAX+1];
               ^~~~~~~~
/home/TANQ/esp/esp-idf/tools/kconfig/confdata.c: In function ‘conf_split_config’:
/home/TANQ/esp/esp-idf/tools/kconfig/confdata.c:837:12: error: ‘PATH_MAX’ undeclared (first use in this function)
  char path[PATH_MAX+1];
            ^~~~~~~~
make[1]: *** [Makefile:171: zconf.tab.o] Error 1
make[1]: Leaving directory '/home/TANQ/esp/esp-idf/tools/kconfig'
make: *** No rule to make target '/home/TANQ/esp/esp-idf/tools/kconfig/conf-idf', needed by '/home/TANQ/esp/esp-idf/tools/kconfig/mconf-idf'.  Stop.

TANQ@ASHANOBTANQ02 MINGW32 ~/esp/esp-idf/examples/get-started/hello_world

Amna13, 2019-04-22:
Hey I have followed all steps in given tutorial https://docs.espressif.com/projects/esp-idf/en/latest/get-started/ but i am getting following attached error on running make menuconfig command. Kindly help me out if you know how to solve this issue.
 
[image]

miaoshifanren, 2019-07-05:
I think you should check your IDF_PATH, I had the same problems before, but I solve them with checking my IDF_PATH.

Atul1390, 2019-08-14:
/usr/bin/env: ‘python’: No such file or directory
make: *** No rule to make target 'check_python_dependencies', needed by 'menuconfig'.  Stop.
 i am getting this error after make menuconfig comand what migh be the issue .?

projectgus, 2019-08-19:
Finally got time to look at the issue mentioned by @william-ferguson-au , which is why this issue was reopened back in Feb:

> Sorry for the slow response. I was able to reproduce the initial problem with IDF v3.2-beta1. A clean
> build where the sdkconfig file is missing comes up with this error:
>
> make: *** No rule to make [...]

On revisiting, I'm not able to reproduce this with a clean build - only if I delete the `sdkconfig` file. The full error which should be printed in this case is this:

```
Can't run interactive configuration inside non-interactive build process.

Open a command line terminal and run 'make menuconfig' from there.
See esp-idf documentation for more details.
make: *** No rule to make target 'menuconfig', needed by 'path/examples/get-started/hello_world/sdkconfig'.  Stop.
```

... which is the case, we can't run menuconfig inside Eclipse (a new announcement is pending about this, hopefully we'll have a better answer soon.)

I'm not able to trigger a situation where I get the error message without the explanatory message above it, even with a high `-jN` number.

***

In the meantime, this issue has picked up a lot of very different bug reports from anyone who is getting errors in their build. It's hard to identify what each issue is, and most of the comments don't have enough detail to debug from.

Therefore, I'm going to close this issue and lock it:

**If you still have issues that look like any of the issues mentioned above, please first double-check the Getting Started Guide for your OS and check that all dependencies were installed, all environment variables set correctly, you didn't accidentally skip any step.**

**If everything looks OK after re-checking then please open a new issue here and provide all of the details which are asked for (IDF version, host OS, full commands and output, etc). With this information then we can help debug the problem.**

closed 2019-08-19 by projectgus