espressif/esp-idf #714: esp_wifi_set_config() documentation lacks detail about conf parameter

HubbyGitter, 2017-06-20:
[`esp_wifi_set_config()`](https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_wifi.h#L659) expects a parameter `wifi_config_t * conf`.

- Is it really a `wifi_config_t * const`? If yes, it could be defined like that. It would help if one has to keep the data after the call.
- Is the data in the struct referred to by the wifi implementation after the function call? Will `esp_wifi_set_config()` simply copy the `conf` pointer and rely on the caller to keep the struct memory allocated? Or is the data inside `conf`, e.g. in [`wifi_sta_config_t`](https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_wifi_types.h#L161), copied, such that it is safe to discard the `struct` after the call?

I came across [this code artefact](https://github.com/Ebiroll/esp32_weather/blob/master/main/app_main.c#L330) from @Ebiroll and was wondering whether my expectation that the wifi function will not copy > 100 bytes and the user code must keep the data safe is correct. Maybe it would be better to mention this explicitly.

HubbyGitter, 2017-06-21:
Actually, the code artefact with the config on the stack mentioned above is identical to t[his one from the https example](https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_request/main/https_request_example_main.c#L121). So the data can be discarded?

liuzfesp, 2017-07-04:
@HubbyGitter yes, you are right, the data can be discarded. We already have a document about WiFi API programming which will give more detailed description about the API usage, and this doc now is in review status, it will be ready in IDF release 2.1.

closed 2017-11-21 by FayeY