espressif/esp-idf #716: assertion failure on gatts_close_connection()

jaracil, 2017-06-20:
Hi.
When I close a gatts connection (ESP32 has gatts server role) this assertion failure is launched:

Assertion failure, esp-idf commit 7ed8c665471200613552e113fd0c1e2d017ea8f7

`/home/pepe/esp32/esp-idf/components/freertos/./timers.c:693 (prvProcessReceivedCommands)- assert failed!

abort() was called at PC 0x400869fb on core 0

0x400869fb: prvProcessReceivedCommands at /home/pepe/esp32/esp-idf/components/freertos/./timers.c:839

Backtrace: 0x40087054:0x3ffcfee0 0x40087153:0x3ffcff00 0x400869fb:0x3ffcff20 0x40086a3b:0x3ffcff60

0x40087054: invoke_abort at /home/pepe/esp32/esp-idf/components/esp32/./panic.c:517

0x40087153: abort at /home/pepe/esp32/esp-idf/components/esp32/./panic.c:517

0x400869fb: prvProcessReceivedCommands at /home/pepe/esp32/esp-idf/components/freertos/./timers.c:839

0x40086a3b: prvTimerTask at /home/pepe/esp32/esp-idf/components/freertos/./timers.c:839

Rebooting...
`

Thanks.

jaracil, 2017-06-21:
After some testing...
Assertion failure only happens when CONFIG_BLE_SMP_ENABLE=n

FayeY, 2017-11-21:
This issue should have been solved, please feel free to reopen it if your problem is still unsolved with the latest IDF.

closed 2017-11-21 by FayeY