espressif/esp-idf #717: How to know Bluetooth pairing has succeeded?

mehrvarz, 2017-06-20:
I do Bt-pairing just like it is done in the a2dp_sink example.  Until esp-idf June 12 I was able to use ESP_GAP_BLE_AUTH_CMPL_EVT to find out if pairing has worked out OK. But with esp-idf June 20, I don't seem to receive this event anymore. Now I only see these error messages:

```
        E (22082) BT: btm_sec_conn_req
        E (22102) BT: btm_sec_connected
        E (22122) BT: btc_dm_sec_cback : unhandled event (8)
        E (22582) BT: btc_dm_sec_cback : unhandled event (5)
        E (26852) BT: btc_dm_sec_cback : unhandled event (6)
        E (26852) BT: btc_dm_sec_cback : unhandled event (8)
```
How is code running on ESP32 supposed to find out if Bt-pairing has succeeded (or not)?

Yulong-espressif, 2017-07-24:
Hi @mehrvarz , the a2dp_sink example is a classic bt demo, not a BLE demo, when you pair succeeded, you will not receive a ESP_GAP_BLE_AUTH_CMPL_EVT, thanks.

mehrvarz, 2017-07-24:
OK. But I think it would be good to have some (other) type of indication on whether classic pairing has succeeded, failed or is still ongoing. It would allow me to provide proper feedback to the user, say, via LED blinking, etc.

mywang-espressif, 2017-08-08:
Hi @mehrvarz , I am afraid that there are actually no security-related APIs for classic bluetooth for now. The ESP_GAP_BLE_AUTH_CMPL_EVT you previously see is not intended to be used in the A2DP example. 
More Generic Access Profile functions for classic BT including pairing/bonding/device discovery/service discovery are in development, which will meet your requirement then.

pufstudio, 2017-09-11:
@mywang-espressif, what about avrcp vendor specific commands like metadata request for
track title, artist, duration, etc. ?

fanthos, 2017-09-14:
Thanks. Is there any plan for avrcp metadata or ETA for that?

ShendoXT, 2017-09-21:
Yes, when can we expect support for metadata?

fanthos, 2017-09-21:
I am reading the bluedroid code, seems that metadata is not that difficult to implement.

pufstudio, 2017-09-21:
I had the same impression after reading docs for bluetooth but I couldn't get it working.
It's like some crucial part of the code is missing...

I made a [thread on the official forums](https://www.esp32.com/viewtopic.php?f=13&t=2581&sid=58c232bdf8f7c33bab93bf7cae75bef2) regarding my findings.

If you get anything working, even incomplete or erroneous commands that are received on the
source side I'd very much appreciate it because I have been snooping with Wireshark and I got nothing which I find very strange.

fanthos, 2017-09-21:
Google already has the metadata support in platform-system-bt and licensed under Apache.

mywang-espressif, 2017-09-26:
Hi @pufstudio, APIs for  avrcp vendor specific commands like metadata are not provided yet, but the lower layer (BTA and stack) has support for them. I think the issue that you cannot get it working is that the feature bit "BTA_AV_FEAT_METADATA" is not set in the argument of function BTA_AvEnable, called in btc_avk.c.
Maybe the metadata message for AVRC can be provided in ESP-IDF release 3.1 or later.

pufstudio, 2017-09-30:
@mywang-espressif, thank you very much. That was a step in the right direction, although I'm not
quite there yet. All the parameters I set are discarded when command is sent.
I'll update my thread with a more detailed info on what I mean.

mywang-espressif, 2017-10-30:
Hi @pufstudio , I saw your progress in developing AVRC metadata from your thread[](https://www.esp32.com/viewtopic.php?f=13&t=2581), I am glad that you make it work. So could I conclude this issue and close this ticket?

pufstudio, 2017-10-30:
Hi @mywang-espressif, Yes I got what I needed so it can be closed.
I submitted a pull request so If anyone else needs this feature code is available.

closed 2017-10-31 by FayeY