espressif/esp-idf #718: Compilation fails with -O3 optimization - vfs_fat.c specifically

benpeoples, 2017-06-20:
```
CC src/vfs_fat.o
/Users/ben/esp/esp-idf/components/fatfs/src/vfs_fat.c: In function 'vfs_fat_readdir':
/Users/ben/esp/esp-idf/components/fatfs/src/vfs_fat.c:546:20: error: 'out_dirent' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     struct dirent* out_dirent;
                    ^
cc1: some warnings being treated as errors
make[1]: *** [src/vfs_fat.o] Error 1
make: *** [fatfs-build] Error 2

```

I have not dug into the issue beyond this.  Compiles fine with -O2.

igrr, 2017-06-21:
This probably means that there is a possibility that `vfs_fat_readdir_r` will return 0 while not initializing `out_dirent`. The code looks like this:

```c++
    FRESULT res = f_readdir(&fat_dir->ffdir, &fat_dir->filinfo);
    if (res != FR_OK) {
        ESP_LOGD(TAG, "%s: fresult=%d", __func__, res);
        return fresult_to_errno(res);
    }
```
Unless I'm missing something, `fresult_to_errno` can only return 0 if `res` is `FR_OK`. On the other hand, the compiler is rarely wrong about such things. Will run this through a static analyzer...

FayeY, 2017-06-25:
Hi @benpeoples , is this problem solved now?

igrr, 2017-06-25:
@FayeY Not yet, please assign to me.

closed 2018-05-18 by igrr