espressif/esp-idf #719: intr_alloc/esp_intr_free: Shared interrupts are disabled unconditionally

devsaurus, 2017-06-20:
A call to `esp_intr_free()`unconditionally disables the interrupt source via `esp_intr_disable()`. This breaks shared interrupts IMO. Disabling should be postponed in case we operate on a shared interrupt and the interrupt is still in use by another handle.

Proposal: Move `esp_intr_disable(handle)` from [L577](https://github.com/espressif/esp-idf/blob/master/components/esp32/intr_alloc.c#L577) after [L603](https://github.com/espressif/esp-idf/blob/master/components/esp32/intr_alloc.c#L603) where any interrupt (shared or unshared) is finally freed.

LDmitryL, 2017-08-09:
@igrr @devsaurus 
There is a hope that will fix this year? A very strong desire to work with these sensors!

igrr, 2017-08-18:
@devsaurus Could you please clarify how you are connecting the interrupts? Are you connecting single interrupt source to a shared interrupt input multiple times? Or are you connecting single interrupt source to multiple CPU interrupts?

devsaurus, 2017-08-18:
@igrr I have a single interrupt source (rmt to be specific) and have two ISRs serving it:
1. My ISR which handles protocol conversion on specific channels and clears the interrupt bit.
2. Everything else is handed down to the driver's ISR.

This works very well with the "shared interrupt" concept, and the code in `esp_intr_alloc` does the right thing building a list of ISRs that are called one after another. It's just `esp_intr_free()` which assumes that an interrupt source has to be disabled immediately.

igrr, 2017-08-19:
I see. I don't think we have considered the case of two handlers for same interrupt source in a shared interrupt, or even in separate interrupts. We may need to check if there are other handlers installed for the same source, assigned to different interrupt numbers.

How do you control the order in which the handlers are called? By enforcing specific order of interrupt registration?

Spritetm, 2017-08-21:
Fyi, we're working on fixing this. It's an interesting issue: the shared interrupts actually were meant for a different use case (multiple peripherals connected to one interrupt) but by a happy coincidence also works on multiple interrupts for the same source. 

There's one problem with this use case, however: disabling an interrupt handler (essentially what esp_intr_disable pertains to do) becomes slightly vague. We can not call the handler anymore, causing dropped interrupt events or worst case an interrupt loop because a hanging interrupt isn't acknowledged anymore. Alternatively, we can disable the interrupt source causing none of the handlers to be called anymore until the interrupt is re-enabled. Thirdly, we can decide that disabling a  handler for a shared source doesn't make sense and refuse to do it.

For your use case, would you have a preference for one of these alternatives?

devsaurus, 2017-08-22:
@igrr @Spritetm Thanks a lot for considering this use case!

> How do you control the order in which the handlers are called? By enforcing specific order of interrupt registration?

Yes, always have the "default" handler registered first and the specific one (if required) second.

> For your use case, would you have a preference for one of these alternatives?

Since I interpreted the shared interrupt scheme from the opposite direction as you pointed out, I'm probably not aware of the implications for the case "multiple peripherals connected to one interrupt". For my specific use case of "multiple interrupts for the same source" I assumed that **all** handles on the shared `intr` are triggered by the same `source`.
In this case, the decision for `free_shared_vector=true` on [line 646](https://github.com/espressif/esp-idf/blob/fc85cb683d4a5e4c2a5b4cb7d607a3790b0cfb96/components/esp32/intr_alloc.c#L646) seems to be key and `esp_intr_disable()` needs to be gated by this. That would map to the 3rd option "refuse to disable interrupt handler" (since other handles on the same `source` are still listening on that `intr`).

In case this approach is too specific and breaks other scenarios, I could also live with the first option and assume that the driver/application mutes the peripheral interrupt source before calling 'esp_intr_free()`. This seems to be good practice and would rule out the risk of interrupt loops IMO.

igrr, 2017-10-13:
Resolved in https://github.com/espressif/esp-idf/commit/c82e51cf79ea732a41454f6144513c08b8e4720e.

closed 2017-10-13 by igrr