espressif/esp-idf #720: Linker failure: undefined reference to `adjtime'

benpeoples, 2017-06-21:
Trying to work up a full NTP (not SNTP) implementation, which requires adjtime() to smoothly change the system clock.

I have a function prototype in `time.h`, but there doesn't appear to be a function definition anywhere.  

I can probably write my own `adjtime`, but am I just missing how to include the definition?

igrr, 2017-06-21:
Currently only `gettimeofday` and `settimeofday` are implemented. The most basic implementation of `adjtime` could fall back to `settimeofday`. Ideas on how to implement gradual adjustment specified by `adjtime` are welcome.

benpeoples, 2017-06-21:
I need a smooth time adjustment for my application.

It looks like two reasonable approaches would be either:

1. Adjust the boot time register (potentially only upon calls to `get_boot_time`)

2. Add some additional logic to the `frc_timer_isr()` to periodically adjust the `s_microseconds` upon ticking.

I'm going to think about this some more, but I think the best bet may be to replace this line: 

`s_microseconds += FRC1_ISR_PERIOD_US;`

in the ISR with something like this, conceptually:

```
s_microseconds += frc1_isr_period_adjust;
s_accumulated_offset += (frc1_isr_period_adjust - FRC1_ISR_PERIOD_US);
if( abs(s_accumulated_offset) > s_target_offset)  frc1_isr_period_adjust = FRC1_ISR_PERIOD_US;

```
So the `adjtime` function would set s_accumulated_offset to 0, and frc1_isr_period_adjust to some value, and then the isr would do the work.   Since this gets called every ~1.6 seconds and if we're targeting a Linux-like shift of about 85 seconds per second, that's typically +/- 20ms in each ISR call we're adding or removing.

Because the timestep happens only every 1.6 seconds, that could be hugely disruptive to time sensitive things, but would be OK in my application.   

---

The mod to get_boot_time would need to store a start time, potentially a rate, and a total amount of time to adjust.   Upon each call, it would apply a proportional amount of the adjustment to the return value.   If the complete adjustment is done, it would adjust the boot_time registers to match the new base time.  

This would have the advantage of being easily over-written by `settimeofday`, and not affect `system_get_time`.   

It also has no overhead except that it slows down `get_boot_time` by the time needed to calculate the offset (while slewing).   

I think I'm taking option 1 unless you see some particular danger in either of these approaches.   I'll post my solution, of course.

igrr, 2017-06-21:
Keep in mind that s_microseconds variable is only used when both FRC1 and RTC timers are used for timekeeping. Adjustments to s_microseconds also don't carry over deep sleep. For this reason I think that adjusting boot time might be a better approach.

benpeoples, 2017-06-21:
Good call on the deep sleep.   Here's my initial implementation, but is untested at the moment.  I'll issue as a PR once I have some basic testing.  Added `adjtime` function and modified `get_boot_time`

Presently ignoring the `outdelta`, in `adjtime`, but that would be fairly easy to add.

`// adjtime_start stores the start time of the slew
// adjtime_total is how many microseconds total to slew
// Maximum is 2^31 microseconds, or about 35 minutes.
// Typically 500,000 microseconds max for NTP
// Rate is the speed at which we add data.
// Unix typical rate is 85 seconds per second
// 6 = 64 seconds per second
// 7 = 128 seconds per second
static uint64_t adjtime_start = 0;
static int32_t adjtime_total = 0;
#define ADJTIME_RATE 7

// Adjtime takes two timeval structs
// delta is the tv to add or subtract from the clock
// outdelta is a way to find out where we are in the process (currently unsupported)
int adjtime(const struct timeval *delta, struct timeval *outdelta)
{
  if(delta == NULL)
    return 0;

  if(llabs((delta.tv_sec * 1000000UL + delta.tv_usec)) > INT_MAX)
    return -1;

  adjtime_start = get_time_since_boot();
  adjtime_total = (delta.tv_sec * 1000000UL + delta.tv_usec);

}

static uint64_t get_boot_time()
{
    uint64_t result;
    int64_t adjustment;
    _lock_acquire(&s_boot_time_lock);
#ifdef WITH_RTC
    result = ((uint64_t) REG_READ(RTC_BOOT_TIME_LOW_REG)) + (((uint64_t) REG_READ(RTC_BOOT_TIME_HIGH_REG)) << 32);
#else
    result = s_boot_time;
#endif
    _lock_release(&s_boot_time_lock);

    if(adjtime_start > 0)
    {
      adjustment = (get_time_since_boot() - adjtime_start) >> ADJTIME_RATE;
      if(adjtime_total < 0)
      {
        if(adjustment <= adjtime_total)
        {
          result = result + adjtime_total;
          set_boot_time(result);
          adjtime_start = 0;
        }
        else
        {
          result -= adjustment;
        }
      }
      else // if adjtime_total = 0 this also fires, but that's an infrequent case.
      {
        if(adjustment >= adjtime_total)
        {
          result = result + adjtime_total;
          set_boot_time(result);
          adjtime_start = 0;
        }
        else
        {
          result += adjustment;
        }
      }
    }

    return result;
}
#endif //defined(WITH_RTC) || defined(WITH_FRC1)
`

benpeoples, 2017-06-22:
OK, Github desktop doesn't seem to want to issue the PR, maybe I'm doing it wrong.   Here's the block of changes, this is tested a bit but not totally thoroughly, so there may be corner problems:

    #if defined(WITH_RTC) || defined(WITH_FRC1)
    static void set_boot_time(uint64_t time_us)
    {
        _lock_acquire(&s_boot_time_lock);
    #ifdef WITH_RTC
        REG_WRITE(RTC_BOOT_TIME_LOW_REG, (uint32_t) (time_us & 0xffffffff));
        REG_WRITE(RTC_BOOT_TIME_HIGH_REG, (uint32_t) (time_us >> 32));
    #else
        s_boot_time = time_us;
    #endif
        _lock_release(&s_boot_time_lock);
    }

    static uint64_t get_time_since_boot(); // FIXME: Is there a better way to do this?

    // adjtime_start stores the start time of the slew
    // adjtime_total is how many microseconds total to slew
    // Maximum is 2^31 microseconds, or about 35 minutes.
    // Typically 500,000 microseconds max for NTP
    // Rate is the speed at which we add data.
    // Unix typical rate is 85 seconds per second
    // 6 = 64 seconds per second
    // 7 = 128 seconds per second
    // Caveats:
    // 1. time adjustment progress does not survive deep sleep, but a completed adjustment does.
    // 2. keep time adjustments shortish, 35 minutes would take 74 hours to adjust.
    static uint64_t adjtime_start = 0;
    static int32_t adjtime_total = 0;
    #define ADJTIME_RATE 7

    // Adjtime takes two timeval structs
    // delta is the tv to add or subtract from the clock
    // outdelta is a way to find out where we are in the process (currently unsupported)
    // Trivial example:
    //    
    //    struct timeval delta;
    //
    //    delta.tv_sec = -1;
    //    delta.tv_usec = 0;
    //
    //    adjtime(&delta,NULL);
    // This removes 1 second from the clock.
    //
    int adjtime(const struct timeval *delta, struct timeval *outdelta)
    {
      if(delta == NULL)
        return 0;

      if(llabs((delta->tv_sec * 1000000L + delta->tv_usec)) > INT_MAX)
        return -1;

      adjtime_start = get_time_since_boot();
      adjtime_total = (delta->tv_sec * 1000000L + delta->tv_usec);

      return 0;
    }

    static uint64_t get_boot_time()
    {
        uint64_t result;
        int64_t adjustment;
        _lock_acquire(&s_boot_time_lock);
    #ifdef WITH_RTC
        result = ((uint64_t) REG_READ(RTC_BOOT_TIME_LOW_REG)) + (((uint64_t) REG_READ(RTC_BOOT_TIME_HIGH_REG)) << 32);
    #else
        result = s_boot_time;
    #endif
        _lock_release(&s_boot_time_lock);

        if(adjtime_start > 0)
        {
          adjustment = (get_time_since_boot() - adjtime_start) >> ADJTIME_RATE;
          if(adjtime_total < 0)
          {
            if(adjustment >= -adjtime_total)
            {
              result = result + adjtime_total;
              set_boot_time(result);
              adjtime_start = 0;
            }
            else
            {
              result -= adjustment;
            }
          }
          else // if adjtime_total = 0 this also fires, but that's an infrequent case.
          {
            if(adjustment >= adjtime_total)
            {
              result = result + adjtime_total;
              set_boot_time(result);
              adjtime_start = 0;
            }
            else
            {
              result += adjustment;
            }
          }
        }

        return result;
    }
    #endif //defined(WITH_RTC) || defined(WITH_FRC1)

Alvin1Zhang, 2019-03-13:
@benpeoples Hi, sorry to get back to you after so long. Does your issue still exist? Thanks.

benpeoples, 2019-03-13:
@Alvin1Zhang This issue has been resolved!  It appears that this patch (written a little more cleanly and with more documentation) is now in `time.c`

Thank you!

closed 2019-03-13 by benpeoples