espressif/esp-idf #721: Why were wifi_config_t.sta.ssid and .password changed from, and how would one set these fields with computed values?

HubbyGitter, 2017-06-21:
- Looking [at this](http://esp32.info/docs/esp_idf/html/d6/dc1/structwifi__sta__config__t.html), it ssems both fields were once `char`s.
 - However, [today both](https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_wifi_types.h#L161) are `uint8_t`s.

One way of setting these fields, [more or less seen here](https://gist.github.com/MakerAsia/37d2659310484bdbba9d38558e2c3cdb#file-esp32_wifi_dns-c-L54), would be:
> (void)memset(&comm_wifi_cfg, 0, sizeof(comm_wifi_cfg));
> (void)strncpy(comm_wifi_cfg.sta.ssid, "SSID", sizeof(comm_wifi_cfg.sta.ssid)-1);
> (void)strncpy(comm_wifi_cfg.sta.password, "PASS", sizeof(comm_wifi_cfg.sta.password)-1);
> comm_wifi_cfg.sta.bssid_set = false;   // to be on the safe side, although many would skip this

But it does not work with the fields being `uint8_t` because the compiler complains about the signedness of the pointer to the destination fields. (Yes, treating this warning as an error is a question of compiler flags. But in many cases, this behaviour is very valuable.)

Of course it's more cool to do follow [this initialisation approach](https://github.com/Ebiroll/esp32_weather/blob/master/main/app_main.c#L330) (although it's [probably not a good idea to do it with an automatic variable on the stack](https://github.com/espressif/esp-idf/issues/714)).

But what if you have to come up with computed strings?

I'm not saying it's impossible, it's simply rather, erm, _unhandy_, and what is the motivation of this datatype anyway? It is not even guaranteed by the C standard that 8 bits can hold a `char`.

nkolban, 2017-06-23:
A guess (on my part) would be that the specification is that ssid and password can be un-interpreted bytes.  The uint8_t is an 8 bit value while a char is a "character".  If an SSID is up to 32 bytes of data then a uint8_t[32] would be the closest representation of an SSID while a char[32] would say that we have 32 characters ... and in C/C++, a char (I believe) doesn't have to be 8 bits ... could it not be 16bits?

HubbyGitter, 2017-06-23:
@nkolban Yes, `char` could be larger than 8 bits, although I _really_ don't want to know how much code out there is actually broken in this regard.

Thank you for pointing out that the SSID is really defined as a sequence of 0 to 32 octets,
 although it [seems that this is not the end of the story](https://en.wikipedia.org/wiki/Service_set_(802.11_network)#Service_set_identifier_.28SSID.29): 
> Note that the 2012 version of the 802.11 standard defines a primitive SSID encoding, an enumeration of UNSPECIFIED and UTF-8, indicating how the array of octets can be interpreted.

Interestingly, since we're living in a C11 world, the potential UTF-8 interpretation of the SSID could be seen as an invitation to use UTF-8 literals (... u8"[SSID]" ...), although that [doesn't change much because these are represented by `char` arrays internally](http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1488.htm).

Why the compiler is happy initialising the `uint8_t[32]` array from regular character constant is not entirely clear to me. Apparently, this initialisation approach is [officially only defined for character and wide character arrays](http://en.cppreference.com/w/c/language/array_initialization). Yes, on the ESP32 platform, characters are represented by a signed 8-bit data type, so a cast is not impossible. But it could actually be non-portable.

So my conclusion is that
- the wifi data structure represents the bare network layer's view,
- (computed) non-human-readable SSIDs are directly supported,
- portable code to e.g. compute / concat SSIDs or passwords from text (artefacts) and an algorithm cannot be based on `str...()` and `mem...()` library functions.

HubbyGitter, 2017-06-26:
@FayeY I'm not sure this is an issue which has a reasonable solution within the scope of the IDF. One has to know how the wifi partner interprets the bytes anyway, and while there is no mandatory standard for this, ASCII is probably common and UTF-8 is a standardised option. So far, so bad, but it can't be generalised -- IDF could only provide helper functions for certain cases.

Regarding the potential problems when converting strings to byte streams: That's really an issue in the scope of the C language and standard library, and its practical relevance is low because the C11 struct initialisation approach works in practice (as long `char` is not defined as something exotic).

jmattsson, 2017-06-27:
SSIDs and passwords aren't limited to printable characters, they're unspecified *octets* and as such `uint8_t` is the appropriate type for those fields. [This stackoverflow entry](https://stackoverflow.com/questions/4919889/is-there-a-standard-that-defines-what-is-a-valid-ssid-and-password) has some more info and links.

liuzfesp, 2017-10-10:
HI @HubbyGitter we delicately modify the type from char to uint8_t, because the SSID is actually not '\0' terminated string.

jack0c, 2018-01-14:
@FayeY Please close this issue.

closed 2018-01-15 by FayeY