espressif/esp-idf #727: GATT UUID128s are in reverse order

rojer, 2017-06-23:
I don't know if this is intended or not, but 128-bit UUIDs have to be specified in reverse order.
so using this uuid:
```c
const uint8_t my_uuid[ESP_UUID_LEN_128] = { 0x00, 0x01, 0x02, ..., 0x0e, 0x0f };
```
will actually result in UUID
```
0f0e0d0c-0b0a-0908-0706-050403020100
```

the reason why this is done is not obvious. so... why?

negativekelvin, 2017-06-23:
They are transmitted as LSBF so commonly stored that way it seems.

> 	The 128-bit (16-byte) UUID as a human readable const-string. Format: XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX Upper and lower case supported. Hyphens are optional, but only upto four of them. The UUID is stored internally as a 16 byte array, LSB (little endian), which is opposite from the string.

https://docs.mbed.com/docs/ble-api/en/master/api/classUUID.html

rojer, 2017-06-23:
well, ok then. i wonder if it can be documented better in the examples, was a bit confusing why i couldn't find my service at first.

closed 2017-06-23 by rojer

hansmbakker, 2018-12-21:
I ran into the same issue, couldn't find this documented. @rojer can you please reopen this issue? I believe it should only be closed when it is documented.

rojer, 2018-12-22:
@hansmbakker i don't feel strongly about it. if you do, feel free to open a documentation request as a separate issue.