espressif/esp-idf #728: esp_ble_gatts_create_attr_tab regression

rojer, 2017-06-23:
@jack0c - after https://github.com/espressif/esp-idf/commit/491932b80cb27239fca4f8e07a36623afafd7a7f `esp_ble_gatts_create_attr_tab` stopped working - characteristics are just not registered anymore.
I isolated it to [this added return statement](https://github.com/espressif/esp-idf/commit/491932b80cb27239fca4f8e07a36623afafd7a7f#diff-f12d21a86d653101bdd60f613897b0a7R575) - if i remove it, everything goes back to normal.

There's another issue with this change: check for "one primary or one secondary service per table" is done twice: once in `btc_gatts_check_valid_attr_tab` and then again while creating services, via `btc_creat_tab_env.is_use_svc`. Maybe there's some reason to doing it twice, but i don't see it...

heyinling, 2017-06-26:
@rojer could you share your application code for this issue? I didn't meet your issue in my own application. I can create service by `esp_ble_gatts_create_attr_tab ` successfully on latest master (3bd3640d).

TianHao-Espressif, 2017-06-26:
@rojer , we do a more test, and there's no problem as you said. After this commit, the check of `esp_ble_gatts_create_attr_tab` is more strict. So you need to check the corresponding event status.

rojer, 2017-06-26:
i extracted a test standalone app from my current in-progress code it reproduces the issue - see attached archive.
basically, after https://github.com/espressif/esp-idf/commit/491932b80cb27239fca4f8e07a36623afafd7a7f there are no more `ADD_CHAR` events following the `CREAT_ATTR_TAB`.
app archive is attached. this is how it works with the change:

```
[Jun 26 21:37:07.629] I (1743) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
[Jun 26 21:37:07.894] I (2013) phy: phy_version: 355.1, 59464c5, Jun 14 2017, 20:25:06, 0, 0
[Jun 26 21:37:07.945] I (2043) main: BT init ok
[Jun 26 21:37:07.945] I (2053) main: LOCAL_IR
[Jun 26 21:37:07.945] I (2053) main: REG if 3 st 0 app 0
[Jun 26 21:37:07.945] I (2053) main: esp_ble_gap_config_adv_data 0
[Jun 26 21:37:07.945] I (2053) main: esp_ble_gatts_create_attr_tab 0
[Jun 26 21:37:07.945] I (2053) main: LOCAL_ER
[Jun 26 21:37:07.964] I (2063) main: CREAT_ATTR_TAB st 0 svc_uuid 5f6d4f53-5f52-5043-5f53-56435f49445f nh 7 hh 0x3ffc378c
[Jun 26 21:37:07.965] I (2073) main: ADV_DATA_SET_COMPLETE st 0
[Jun 26 21:37:07.965] I (2073) main: ADV_START_COMPLETE st 0
```

both return value from `esp_ble_gatts_create_attr_tab` and `status` passed to the `CREAT_ATTR_TAB` event are 0, meaning - OK, but theer are no `ADD_CHAR ` events an the service is not visible when enumerating.

if i remove the return statement, it starts to work again:

```
[Jun 26 21:39:30.740] I (1742) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
[Jun 26 21:39:30.995] I (2002) phy: phy_version: 355.1, 59464c5, Jun 14 2017, 20:25:06, 0, 0
[Jun 26 21:39:31.046] I (2032) main: BT init ok
[Jun 26 21:39:31.046] I (2042) main: LOCAL_IR
[Jun 26 21:39:31.046] I (2042) main: REG if 3 st 0 app 0
[Jun 26 21:39:31.046] I (2042) main: esp_ble_gap_config_adv_data 0
[Jun 26 21:39:31.046] I (2042) main: esp_ble_gatts_create_attr_tab 0
[Jun 26 21:39:31.047] I (2042) main: LOCAL_ER
[Jun 26 21:39:31.068] I (2052) main: CREAT_ATTR_TAB st 0 svc_uuid 5f6d4f53-5f52-5043-5f53-56435f49445f nh 7 hh 0x3ffc378c
[Jun 26 21:39:31.069] I (2062) main: ADV_DATA_SET_COMPLETE st 0
[Jun 26 21:39:31.069] I (2062) main: CREATE st 0 svch 40 svcid 5f6d4f53-5f52-5043-5f53-56435f49445f 0 primary
[Jun 26 21:39:31.090] I (2072) main: ADD_CHAR st 0 ah 42 svch 40 uuid 5f6d4f53-5f52-5043-5f64-6174615f5f5f
[Jun 26 21:39:31.091] I (2082) main: mos_rpc_data_ah = 42
[Jun 26 21:39:31.091] I (2082) main: ADD_CHAR st 0 ah 44 svch 40 uuid 5f6d4f53-5f52-5043-5f72-785f63746c5f
[Jun 26 21:39:31.113] I (2092) main: mos_rpc_rx_ctl_ah = 44
[Jun 26 21:39:31.113] I (2102) main: ADD_CHAR st 0 ah 46 svch 40 uuid 5f6d4f53-5f52-5043-5f74-785f63746c5f
[Jun 26 21:39:31.113] I (2102) main: mos_rpc_tx_ctl_ah = 46
[Jun 26 21:39:31.114] I (2112) main: ADV_START_COMPLETE st 0
[Jun 26 21:39:31.117] I (2112) main: START st 0 svch 40
```

and service is enumerated properly.
i'm not on the master, but only a couple days behind - this is my branch: https://github.com/cesanta/esp-idf/commits/2.0-r8
[test_app_728.zip](https://github.com/espressif/esp-idf/files/1103446/test_app_728.zip)

heyinling, 2017-06-27:
@rojer `esp_ble_gatts_create_attr_tab ` won't send `ADD_CHAR` event to app now. sending `ADD_CHAR` is not part of the original design.

The proper way to get attribute handle is to get them from `ESP_GATTS_CREAT_ATTR_TAB_EVT`.

```c
    struct gatts_add_attr_tab_evt_param{
        esp_gatt_status_t status;       /*!< Operation status */
        esp_bt_uuid_t svc_uuid;         /*!< Service uuid type */
        uint16_t num_handle;            /*!< The number of the attribute handle to be added to the gatts database */
        uint16_t *handles;              /*!< The number to the handles */
    } add_attr_tab;                     /*!< Gatt server callback param of ESP_GATTS_CREAT_ATTR_TAB_EVT */
```

first check if `add_attr_tab.num_handle` equals the length of your table, then get handle by `add_attr_tab.handles`.

rojer, 2017-06-27:
ok, that's fine, i can modify the code to extract attr handles from add_attr_tab.
however, the service is still not enumerated during scan.

heyinling, 2017-06-27:
service data will not be auto enumerated in adv data. we need to config service uuid in adv data.

You can either use `esp_ble_gap_config_adv_data` (setting `service_uuid_len` and `p_service_uuid` of `esp_ble_adv_data_t`, refer to example `gatt_server`), or use `esp_ble_gap_config_adv_data_raw`, manually config adv raw data according to [Spec CSSv7](https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=421047&_ga=2.237809350.716314775.1498546720-1420322536.1477580329).  I prefer to use `esp_ble_gap_config_adv_data_raw ` to set adv data, the adv data is entirely under control with this API.

rojer, 2017-06-27:
@heyinling i already do that, see line 161 in main.s. again, everything works up until that specific change in esp-idf that i described.

rojer, 2017-06-27:
notice that `ESP_GATTS_START_EVT` never arrives, so service is never started.

rojer, 2017-06-27:
... and that is because `ESP_GATTS_CREATE_EVT` does not arrive.

heyinling, 2017-06-27:
need to start service in processing event ESP_GATTS_CREAT_ATTR_TAB_EVT, other create/add event now will not be passed to APP.  and got totally wrong on previous problem, i was thinking there're problems with advertising.

rojer, 2017-06-27:
but i need service handle for `esp_ble_gatts_start_service` and `ESP_GATTS_CREAT_ATTR_TAB_EVT` doesn't give me that?

heyinling, 2017-06-27:
service is also defined in your table, you can get service handle by the index you defined service in table

rojer, 2017-06-27:
ok, that works, thanks!
in fact, i still see CREATE and ADD_CHAR events, they still arrive, but the key was to start the service from CREAT_ATTR_TAB handler. ok, i can live with that.

thanks!

closed 2017-06-27 by rojer