espressif/esp-idf #730: I2C API uses portBASE_TYPE /  BaseType_t instead of portTICK_TYPE / TickType_t for timeouts; I2C implementation does not account for OS ticks overflow and could block for 500 days; and I2C API documentation does not relate to the tick base

HubbyGitter, 2017-06-24:
The I2C API uses `portBASE_TYPE` (which is supposed to be signed and defined as `int` for ESP32 IDF)
 [here](https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/i2c.h#L344), [here](https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/i2c.h#L361) and [here](https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/i2c.h#L378) to represent a number of OS ticks, which would better be represented by `portTICK_TYPE` (old) / `TickType_t` (new) in the context of the FreeRTOS API -- negative values don't make sense anyway.
<hr>

While [FreeRTOS itself takes care of tick overflows](https://github.com/espressif/esp-idf/blob/master/components/freertos/tasks.c#L1300), the [I2C implementation does not](https://github.com/espressif/esp-idf/blob/master/components/driver/i2c.c#L937).

With ticks @ 100 Hz  and TickType_t as unsigned 32 bit, applications may see unexpected results after roughly 500 days of CPU uptime. All it needs is a relatively generous timeout and an API call shortly before the tick counter overflow. Maybe someone even experimets with a 16 bit tick count, which is an option for FreeRTOS and would increase the likelihood drastically.

The other weakness of the implementation lies in the task synchronisation in combination with the computation of the destination tick count. If the functions have to wait longer for the first semaphore than the delay provided via the call, then the subtraction will underflow and the call timeout will effectively be around 500 days.

This scenario can occur if the task acessing I2C is interrupted or the semaphore is held by higher priority tasks "for long enough".

Both problems could be easily fixed at the same time by using `vTaskDelayUntil()` instead of `vTaskDelay()`.
<hr>

The documentation, [e.g. here](https://github.com/espressif/esp-idf/blob/master/components/driver/i2c.c#L959), simply refers to `ticks`. Since I2C has to be configured with a `clk_speed`, it might be helpful for the user if it was made clear that these are really OS ticks, maybe with e reference to `portTICK_PERIOD_MS`.

tdesmet, 2017-06-27:
Could https://github.com/espressif/esp-idf/issues/680 be related to this?

HubbyGitter, 2017-06-27:
@tdesmet It _could_ be the case that you suffer from the "other weak aspect" I described, with the likelihood depending on you using one or two cores, having higher priority tasks which can let your I2C calls wait, etc. But the symptoms you mention go beyond what the bug I mentioned can do (unless timeout handling would be broken, which I didn't check).

Having a quick glance, it seems to me that one more [code line in `i2c_master_cmd_begin()` might be suspicious](https://github.com/espressif/esp-idf/blob/master/components/driver/i2c.c#L942). There, the semaphore is polled (instead of being waited for) and the return value ignored (`xTicksToWait` is `0`). This means that if the semaphore cannot be obtained, the later `xSemaphoreGiveFromISR()` calls in `i2c_master_cmd_begin_static()` will have no match.

Interestingly, while[ FreeRTOS requires that a semaphore _must_ be taken before it can be given from a non-ISR context](http://www.freertos.org/a00123.html), the same is [not mentioned for the ISR case](http://www.freertos.org/a00124.html). But I would assume that the rule must still be obeyed, because FreeRTOS maps semaphores to queues, and the command simply sends an item to the queue, this might get the semaphore mechanism out of balance.

Whether or not this is a bug depends on FreeRTOS details and a potential assertion that the semaphore can always be obtained without waiting (which would be worth a comment in the source code).

If I were you I'd probably try to replace that by `res = xSemaphoreTake(p_i2c->cmd_sem, ticks_to_wait); if (res == pdFALSE) { return ESP_ERR_TIMEOUT; } and see what happens. ;)

HubbyGitter, 2017-06-28:
@tdesmet My above suggestion probably won't fix your problem, but it could help reduce problems. The assumption that `p_i2c->cmd_sem` can successfully be taken on entry to `i2c_master_cmd_begin()` is in principle valid _if_ the other code is robust.

But in the sense of a more robust code, I believe [this line](https://github.com/espressif/esp-idf/blob/master/components/driver/i2c.c#L942) should be chaged to `res = xSemaphoreTake(p_i2c->cmd_sem, 0); if (res == pdFALSE) { return ESP_FAIL; }`. This suggestion is valid for a case where timeout handling can be cleanly under control such that the semaphoe _will definitely_ be available _immediately_ after `i2c_master_cmd_begin()` detects a timeout and releases the `p_i2c->cmd_mux` mutex. Regarding `handle_timeout()`, see below.

While the priority of the task calling `i2c_master_cmd_begin()` is lifted to avoid conflicts with other tasks calling the same function, this does not necessarily help much against the race conditions in the code, which I described initially. E.g. if the execution of `i2c_master_cmd_begin()` "starves" between `[portTickType ticks_end = xTaskGetTickCount() + ticks_to_wait;`](https://github.com/espressif/esp-idf/blob/master/components/driver/i2c.c#L937) and [`ticks_to_wait = ticks_end - xTaskGetTickCount();`](https://github.com/espressif/esp-idf/blob/master/components/driver/i2c.c#L959) and the timeout is tight, then a timeout will be signalled although it does not necessarily occur.

My question would be, is timeout handling cleanly implemented? Upon a timeout "detection", `i2c_master_cmd_begin()` simply sets `p_i2c->status = I2C_STATUS_DONE;` which also tells `i2c_master_cmd_begin_static()` to do nothing if more data is signalled via interrupt. -- not even give back `p_i2c->cmd_sem`.

I think there are two problems with this. First, again, there is a race condition between waiting for `p_i2c->cmd_sem` and setting `p_i2c->status = I2C_STATUS_DONE;`, and second, after the first timeout occurs, `p_i2c->cmd_sem` is in trouble.

You could try the following modification, but be aware that _if_ it does anything good then it would only solve the behaviour after a first timeout:

```C
res = xSemaphoreTake(p_i2c->cmd_sem, ticks_to_wait);
if (res == pdFALSE) {
    (void)xSemaphoreGive(p_i2c->cmd_sem);   /* <---- ADD THIS LINE */
    ret = ESP_ERR_TIMEOUT;
} else if (p_i2c->status == I2C_STATUS_ACK_ERROR) {
```

I'd recommend also making the other modification to be warned about problems when taking the semaphore:

```C
if (res == pdFALSE) {
    return ESP_ERR_TIMEOUT;
}
/* vvv replaced vvv */
res = xSemaphoreTake(p_i2c->cmd_sem, 0);
if (res == pdFALSE) {
    return ESP_FAIL;
}
/* ^^^ replaced ^^^ */
i2c_reset_tx_fifo(i2c_num);
i2c_reset_rx_fifo(i2c_num);
```

FayeY, 2017-11-22:
This issue should have been solved on the latest IDF, if it is not, please feel free to reopen it.

closed 2017-11-22 by FayeY