espressif/esp-idf #732: Internal Temperature Sensor: Presumed Silicon Erratum; and Several Related Documentation Issues

HubbyGitter, 2017-06-26:
The internal temperature sensor is an interesting feature. Yes, the sensor yields results which are heavily influenced by current silicon power dissipation, with two major contributors (according to my observations) being WiFi and 240 MHz operation (which should not be much of a surprise). And yes, the sensor is certainly not very precise or even calibrated. However, if one wants to monitor system health, it does not matter what the reason for a high temperature is.

Having said that, I find it interesting that regarding non-ULP usage of the sensor, the undocumented `temprature_sens_read()` as well as the test case in `test_tsens.c` do not deploy the `SENS_TSENS_RDY_OUT` bit in the `SENS_SAR_SLAVE_ADDR3_REG` register. Rather, they rely on `ets_delay_us()` which, [according to its documentation](https://github.com/espressif/esp-idf/blob/master/components/esp32/include/rom/ets_sys.h#L366), is not necessarily meant to be used inside FreeRTOS tasks.

In addition, both implementations are based on the `SENS_FORCE_XPD_SAR` bits in the `SENS_SAR_MEAS_WAIT2_REG` register. While [their semantics is left open at the point of their declaration](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/sens_reg.h#L88), one gets the idea [from this comment in hall sensor related code](https://github.com/espressif/esp-idf/blob/master/components/driver/rtc_module.c#L585). Unfortunately, the register is also not mentioned in the technical reference manual.

After some tests, I have seen no sign of life from `SENS_TSENS_RDY_OUT`, and since the reference implementation prefers a fixed delay, here is my conclusion.

- The dysfunctional `SENS_TSENS_RDY_OUT` bit should be mentioned in the "ECO and Workarounds" document, along with an explanation how to calculate the required delays (the first one seems to be fixed to 100 µs, but the second one seems to depend on the divider and probably the clock base -- but how?) Also, it should be clarified whether the use of `ets_delay_us()` is appropriate, and whether it has any side effects
- The `SENS_SAR_MEAS_WAIT2_REG` should be documented in the technical reference manual.
- Please add the temperature sensor in fig. 74 and table 84 of the techical reference manual.
- Please Explain the exact meaning of  `SENS_TSENS_CLK_DIV` one one hand and `Wait_Delay` for the ULP `TSENS` command on the other hand, how the two values relate to each other, and how long the measurement will actually take.
- Please add the description for `SENS_FORCE_XPD_SAR` at the point of its declaration.

igrr, 2017-06-27:
(We'll add the relevant bits of this as comments in the header files; just an explanation in the meantime)

`SENS_TSENS_CLK_DIV` sets the clock divider for TSENS module. Source clock is RTC_FAST_CLK (~8MHz usually). The value of `SENS_TSENS_CLK_DIV` can be any in the range from 2 to 255.

In normal operation, TSENS is controlled by the ULP. ULP has a `TSENS` instruction, which takes `wait_delay` value as its second argument. This value sets measurement duration, in TSENS cycles (i.e. RTC_FAST_CLK/SENS_TSENS_CLK_DIV). On the ULP, TSENS instruction retires once the measurement is complete and the result is written into the register.

When TSENS is instead controlled by software using RTC registers, there is no mechanism to count down `wait_delay` of `RTC_FAST_CLK/SENS_TSENS_CLK_DIV` cycles. Software needs to set `SENS_TSENS_POWER_UP` from 0 to 1 to start TSENS, and then needs to set `SENS_TSENS_DUMP_OUT` to stop the measurement. In the test code this delay is achieved using `ets_delay_us`. `ets_delay_us` does not have side effects, but due to preemption real delay may be higher than the requested one. It is also a "busy wait", which is the reason why we don't generally recommend to use it as a go-to solution for delays in FreeRTOS tasks. In a fleshed out driver this should rather be achieved using some other mechanism, such as using one of the Timer Group timers with an interrupt.
A few (~3-4) TSENS cycles after software sets `DUMP_OUT`, TSENS writes the measured value into `SENS_TSENS_OUT` register and signals this by setting `SENS_TSENS_RDY_OUT` bit high.
One can replace the second delay in the test code with something like
```c++
    while(REG_GET_FIELD(SENS_SAR_SLAVE_ADDR3_REG, SENS_TSENS_RDY_OUT) == 0) {
        ;
    }
```
which would work as well, so `SENS_TSENS_RDY_OUT` is not dysfunctional.

I'll pass your TRM related requests to the documentation team.

igrr, 2017-06-27:
Regarding documentation changes,
> Please add the temperature sensor in fig. 74 and table 84 of the techical reference manual.

Do you mean that TSENS should be somehow related to the SARADC in the diagram?

HubbyGitter, 2017-06-27:
@igrr Thank you for this clear and extensive response!

Sorry for the false claim that `SENS_TSENS_RDY_OUT` appears to be dysfunctional. I can confirm that the `test_tsens.c` code, when replacing the second `ets_delay_us()` by your `while`, works for me.

Would it make sense to change the test case code to actually validate `SENS_TSENS_RDY_OUT` this way? The very short busy waiting seems to make sense.

What irritates me is that I deterministically get different readings from `temprature_sens_read()` and the `test_tsens.c` code (e.g. 116 vs. 124) when I run them one after the other without external effects kicking in . While both wait for 100 µs before they pick up the results, there are two differences between the two implementations:

- The initialisation in `test_tsens.c` is more clean; `temprature_sens_read()` switches to non-ULP control without `SENS_TSENS_POWER_UP` and `SENS_TSENS_DUMP_OUT` being set first.
- The `SENS_TSENS_CLK_DIV`, which has a default of 6, is set to `10` in `test_tsens.c` while `temprature_sens_read()` uses `2`.

You said that `SENS_TSENS_CLK_DIV` is "out of control" of non-ULP software. All the more it would be good to know what effects it has and whether the waiting time (so far always 100 µs) needs to be adjusted in accordance. If I modify the `test_tsens.c` code to use a `SENS_TSENS_CLK_DIV` of `2`, it will indefinitely wait for `SENS_TSENS_RDY_OUT`. With `3`, it will work again and the readings are close to those of `temprature_sens_read()`. And with `20`, the differences are similar to using `10`.

Does this mean that a `SENS_TSENS_CLK_DIV` of 10 and waiting for 100 µs is the correct setting for an accurate reading? It's a bit suspicious that `temprature_sens_read()` also waits for 100 µs while it uses a much smaller clock divider.

<hr>

Regarding fig. 74. Currently, the whole TSENS ADC functionality seems to be coming out of nowhere in the non-ULP domain. (The same is true in the ULP domain but at least one doesn't have to worry because the FSM is taking care of it all.)  It yields 8 bits no matter what you do, and neither the routing nor the occupied ADC and / or controller are mentioned -- unlike the Hall sensor.

So , which SAR ADC is used, and which FSM and power is really switched on and off by `SENS_FORCE_XPD_SAR` (aka what other functionalities are potentially affected by taking away the TSENS ADC from the ULP / RTC domain)?

igrr, 2017-06-27:
> Would it make sense to change the test case code to actually validate SENS_TSENS_RDY_OUT this way? The very short busy waiting seems to make sense.

We try not to busy-wait on reading from RTC clock domain, if this is possible (in some cases, it is the only way). Transactions into that clock domain happen at RTC_FAST_CLK speed. Polling an RTC register from one CPU increases latency for the other CPU, if it tries to access RTC at the same time. Plus, delay is fairly predictable here. That is the reason why I have originally used ets_delay_us in that code. It is also acceptable to replace it with polling, which won't take long time in this particular case. Because this is just a piece of test code, it's not really important.
Another point to note though is that `SENS_TSENS_RDY_OUT` goes high for one TSENS clock cycle. Which may explain why you never see that it gets set if you set divider to 2 (i.e. it is set for two RTC_FAST_CLK cycles and then cleared). From software perspective this is not very convenient, and in the next chip the behavior is changed so that this bit gets latched.

With regards to the TSENS clock divider and total delay... I haven't explained this very well I think.
Based on my understanding, only the total measurement time should matter for the output value. There is no "right" measurement time. The longer the measurement time, the higher the dynamic range of the reported values. Hence, if you plan to calibrate TSENS, you need to do this for a specific delay time. Your observations do not fully agree with this, in the sense that you see that for low values of the divider the output values are different, even if total time is the same. I'll need to check with my colleagues about this issue. This doesn't matter much as long as parameters (divider and total delay) are fixed, but still might be interesting to figure out why this happens.
The fact that the value obtained with the divider of 20 is similar to the one obtained with a divider of 10 (but not with a divider of 2) makes me think that this effect is indeed an edge case for high TSENS clock speeds.

>So , which SAR ADC is used, and which FSM and power is really switched on and off by `SENS_FORCE_XPD_SAR`

`SENS_FORCE_XPD_SAR` is indeed the only thing which is common between TSENS and the two SARADCs. Power control of TSENS is done via the same register as the SARADCs. Aside from that, TSENS is a stand-alone block, it doesn't interact with SARADCs in any way. `SENS_FORCE_XPD_SAR` affects TSENS, SARADCs, and the Hall sensor.

HubbyGitter, 2017-06-27:
- The `test_tsens.c` is he most prominent piece of officially available information as of now, hence my question. Disassembling `temprature_sens_read()` seemed only worthwhile because I learned about the CPU. (Although it finally revealed a few interesting aspects.)
- Polling is a no-go in my view if `SENS_TSENS_RDY_OUT` is only high for one cycle, no matter how long that cycle is, because anything can happen to the polling task while it polls.
- I guess using `ets_delay_us()` still bears the risk that the CPU might be accelerated while the busy waiting takes place. Not very likely, but possible.
- "Next chip" = "next chip revision"?
- Talking of revisions, I guess I have a V1.0 but `EFUSE_BLK0_RDATA3_REG` is all zero, am I missing something there, too? And by the way, the header file comments are different from the reference manual regarding this register. According to the header file, at least one bit is supposed to be set for chip version one, as far as I understand it.

Thank you very much for sharing your knowledge.

igrr, 2017-06-27:
We have a task for adding an API for temperature sensor, which will likely be addressed for the next major version; that implementation should take all these things into account. The point about polling being unreliable due to task preemption is a very good one. If you have other points to make about temperature sensor API, please use the issue you have linked above.

>Talking of revisions, I guess I have a V1.0 but EFUSE_BLK0_RDATA3_REG is all zero

The best way to check is to run get-started/hello_world example — it will print the chip revision. If it shows rev.0, then you indeed have rev.0.

> "Next chip" = "next chip revision"?

Sorry, can't share more details at this point.

HubbyGitter, 2017-06-27:
@igrr Thank you! Posted screenshots [over there](https://github.com/espressif/esp-idf/issues/146).

eyaleb, 2017-06-29:
> The best way to check is to run get-started/hello_world example — it will print the chip revision. If it shows rev.0, then you indeed have rev.0.

A bit OT, but I have a rev. 0 module. When I check
```
esp_chip_info_t chip_info;
esp_chip_info(&chip_info);
printf ("Chip model %d, revision %d, #cores %d\n",
                chip_info.model, chip_info.revision, chip_info.cores);
```
I get
```
Chip model 0, revision 0, #cores 2
```
yet the header `esp_system.h` says
```
typedef enum {
    CHIP_ESP32 = 1, //!< ESP32
} esp_chip_model_t;
```
Should the model say `1` for the esp_32 (using a WROOM-32)?

igrr, 2017-06-29:
That's a bug, `get_chip_info_esp32` doesn't set `model` field of `out_info` at all.

HubbyGitter, 2017-06-29:
@eyaleb Yes, [according to the header](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/efuse_reg.h#L82), I expected `EFUSE_RD_CHIP_VER_REV1` to be `1`, but on my chip it isn't, so I guess this story starts at hardware level already. Or does anybody get a `1` there?

igrr, 2017-06-29:
@HubbyGitter what is the date code on the chip?

HubbyGitter, 2017-06-29:
@igrr What's printed on the chip is extremely hard to read. Very dark. Had to take a macro photo with specific angle related to light source. If I had to guess, the second line probably reads "392016" (the "3" is not too clear but unlikely to be an 8, and rather similar to the unidentifiable digit in "ESP32"...". Could be a calendar week?

negativekelvin, 2017-06-29:
Yes it is a week and version 1 starts at 122017

HubbyGitter, 2017-06-29:
@negativekelvin Did you mean "revision 1 started at 12/2016", or are you saying that all current chips are "V 0 Rev 0 or 1" and 12/2017 there will be a "V 1 Rev something"?

Either way, the bottom line is that as of today, it's not possible to write a piece of code which will identify any future chip versions (referring to major and minor information) correctly, even if it were compatible and just wanted to output the chip version for the sake of debugging.

Is there a chance you _now_ agree on a future-proof format for the version, revision etc, such that we don't have to update and flash software later just to find out on which silicon it's running?

igrr, 2017-06-29:
> Either way, the bottom line is that as of today, it's not possible to write a piece of code which will identify any future chip versions

How is this so, aside from the bug mentioned by @eyaleb above?

There is currently one supported chip (ESP32) and two revisions: v0 and v1, as well as a few package variants (q6/q5/q5 with flash). All can be identified using this API.

If there are more chip models or revisions of the existing model, they will receive incremental numbers (2, 3, etc).

> revision 1 started at 12/2017

The first batch which wasn't pre production was indeed on week 12 of 2017.

HubbyGitter, 2017-06-29:
@igrr  Sorry for my imprecise wording. I meant code that is written and _compiled_ (and potentially distributed) today. Only applies to such scenarios as:

- You're developing something, it works for you, you flash it, the board dies, you order another one and flash it, firmware doesn't work correctly, you look at the debug output, it indicates an unsuspicious chip revision, you send the board back claiming that it has a defect.
- You're developing some piece of high quality firmware, don't disclose the source, but leave the rest to the user. Users complain about firmware not working, you ask them to supply debug output, and it's not showing anything unusual regarding the chip.

I just seem to think that reserving more than a single bit and defining a semantics for them upfront wouldn't hurt much and might be very helpful in some situations.

igrr, 2017-06-29:
I don't think that you can assume that the code compiled with ESP-IDF, say, v2.1 will even start up on subsequent revisions or models of the chip. Even if your C code will not need any changes, recompilation against newer ESP-IDF version may be required. Current versions of ESP-IDF are in now way guaranteed to produce software which runs on newer revisions or models.

> reserving more than a single bit and defining a semantics for them upfront 
Thank you for the suggestion, that would indeed be a good thing to do.

HubbyGitter, 2017-06-29:
@igrr 

> I don't think that you can assume that the code compiled with ESP-IDF, say, v2.1 will even start up on subsequent revisions or models of the chip.

I'm referring to scenarios where it _does_ start and then later something goes wrong e.g. with some periphery, and you're trying to exclude root causes, and your firmware is indicating a wrong chip revision, and then you're wasting time and money based on a false assumption.

> Thank you for the suggestion, that would indeed be a good thing to do.

Thank you for your feedback, and keep up the excellent work.