espressif/esp-idf #734: freertos/./heap_regions.c:369 (vPortFreeTagged)- assert failed!

joysfera, 2017-06-26:
Hello there,

I've got an application that waits for data on TCP socket, then sends them to UART, waits 100 ms, reads data from UART and send them via TCP. And it's crashing randomly in FreeRTOS' heap_regions.c like this:

```
/home/joy/esp/esp-idf/components/freertos/./heap_regions.c:369 (vPortFreeTagged)- assert failed!
abort() was called at PC 0x400850d4 on core 0
0x400850d4: vPortFreeTagged at /home/joy/esp/esp-idf/components/freertos/./heap_regions.c:410

Backtrace: 0x4008790c:0x3ffbf7e0 0x40087a0b:0x3ffbf800 0x400850d4:0x3ffbf820 0x40087f51:0x3ffbf840 0x40081f59:0x3ffbf860 0x4000bec7:0x3ffbf880 0x4010d3b5:0x3ffbf8a0 0x40108124:0x3ffbf8c0
0x4008790c: invoke_abort at /home/joy/esp/esp-idf/components/esp32/./panic.c:519
0x40087a0b: abort at /home/joy/esp/esp-idf/components/esp32/./panic.c:519
0x400850d4: vPortFreeTagged at /home/joy/esp/esp-idf/components/freertos/./heap_regions.c:410
0x40087f51: vPortFree at /home/joy/esp/esp-idf/components/esp32/./heap_alloc_caps.c:323
0x40081f59: _free_r at /home/joy/esp/esp-idf/components/newlib/./syscalls.c:32
0x4010d3b5: sys_timeouts_mbox_fetch at /home/joy/esp/esp-idf/components/lwip/core/timers.c:570
0x40108124: tcpip_thread at /home/joy/esp/esp-idf/components/lwip/api/tcpip.c:474
```

or like this:
```
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40084fd2  PS      : 0x00060b33  A0      : 0x80087fef  A1      : 0x3ffc28b0  
0x40084fd2: pvPortMallocTagged at /home/joy/esp/esp-idf/components/freertos/./heap_regions.c:410

A2      : 0x00015c18  A3      : 0x00000000  A4      : 0x0edc0005  A5      : 0x0000064c  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x0edc0005  A9      : 0x3fff000a  
A10     : 0x3ffb1308  A11     : 0x00060f20  A12     : 0xff000000  A13     : 0x80ffffff  
A14     : 0x3f40a6a4  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0edc0009  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x40084fd2:0x3ffc28b0 0x40087fec:0x3ffc28d0 0x4008802d:0x3ffc28f0 0x40081f4d:0x3ffc2910 0x4000beaf:0x3ffc2930 0x400fe25e:0x3ffc2950 0x400fda8c:0x3ffc2980 0x400f9ebf:0x3ffc29e0
0x40084fd2: pvPortMallocTagged at /home/joy/esp/esp-idf/components/freertos/./heap_regions.c:410
0x40087fec: pvPortMallocCaps at /home/joy/esp/esp-idf/components/esp32/./heap_alloc_caps.c:372
0x4008802d: pvPortMalloc at /home/joy/esp/esp-idf/components/esp32/./heap_alloc_caps.c:306
0x40081f4d: _malloc_r at /home/joy/esp/esp-idf/components/newlib/./syscalls.c:27
0x400fe25e: esf_buf_alloc at ??:?
0x400fda8c: wdevProcessRxSucDataAll at ??:?
0x400f9ebf: ppTask at ??:?
```

My application does not allocate memory dynamically, has minimal static buffers, and a large stack (tried 8, 16 and 32 kB). The TCP connection is handled by mongoose library. I don't see any obvious bug in my code and I don't like it crashing deep in FreeRTOS memory management called from lwip.

The esp-idf is up-to-date. ESP32 chip revision = 0. Any idea what's wrong? Or what I could try to debug it better? Thanks!

projectgus, 2017-06-27:
These crashes & assertion indicate heap corruption, so something in the heap is corrupted (not necessarily by the calling code in the stack trace, the caller is just the "messenger" who finds the problem.

Heap corruption is hard to debug, because it can be introduced from any place. Are you able to share any of your code with us?

joysfera, 2017-06-27:
The code looks as follows (very simplified):
```
xTaskCreate(&webserver, "webserver", 8192, NULL, 5, NULL);
const char *err = NULL;
static void webserver(void *pvParameter)
{
    struct mg_mgr mgr;
    mg_mgr_init(&mgr, NULL);
    struct mg_bind_opts opts;
    memset(&opts, 0, sizeof(opts));
    opts.error_string = &err;
    struct mg_connection *c = mg_bind_opt(&mgr, "8000", nc_handler, opts);
    for (;;)
    {
        mg_mgr_poll(&mgr, 1000);
    }
}

uint8_t buf[512];
static void nc_handler(struct mg_connection *nc, int ev, void *p) {
  switch (ev) {
    case MG_EV_ACCEPT:
    case MG_EV_CONNECT:
      break;

    case MG_EV_CLOSE:
      break;

    case MG_EV_RECV:
        int len = UART_send_receive((uint8_t *)nc->recv_mbuf.buf, nc->recv_mbuf.len, buf, sizeof(buf));
        mg_send(nc, buf, len);
      break;

    default:
      break;
  }
}

static int UART_send_receive(uint8_t buf[], uint8_t len, uint8_t *outbuf, size_t outlen)
{
    tx_no_echo(buf, len); // calling uart_write_bytes() and discarding local echo

    vTaskDelay(100 / portTICK_RATE_MS);

    int l = uart_read_bytes(uart_num, outbuf, outlen, 20 / portTICK_RATE_MS);
    return l;
}
```

However, I have just updated the mongoose library from mongoose-master (April 2017) to latest mongoose-dev because I read in their git log that on May 27 they changed the way they call lwip from threads.

It's no longer crashing like before so I'd say this issue could be closed. It doesn't work properly yet but it can be a timing or other issue when dealing with UART data so I'm going to debug that now. The original heap corruption seems to be gone and that's great. Thank you for your help.

projectgus, 2017-06-27:
OK, thanks for the update. Please reopen if you think the bug isn't fixed (and may be caused by IDF.)

closed 2017-06-27 by projectgus