espressif/esp-idf #736: Sniffer mode: wifi_pkt_rx_ctrl_t not showing 11n traffic?

anoncontrib, 2017-06-27:
https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_wifi_types.h

```
typedef struct {
    signed rssi:8;            /**< signal intensity of packet */
    unsigned rate:5;          /**< data rate */
    unsigned :1;              /**< reserve */
    unsigned sig_mode:2;      /**< 0:is not 11n packet; 1:is 11n packet */
    unsigned :16;             /**< reserve */
    unsigned mcs:7;           /**< if is 11n packet, shows the modulation(range from 0 to 76) */
    unsigned cwb:1;           /**< if is 11n packet, shows if is HT40 packet or not */
    unsigned :16;             /**< reserve */
    unsigned smoothing:1;     /**< reserve */
    unsigned not_sounding:1;  /**< reserve */
    unsigned :1;              /**< reserve */
    unsigned aggregation:1;   /**< Aggregation */
    unsigned stbc:2;          /**< STBC */
    unsigned fec_coding:1;    /**< if is 11n packet, shows if is LDPC packet or not */
    unsigned sgi:1;           /**< SGI */
    unsigned noise_floor:8;   /**< noise floor */
    unsigned ampdu_cnt:8;     /**< ampdu cnt */
    unsigned channel:4;       /**< which channel this packet in */
    unsigned :12;             /**< reserve */
    unsigned timestamp:32;    /**< timestamp */
    unsigned :32;             /**< reserve */
    unsigned :32;             /**< reserve */
    unsigned sig_len:12;      /**< It is really lenth of packet */
    unsigned :12;             /**< reserve */
    unsigned rx_state:8;      /**< rx state */
} wifi_pkt_rx_ctrl_t;

```

I have set the wifi card into sniffer mode but I am not getting any 11n packets? and hence none of the 11n fields are getting populated. They are always 0. Also, the "rate" field seems to be only the values "11" or "0" for me. Only the packet length, RSSI and Channel are being populated for every packet. 

I am not sure what do I have to do to capture 11n packets? and get the rest of the fields populated? Does any special mode have to be set?

anoncontrib, 2017-07-02:
Friends, 

Any update on this? I am developing a product and this sadly seems to be a deal breaker. Can any of the folks working for the company please respond? We bought these chips because the manufacturer explicitly mentions "sniffer" as a mode for Wi-Fi on their website.

negativekelvin, 2017-07-02:
Maybe @TimXia or @liuzfesp can help

liuzfesp, 2017-07-03:
ＨＩ ＠anoncontrib, which kind of type the sniffer captured? WIFI_PKT_DATA? or WIFI_PKT_MISC?

anoncontrib, 2017-07-03:
Its capturing WIFI_PKT_DATA  and WIFI_PKT_MGMT.  

I am not sure what we get when its is WIFI_PKT_MISC?  seems almost like an error condition to me as the header file says the payload is NULL? 

Also in older versions there was a WIFI_PKT_CTRL as well? but has been removed now?

liuzfesp, 2017-07-03:
Yes, @anoncontrib WIFI_PKT_CTRL is removed because we actually don't filter the ctrl packet (such as CTS, RTS, ACK etc) to CPU and the ctrl packet is dropped in the hardware according our current design.
As the issue your referred, could you let me know the IDF commit ID please? Then I can check it...

anoncontrib, 2017-07-03:
This is the line:

https://github.com/espressif/esp-idf/blob/99d5959d8a319c26b5b907d26a4a5993a2775594/components/esp32/include/esp_wifi_types.h#L251   

It says if the type is  WIFI_PKT_MISC  then the payload is NULL.  So what is this packet then? 

Also, I am getting packets with WIFI_PKT_DATA & WIFI_PKT_MGMT  but it is never 11n as sig_mode is always 0.  Why is this happening?

anoncontrib, 2017-07-03:
Sorry to bug you so much but could you please let me know?  I internally championed the ESP32 in our company and I'll be in pretty hot soup if I don't get to a conclusion by this week :(

liuzfesp, 2017-07-04:
Hi @anoncontrib sorry for my late response, just a little busying....
I just test the sniffer with my device, the result:
sig_mode=0 rate=0 mcs=0
**sig_mode=1 rate=11 mcs=7   =====> sig_mode 1 means 11N packets, the rate is MCS 7**
sig_mode=1 rate=11 mcs=0
sig_mode=0 rate=0 mcs=0
sig_mode=0 rate=0 mcs=0
sig_mode=0 rate=0 mcs=0
sig_mode=0 rate=11 mcs=0
sig_mode=1 rate=11 mcs=7
sig_mode=1 rate=11 mcs=7
sig_mode=0 rate=0 mcs=0
sig_mode=0 rate=6 mcs=0
sig_mode=0 rate=6 mcs=0
sig_mode=0 rate=0 mcs=0

anoncontrib, 2017-07-04:
@liuzfesp Thanks! Really appreciate the help. 

Can you please share the code if you don't mind. I seem to be doing everything right but still not seeing any sig_mode = 1 packets. I just want to make sure I have not left out anything. 

Please looking forward to your response.

liuzfesp, 2017-07-04:
Hi @anoncontrib my promiscuous callback function `void ssc_sniffer_cb(void *buf, wifi_promiscuous_pkt_type_t type)
{
        wifi_pkt_rx_ctrl_t *ctl = (wifi_pkt_rx_ctrl_t*)but;
        switch (type) {
        case WIFI_PKT_MGMT:
                g_sniffer_cnt.rx_mgmt++;
                break;
        case WIFI_PKT_DATA:
                //printf the debug info here
                g_sniffer_cnt.rx_data++;
                break;
        case WIFI_PKT_MISC:
                g_sniffer_cnt.rx_misc++;
                break;
        }
        g_sniffer_cnt.rx_total++;
        return;
}
` Before we can get the 11N packet, we need to make sure there is 11N packets in the air...

FayeY, 2017-07-24:
Hi @anoncontrib , did the above suggestion help you resolve the issue?

closed 2017-08-01 by FayeY

malaimoo, 2017-12-14:
@liuzfesp Hello, Can idf add the WIFI_PKT_CTRL frame? I just need this in my product.