espressif/esp-idf #737: Has any one successfully tested the example of gatt_security_server?

wolfking0815, 2017-06-27:
Hi,
Has anyone successfully tested these codes? because I have tried without any success. Especially, how to enter the pinkey notified? For example, if it was "123456", should I enter "123456" in the peer device? Any advice is helpful. Thanks in advance.

Best regards,
kaikai

Yulong-espressif, 2017-06-28:
@wolfking0815 Yes, when you try this example, you should open the log, and the log will show the pinkey value, then you should enter this key in the peer device. And can you update your SDK? We change the way to use this example, there will not need to anter the pinkey for this example in the new SDK, we also fixed some BLE security connection issues in the new SDK. Thanks.

wolfking0815, 2017-06-28:
Hi Yulong,
Yeah, I have updated my SDK to the commit 574d0cf which is pushed by you. And used the example codes in the IDF. However, I can not successfully bond my Android 5/7 or iOS devices to ESP32, whether the IO_CAP is NONE or OUT. Could you make more explanations? Thanks in advance.

Best regards,
Kaikai

wolfking0815, 2017-06-28:
Hi Yulong, 
After tried with the latest SDK and the example codes (IO_CAP_OUT), with commit 6fa4712, the log of ESP32 said that the authentication status is "success", however, my Android 5 phone still said that ERROR PIN CODE. The log is as follows:

[0;31mE (811552) BT: io_cap = 4[0m
[0;31mE (811552) BT: rcvd auth_req: 0x05, io_cap: 0                         loc_oob_flag: 0 loc_enc_size: 16,local_i_key: 0x03, local_r_key: 0x07
[0m
[0;31mE (811562) BT: set auth_req: 0x05, local_i_key: 0x03, local_r_key: 0x07
[0m
[0;32mI (811572) SEC_GATTS_DEMO: remote BD_ADDR: f880c9756153[0m
[0;32mI (811572) SEC_GATTS_DEMO: address type = 0[0m
[0;32mI (811572) SEC_GATTS_DEMO: pair status = success[0m
[0;32mI (811582) SEC_GATTS_DEMO: pair status = 0[0m
[0;32mI (811582) SEC_GATTS_DEMO: confirm to the peer device[0m
[0;32mI (811592) SEC_GATTS_DEMO: remote BD_ADDR: 011000000000[0m
[0;32mI (811602) SEC_GATTS_DEMO: address type = 0[0m
[0;32mI (811602) SEC_GATTS_DEMO: pair status = success[0m
[0;32mI (811612) SEC_GATTS_DEMO: pair status = 0[0m
[0;32mI (811612) SEC_GATTS_DEMO: confirm to the peer device[0m
[0;31mE (811692) SEC_GATTS_DEMO: The passkey Notify number:362317[0m
[0;31mE (812102) BT: gatt_disc_cmpl_cback() - Register for service changed indication failure[0m
[0;32mI (820152) SEC_GATTS_DEMO: remote BD_ADDR: f880c9756153[0m
[0;32mI (820152) SEC_GATTS_DEMO: address type = 0[0m
[0;32mI (820152) SEC_GATTS_DEMO: pair status = success[0m
[0;32mI (820152) SEC_GATTS_DEMO: pair status = 0[0m
[0;32mI (820162) SEC_GATTS_DEMO: confirm to the peer device[0m
[0;31mE (820292) BT: btm_sec_clr_temp_auth_service() - no dev CB
[0m
[0;31mE (820292) BT: Device not found
[0m
[0;31mE (820292) BT: btm_sec_disconnected clearing pending flag handle:0 reason:19
[0m
[0;31mE (820292) BT: BTA got unregistered event id 31
[0m
[0;31mE (820302) BT: BTA got unregistered event id 31
[0m
[0;32mI (820312) SEC_GATTS_DEMO: remote BD_ADDR: f880c9756153[0m
[0;32mI (820312) SEC_GATTS_DEMO: address type = 0[0m
[0;32mI (820312) SEC_GATTS_DEMO: pair status = success[0m
[0;32mI (820322) SEC_GATTS_DEMO: pair status = 0[0m
[0;32mI (820322) SEC_GATTS_DEMO: confirm to the peer device[0m
[0;32mI (820332) SEC_GATTS_DEMO: remote BD_ADDR: 001000000000[0m
[0;32mI (820342) SEC_GATTS_DEMO: address type = 0[0m
[0;32mI (820342) SEC_GATTS_DEMO: pair status = success[0m
[0;32mI (820352) SEC_GATTS_DEMO: pair status = 0[0m
[0;32mI (820352) SEC_GATTS_DEMO: confirm to the peer device[0m

So my question is: 1) is the notify pinkey exactly the same I should put into the peer device? 2) when the AUTH_CMPL event caught by ESP32, should I send out another reply event to the peer device? Thanks in advance. 

Some suggestions: please make the comments of example codes be the same as the comments of header file, because the mismatch of IO_CAP in the comments introduces some misunderstanding. Thanks.

Best regards,
kaikai

wolfking0815, 2017-06-29:
Hi, Yulong,
This issue has been solved, and you can close it now. Thanks for your help.

Best regards,
kaikai

closed 2017-06-30 by FayeY

NicolasNS, 2017-07-31:
I have the same issue with the latest commit and my Iphone 5c. Is it working for some one ?