espressif/esp-idf #739: make size, make size-components, make size-files doesn't seem to work with tip.

madscientist42, 2017-06-28:
Captured result from an attempt to use...

> FEarl@ENG110 MINGW32 /d/PetProjects/esp32/esp-idf/examples/bluetooth/ble_adv
> $ make size
> Traceback (most recent call last):
>   File "D:/PetProjects/esp32/esp-idf/tools/idf_size.py", line 229, in <module>
>     main()
>   File "D:/PetProjects/esp32/esp-idf/tools/idf_size.py", line 153, in main
>     memory_config, sections = load_map_data(args.map_file)
>   File "D:/PetProjects/esp32/esp-idf/tools/idf_size.py", line 47, in load_map_data
>     sections  = load_sections(map_file)
>   File "D:/PetProjects/esp32/esp-idf/tools/idf_size.py", line 82, in load_sections
>     scan_to_header(map_file, "Linker script and memory map")
>   File "D:/PetProjects/esp32/esp-idf/tools/idf_size.py", line 43, in scan_to_header
>     raise RuntimeError("Didn't find line '%s' in file" % header_line)
> RuntimeError: Didn't find line 'Linker script and memory map' in file
> make: *** [/d/PetProjects/esp32/esp-idf/make/project.mk:395: size] Error 1

Here's a grep for the landmark in the .map file:

> FEarl@ENG110 MINGW32 /d/PetProjects/esp32/esp-idf/examples/bluetooth/ble_adv/build
> $ grep Linker *.map
> Linker script and memory map

For some unknown reason, it seems like Python's clipping off the map file out of the example build - the map file has the landmark section sought for the processor code.  I'm looking for causes right now.  Will update the issue and indicate in notes as soon as I see what gives.

madscientist42, 2017-06-28:
Looks like the Regex for the memory configuration section processor doesn't work as expected.  It skims to that header for the section, can't find any lines because it's not a valid regex to parse the line, and since the code always has "result" as empty...it inhales the rest of the file.  Oops.

madscientist42, 2017-06-28:
I'll tinker a bit with it and see if I can't help out with a fix here.

projectgus, 2017-06-28:
Thanks @madscientist42 . Maybe this is something line ending related?

projectgus, 2017-06-29:
Thanks again for reporting this, and sorry for the hassle. It was exactly like you said, and is fixed now (plus a sanity check to get a better error if scan_for_header() inhales the entire file.)

Fix will be merged shortly, but the change is here if you're interested:
```
-    RE_MEMORY_SECTION = r"(?P<name>[^ ]+) +0x(?P<origin>[\da-f]+) 0x(?P<length>[\da-f]+)"
+    RE_MEMORY_SECTION = r"(?P<name>[^ ]+) +0x(?P<origin>[\da-f]+) +0x(?P<length>[\da-f]+)"
```

madscientist42, 2017-06-29:
Thanks- didn't have time yesterday like I'd thought I would with the project they had me on at work combined with getting ready to a SF convention that many of the major SF authors use as a get-to-gather on this weekend.  Greatly appreciated and I'll apply the patch locally because I'm trying to find an answer for the associated project that uses an ESP32.

closed 2017-07-13 by igrr