espressif/esp-idf #740: ESP32 with sd card shield and sd_card_example_main.c

Staubgeborener, 2017-06-28:
I try to get the sd_card_example_main.c working. But it fails. I tried it with a solid sd card adapter and also with a sd card shield.
To break down everything: How is the correct circuit for the esp32 and a sd card shield to get this script working? I have this one: https://d3s5r33r268y59.cloudfront.net/27271/products/thumbs/2014-05-16T16:00:29.024Z-237.jpg.2560x2560_q85.jpg
The pinout is: GND, VCC, MOSO, MOSI, SCK, CS

If i try my diy solid sd card adapter with the circuit in the readme file, i'll get
```
E (1170) sdmmc_req: handle_idle_state_events unhandled: 00000004 00000000
E (1170) sdmmc_cmd: sdmmc_card_init: send_op_cond (1) returned 0x107
E (1180) example: Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place.
```

negativekelvin, 2017-06-28:
IDF does not support spi mode yet, only sdio.  So you could try 1-bit sdio mode or try the arduino example http://www.iotsharing.com/2017/05/how-to-use-arduino-esp32-to-store-data-to-sdcard.html

Staubgeborener, 2017-06-29:
The arduino example works fine, but i want to use the esp-sdk / esp-idf instead.  Now i have a solid sd-card adapter connected to my esp32 with the circuit posted in the readme file ( https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#hardware )
This should work with esp-sdk, eh? Tried this with 4-line SD mode and also 1-line SD mode (by uncommenting line 29 of the sd_card_example_main.c code).  I'll get

```
E (1208) sdmmc_cmd: sdmmc_card_init: send_if_cond (1) returned 0x109
E (1208) example: Failed to initialize the card (265). Make sure SD card lines have pull-up resistors in place.
```

negativekelvin, 2017-06-29:
Post a photo/diagram of your setup.

Staubgeborener, 2017-06-29:
Everything works now. I implemented the Arduino component in my esp-idf, so i can use the librarys
```
#include "SPI.h"
#include "SD.h"
```
These operates fine.

FayeY, 2017-07-02:
Hi @Staubgeborener , is this problem solved now?

Staubgeborener, 2017-07-31:
Hello @FayeY. Sorry for the late response. Yes, it is (with SPI, but not [this](https://github.com/espressif/esp-idf/issues/817) case.)

closed 2017-08-09 by FayeY

TurkeyLegsWilliam, 2018-06-22:
hey @Staubgeborener, how did you manage to get it working? I'm trying to figure out what's wrong with how I'm doing it

Staubgeborener, 2018-06-24:
@TurkeyLegsWilliam Oh man, sorry, but its now over a year ago. I saw in my previous posts, that i used some librarys from the arduino ide ``#include "SPI.h"`` and ``#include "SD.h"``. It seems, that i just copypasted those librarys into my esp-idf folder to use them. I just looked into my old code, but there is nothing like that. I used these libraries:
``#include "driver/sdmmc_host.h"
#include "driver/sdspi_host.h"
#include "sdmmc_cmd.h"``
together with the sd card [example](https://github.com/espressif/esp-idf/blob/master/examples/storage/sd_card/main/sd_card_example_main.c). Sorry, that I can not say anymore, but it's too long ago. Nowadays i use [esp32 boards with included sd card support](https://de.aliexpress.com/item/DSTIKE-WiFi-Paket-Monitor-V3-Preflashed-D-duino-32-SD-ESP32-Wrover-OLED-WiFi-BLE-IOT/32847023236.html), so i don't have that kind of problems anymore.