espressif/esp-idf #742: SYSTEM_EVENT_AP_PROBEREQRECVED Not Working

alexgrauer, 2017-06-28:
Hi all,

Im trying to work with the AP_PROBEREQRECVED event but its not working. All of the other are ok, but it seems as if this one is not being triggered.
I already asked some other people to try it and they had the same result.

Any clue?
Im using the ESP-WROOM-32 with the Arduino IDE.

Thanks very much

liuzfesp, 2017-07-04:
Hi @alexgrauer currently this event has been forbidden in latest IDF because we will have too much SYSTEM_EVENT_AP_PROBEREQRECVED if we enable it, we are going to optimize this event in the release next to 2.1.

alexgrauer, 2017-07-04:
Thanks @liuzfesp for the answer. I´ll wait for release 2.1 then!

liuzfesp, 2018-01-15:
@alexgrauer I think this feature will not be added into v3.0 according to the release plan.

closed 2018-06-22 by FayeY