espressif/esp-idf #743: Secure boot issue

rillhu, 2017-06-29:
I tried the secure boot feature following the steps [ http://esp-idf.readthedocs.io/en/latest/security/secure-boot.html#re-flashable-software-bootloader](url)

I generate the pem useing 
`python espsecure.py generate_signing_key secure_boot_signing_key.pem`

Then I run `make bootloader `and using the second set command as below to flash the bin

`python /d/ESP32/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM4 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 /d/ESP32/test_secure_project/build/bootloader/bootloader-reflash-digest.bin`

Then I run `make flash` and code runs as expect.

However, when I try to apply the pem to another project with the same above steps, issue happens. ESP32 always print out below info:

> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0x00
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0008,len:8
> load:0x3fff0010,len:6492
> ho 0 tail 12 room 4
> load:0x40078000,len:17184
> load:0x40080000,len:252
> secure boot check fail
> ets_main.c 371 

Could expert help me here? What's wrong with my operation?

rillhu, 2017-07-20:
May I get some feedback? 👍

projectgus, 2017-08-21:
Hi @rillhu ,

Sorry it took so long for someone to get back to you.

When you say:
> However, when I try to apply the pem to another project with the same above steps, issue happens

Can you be more specific, please? What is different about this other project? Are you using the same ESP32 chip this time, or a different chip? Have you written the secure boot key to the efuse of the new chip with `espefuse.py`, same as the old chip?

projectgus, 2017-09-07:
Hi @rillhu ,

I'm going to close this for now, but please reopen it if you reply with more information or specific questions.

Angus

closed 2017-09-07 by projectgus

chegewara, 2017-12-29:
@projectgus Hi, im trying to erase esp32 with flash download tool and im getting some errors. Until now ive got issue to flash/erase this chip at all, now during errase some efuse flags got burned but im getting this error:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
```

projectgus, 2018-01-01:
> flash read err, 1000

This messages implies that the software bootloader (which should be flashed at 0x1000) is missing or corrupt. Double check the .bin files you're flashing are correct for each offset in the flash (bootloader.bin should be flashed at 0x1000). If you can post a screenshot of the Windows GUI Flasher tool, this may help.

> test offset :  4096 0x1000

4096 is too short for a bootloader, so maybe this isthe partition table file being flashed to 0x1000? If so, this explains the error.

> How do i know that efuse flags got burned? Because those 2 messages:
>
>   first erase tryout:
>   .Burning efuse CONSOLE_DEBUG_DISABLE (Disable ROM BASIC interpreter fallback) 0x0 -> 0x1. This is > an irreversible operation.
>    next erase tryout:
>    Efuse CONSOLE_DEBUG_DISABLE is already burned.

This is not related to these boot problems. ESP-IDF now disables the undocumented "easter egg" ROM BASIC interpreter by default, to avoid accidental resetting into this mode.

chegewara, 2018-01-01:
```
flash read err, 1000

This messages implies that the software bootloader (which should be flashed at 0x1000) is missing or corrupt. Double check the .bin files you're flashing are correct for each offset in the flash (bootloader.bin should be flashed at 0x1000). If you can post a screenshot of the Windows GUI Flasher tool, this may help.
```
I am using exact the same bootloader and partition file to flash 2 chips, one is DevKitC and one is just a esp32 wroom. With DevKitC ive got no issue.

`Burning efuse CONSOLE_DEBUG_DISABLE (Disable ROM BASIC interpreter fallback) 0x0 -> 0x1`
I know what this means, just wanted to show that something got flashed, or at least some operations got performed. In this case it was burning efuse but thats all.

projectgus, 2018-01-01:
I see, I didn't understand your original comments.

IF it's a bare ESP32 WROOM and you're sure you have the correct files at the correct offsets then it probably has power problems. WROOMS are not really designed to be used "bare", although lots of people use them this way. Check you ahve enough low-ESR capacitance on the VCC/GND connections, and a high current rated 3.3V supply with short/thick leads to the WROOM itself.

chegewara, 2018-01-01:
I think there should not be problem with voltage, if USB<-->UART is not providing stable 3.3V then i think so getting it from DevKitC should be enough. I didnt concidered capacitance issues, this may be the problem. 
Just the issue can be read somehow from this message since boot values are different in this 2 cases and this info is not available to us users:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)`

chegewara, 2018-01-01:
@projectgus Also in my first post i sid i am trying to erase chip, not flash it and im getting errors during ERASE.

projectgus, 2018-01-01:
I'm not sure about the error when erasing the chip from the GUI tool. This tool is not a part of esptool.py.

Is it possible the flash chip was actually successfully erased? This would explain the "flash read err, 1000" on boot (a fully erased chip doesn't contain a bootloader so you will get this error.)

Are you flashing new binaries to the chip after you erase it?

If you can explain exactly what you did, in what sequence, and seeing what output, then this would be very helpful.

chegewara, 2018-01-02:
I did try to flash this chip few months ago with esptool.py but with no success. I cant read reliable info about that chip with esptool.py. Now i thought i will try with flash download tool. Still cant flash it. Im affraid i might broke it during soldering or something.

This error code messages are exactly the same with flash download tool and with esptool.py because i can read it from serial port with putty. 

`Is it possible the flash chip was actually successfully erased? This would explain the "flash read err, 1000" on boot (a fully erased chip doesn't contain a bootloader so you will get this error.)`

You are probably right and chip is erased, but i cant flash it no matter what im doing. I will just assume it is broken and stop waisting yours and mine time. Thanks for help.

projectgus, 2018-01-02:
>  I will just assume it is broken and stop waisting yours and mine time. Thanks for help.

In all likelihood this is a power/capacitance issue. Given how cheap ESP32 dev boards are becoming, buying another one instead of using the bare module is probably the easiest option.

chegewara, 2018-01-02:
Ive got 2 DevKitC already, and its not as cheap to me. My earnings is less than $3 per hour.