espressif/esp-idf #745: Bluetooth Problem with  ESP-IDF Updating to Release v2.1

figueiredoAF, 2017-06-29:
Hello,

I was using ESP-IDF 2.0 in my project and everything was going well. I was using bluetooth and wifi to command my device locally and remotely, the functionality was in use two months ago.

I decided to upgrade my ESP-IDF to 2.1 release. So the application crashed, and now I'm getting the following error:

E (4550) BT: Each service table is only a service or second service.
Guru Meditation Error of type LoadProhibited occurred in core 0. Exception was not processed.

Anyone know what's going on?

Regards,
Alex Fernandes.

negativekelvin, 2017-06-29:
Did you read the breaking changes section

https://github.com/espressif/esp-idf/releases/tag/v2.1-rc1

heyinling, 2017-06-30:
@figueiredoAF the error log `E (4550) BT: Each service table is only a service or second service.` means we can only define one primary service in one table.

And I tried to create table with 2 primary services. I got error reply without exception.

plz check @negativekelvin 's suggestion. If still doesn't work please post your application code.

figueiredoAF, 2017-06-30:
Hello,

Sorry, thank you, I was creating an attribute table with more than one primary service. At the ESP-IDF 2.0 release I could do it.

@negativekelvin, I checked the ESP-IDF Pre-release v2.1-rc and have no mention of this incompatibility. I suggest you to mention this change.

Greetings,

closed 2017-06-30 by figueiredoAF