espressif/esp-idf #748: Interrupts

chegewara, 2017-06-30:
I have modified a little bit example gpio to generate interrupts from buttons click and not from gpio state change, also ive changed intr_type from ANYEDGE to POS/NEGEDGE and this is what ive found:
- 1 click button sometimes generates interrupt on both edges, 

Additionally, but maybe its problem with my program sometimes 1 click generates few interrupts.

Spritetm, 2017-06-30:
Do you debounce your button in hardware? If not, what you're seeing is contact bounce.

chegewara, 2017-06-30:
What do you mean?

Spritetm, 2017-06-30:
https://www.google.com.hk/?gws_rd=cr,ssl#safe=strict&q=contact+bounce

chegewara, 2017-06-30:
Then it not software but hardware issue?

Spritetm, 2017-06-30:
The issue originates in your hardware, yes.

chegewara, 2017-06-30:
One more question,  button debouncingdoes not explain why with POS/NEGEDGE intr_type sometimes are generated interrupts on both edges?

Spritetm, 2017-06-30:
How do you make sure you're not actually feeding positive and negative edges into your GPIO? What is your signal source?

chegewara, 2017-06-30:
Im not cause its an ordinary cheap button. The point is i want interrupt to be triggered only on one edge, thats why is intr_type parameter for as i understand. But ocasionally vallue read from pin that triggered interrupt is not the one im expecting. 
As i understand, with pin pullup and intr_type=POSEDGE, interrupt could be triggered when pin state is changing from low to high, means pin level=1. Or am i wrong.

Spritetm, 2017-06-30:
Again, contact bounce. When you close or open a switch, the contact makes (positive edge) and breaks (negative edge) a bunch of times.

chegewara, 2017-06-30:
Again thanks, and sorry for bothering.

closed 2017-06-30 by chegewara