espressif/esp-idf #749: BLE: Calling esp_ble_gatts_send_indicate() always causes a confirmation to be returned.

nkolban, 2017-06-30:
In my testing when I invoked esp_ble_gatts_send_indicate() to submit a notification and specify the last parameter as `false` ... which should mean `bool need_confirm` I seem to be finding that I then receive the GATT event `ESP_GATTS_CONF_EVT` which I take to mean that a confirmation was received.  This surprises me as I had assumed that since I was not asking for a confirmation, one would not arrive.

I don't yet know whether this is a problem with the GATT controller (my Android phone) or my GATT peripheral (ESP32).  It could be that the ESP32 BLE stack is not sending the correct bit/flag in the notification or it could mean that the GATT controller (Android) is not honoring it correctly.  Either way, I wanted to raise the issue in case it is of value for a test / examination.

Yulong-espressif, 2017-06-30:
@nkolban Thanks for your question, this is normal state, our purpose of the design for gatts module is that each call of the gatts API should have a callback. So when you used the esp_ble_gatts_send_indicate API and set the need_confirm to false, then will callback by the bluedroid stack directly when you has success or fail to sent the message. If you set the need_confirm to ture, then will callback when the peer device has response the command or the timeout detected. Thanks. Hope this answer can help you.

nkolban, 2017-06-30:
Ahhh ... thank you my friend.  I understand now.  Again, thank you for the quick response.

closed 2017-06-30 by nkolban