espressif/esp-idf #750: System halts when Flashing Core Dump

APokorny, 2017-06-30:
I try to enable flashing core dumps to a core dump partition. As test case I wrote a small arduino application that writes to unmapped memory. During the panic handler the system halts in spi flash routines. It halts after the while loop:
https://github.com/espressif/esp-idf/blob/master/components/spi_flash/cache_utils.c#L258
during the first invocation of the functions (the test application causes the memory access error on core 1). 

If I understand that correctly this code deactivates the cache on core 0, and then after that calls it again to deactivate the cache on core 1. It seems that during the panic handler any attempt to set that register to zero for either of the cores halts the panic handler execution.

Removing that i.e. in 
https://github.com/espressif/esp-idf/blob/master/components/spi_flash/cache_utils.c#L168
seems to produce a core dump in the partition - as in I found the magic and and begin markers on the next boot inside the partition.

igrr, 2017-06-30:
Please check whether IDF version you are using has this recent change: https://github.com/espressif/esp-idf/commit/21f02a6955cdf6cd37c6c4018e1ed683a95a9abb
It fixes the bug with DPORT access inside panic handler.

APokorny, 2017-06-30:
We currently pick the version used with the esp32 Arduino SDK 1e0710f1.
I switched over to v2.1 - which contains the patch. That solves the problem.

closed 2017-06-30 by APokorny