espressif/esp-idf #751: ble advertising service data

youyudedasan, 2017-06-30:
hello all!
i tested the project gatt_server, BLE advertised very well! I also add manufacturer data, and it worked well. Here is advertisin data struct.
`static uint8_t test_service_uuid128[32] = {
    /* LSB <--------------------------------------------------------------------------------> MSB */
    //first uuid, 16bit, [12],[13] is the value
    0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xAB, 0xCD, 0x00, 0x00,
    //second uuid, 32bit, [12], [13], [14], [15] is the value
    0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xAB, 0xCD, 0xAB, 0xCD,
};

//static uint8_t test_manufacturer[TEST_MANUFACTURER_DATA_LEN] =  {0x12, 0x23, 0x45, 0x56};
static esp_ble_adv_data_t test_adv_data = {
    .set_scan_rsp = false,
    .include_name = true,
    .include_txpower = true,
    .min_interval = 0x20,
    .max_interval = 0x40,
    .appearance = 0x00,
    .manufacturer_len = TEST_MANUFACTURER_DATA_LEN,
    .p_manufacturer_data = &test_manufacturer[0],
    .service_data_len = 0,
    .p_service_data = NULL,
    .service_uuid_len = 32,
    .p_service_uuid = test_service_uuid128,
    .flag = (ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT),
};`
Then i tried to add servise data, but i could not find servise data with a BLE dongle.Only service UUID was showed.Is there something that I missed? Here is new advertisin data struct.
`static uint8_t test_service_uuid128[32] = {
    /* LSB <--------------------------------------------------------------------------------> MSB */
    //first uuid, 16bit, [12],[13] is the value
    0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xAB, 0xCD, 0x00, 0x00,
    //second uuid, 32bit, [12], [13], [14], [15] is the value
    0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xAB, 0xCD, 0xAB, 0xCD,
};

static uint8_t test_manufacturer[TEST_MANUFACTURER_DATA_LEN] =  {'A','B','C','D'};
static uint8_t test_servise[TEST_SERVISE_DATA_LEN] = {'s','e','r','v','i','s','e'};
static esp_ble_adv_data_t test_adv_data = {
    .set_scan_rsp = false,
    .include_name = true,
    .include_txpower = true,
    .min_interval = 0x20,
    .max_interval = 0x40,
    .appearance = 0x00,
    .manufacturer_len = 0,//TEST_MANUFACTURER_DATA_LEN,
    .p_manufacturer_data = NULL,//&test_manufacturer[0],
    .service_data_len = TEST_SERVISE_DATA_LEN,
    .p_service_data = &test_servise[0],
    .service_uuid_len = 32,
    .p_service_uuid = test_service_uuid128,
    .flag = (ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT),
};`

heyinling, 2017-06-30:
plz check if your service data is too long, exceed 31 bytes. If too long the adv API will cut off some types of data.

and I prefer to use set raw data API to set adv data, it gives you full control of the data you want to advertise.

youyudedasan, 2017-06-30:
@heyinling thanks for reply!
as you said, I check raw data API. I find the answer to my qustion

FayeY, 2017-07-02:
Hi @youyudedasan , is this problem solved now? Can we close this issue?

closed 2017-07-03 by youyudedasan