espressif/esp-idf #752: Wifi+BLE error while scanning wifi

Emaxinvader, 2017-06-30:
If I scan for wifi network when BLE GATT server and Wifi are connected, I get sometimes this error "E (33262) wifi: pm.c 953" and all connections are closed.
What does it mean?

I'm using the defult configuration of BLE examples with esp-idf v2.1rc1.

FRDavid, 2017-07-03:
+1

In my case I reached the error in the following way:
- Turn on BLE to advertise
- Turn on WiFi
- Scan networks, connect to a network, disconnect from network, scan networks

After a couple of scans the message above mentioned appears and seems wifi task is in a bucle or similar since next message is shown:

Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE

BR,
David

Roody75, 2017-09-12:
From time to time (about 1 in 5 tries) the same issue occures during WPS scan as well (@HEAD revision), mostly if I start WPS in ESP before I start it in the AP:

```
E (184476) wifi: pm.c 953
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: wifi
```

jack0c, 2018-03-22:
@Roody75 @Emaxinvader @FRDavid It should be a bug, can you provide commit ID.

TimXia, 2018-03-29:
@Roody75 @Emaxinvader @FRDavid The bug has been fixed in IDFv2.1. Please have a try.

closed 2018-03-30 by FayeY