espressif/esp-idf #753: BLE power consumption

HardwireIO, 2017-06-30:
We made some tests on power consumption, because we're interested to use device with a battery.
Even though BLE was born for low power applications, we're really in trouble to reach a real low power situation.

Here what we got:
- CPU @ 80Mhz, BLE on and 20ms of advertising --> 115mA of power consumption
- CPU @ 80Mhz, BLE on and 10sec of advertising --> 110mA of power consumption
- CPU @ 80Mhz, BLE off --> 30mA of power consumption

It looks like BLE is not really low power. Is there any way to reduce power consumption of BLE peripheral?

jmarcelino, 2017-07-04:
There was some great discussion about this on the forum:

https://www.esp32.com/viewtopic.php?f=12&t=1198

closed 2017-11-14 by FayeY

CT3, 2018-03-01:
Does anyone know if the ESP-IDF Pre-release 3.0-rc1 has the problem fixed?