espressif/esp-idf #755: BLE GATTC graceful disconnect not works

SiarheiVolkau, 2017-07-01:
When i try to disconnect connected Peripheral(s) no real disconnection happened.
Was tried esp_ble_gattc_close() and esp_ble_gattc_app_unregister() methods, IDF is [9b955f4](https://github.com/espressif/esp-idf/commit/9b955f4c9f1b32652ea165d3e4cdaad01bba170e).
Peripheral stays connected in the background unlimited time.

Thank you.
Siarhei

heyinling, 2017-07-03:
@SiarheiVolkau `esp_ble_gattc_close` will do disconnect if there's no other gattc/gatts app using this connection. plz check if you have created multiple gattc/gatts apps.

and there's a GAP API `esp_ble_gap_disconnect` which will do disconnect regardless of gatt apps.

SiarheiVolkau, 2017-07-03:
@heyinling i create new app for each connection, so there's no one using this connection anymore. Will check esp_ble_gap_disconnect, thx.

heyinling, 2017-07-03:
@SiarheiVolkau if the app is using this connection depends on if it receives the connect event. the stack will dispatch to all open apps. Need to check that on your application code.

SiarheiVolkau, 2017-07-03:
@heyinling when i try to use `esp_ble_gap_disconnect()` i got this error:
`BT: L2CA_RemoveFixedChnl()  CID: 0x0004  BDA: 00158300b985 not connected`
and no one event happens in gattc or gap.
Device stays connected.

heyinling, 2017-07-03:
are you try to connect to a slave which send security request and you did not do pairing with it?

If you have 2 ESP32 you may try to connect to example gatt_server and gatt_security_server. If it only happen on gatt_security_server, that's the same issue I met.

SiarheiVolkau, 2017-07-03:
@heyinling  No, my devices not use security. And yes i have 2 esp's will try to do as you said.

SiarheiVolkau, 2017-07-03:
@heyinling Yes you are right, disconnection works in gattc_demo code. Thanks.

SiarheiVolkau, 2017-07-06:
I found the problem, disconnection not work when `CONFIG_CLASSIC_BT_ENABLED=y` but not works in other case.
I suspect this is a bug somewhere in the code, because should work in BLE only mode too.

closed 2017-07-06 by SiarheiVolkau