espressif/esp-idf #756: network disconnects after some time and then reconnects ..

shirish47, 2017-07-01:
I am trying to develope MQTT based application. I am using https://github.com/tuanpmt/esp32-mqtt example code and testing with nodeRed. It works fine after multiple reboot by itself but then it disconnects after some time and reconnects ..is it the issue of IDF?? here is the issue I posted on MQTT library git. https://github.com/tuanpmt/esp32-mqtt/issues/11 topic name is different then the problem so please forgive me.

Also I am storing wifi in flash and also I erased a flash completely for making sure flash wasn't corrupted.

shirish47, 2017-08-04:
```
[APP] Publish data[2/2 bytes]
[APP] DATA: hil
o??
u there ??‚˘CßârüpZ 
f task=mqt a=0x3ffafec8 h=1a2b3c4d t=0a3f3f6f
abort() was called at PC 0x40083779 on core 0
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40009203:0x3ffc4bf0 0x40082ecc:0x3ffc4c10 0x400833d2:0x3ffc4c30 0x400864a0:0x3ffc4c50 0x400820d4:0x3ffc4c70 0x4000beca:0x3ffc4c90 0x40111840:0x3ffc4cb0 0x401134a8:0x3ffc4cd0 0x401135b5:0x3ffc4d20 0x401136a5:0x3ffc4d40

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

FayeY, 2017-11-21:
Hi, you can refer to our example for MQTT usage https://github.com/espressif/esp-idf/tree/master/examples/protocols/aws_iot.

closed 2017-11-21 by FayeY