espressif/esp-idf #758: gatt_server_service_table example does not restart advertising after client disconnect

RobinCutshaw, 2017-07-01:
When a gatt client connects and then disconnects from the gatt server service table example, advertising is not restarted.

Something like this should work:
`
>         case ESP_GATTS_DISCONNECT_EVT:
>             // start advertising again after disconnect
>             esp_ble_gap_start_advertising(&heart_rate_adv_params);
>             break;
`

Yulong-espressif, 2017-08-09:
@RobinCutshaw after the gatts receive the connection event from the gatt client, it will stop the adverting, so when disconnect, the application should sent the advertising again. Help this answer can help you.

RobinCutshaw, 2017-08-09:
I don't think you understand the problem.  The issue is that the example does NOT restart advertising once the connection is dropped.  I've submitted a pull request with a fix.

LEMAOR, 2017-08-17:
i have it like this:
case ESP_GATTS_DISCONNECT_EVT:
            ble_set_adv_data();

Yulong-espressif, 2017-10-09:
@RobinCutshaw Thank you very much.
@LEMAOR it will not restart advertising if you write it like you show:
case ESP_GATTS_DISCONNECT_EVT:
esp_ble_gap_config_adv_data(&heart_rate_adv_config);

you should change it like the following:
case ESP_GATTS_DISCONNECT_EVT:
        esp_ble_gap_start_advertising(&heart_rate_adv_params);
        break;

closed 2017-10-16 by FayeY