espressif/esp-idf #761: Simple Bluetooth Example

alexgrauer, 2017-07-02:
Hi all,

Im just trying to do a simple Bluetooth connection between a mobile phone and the ESP32 to exchange some basic information.

I wasnt able to find any example. Is Classic Bluetooth support available already?
Any example?

Thanks very much in advance

lucazader, 2017-07-02:
Hi Alex,

BLE should be good enough for what you want. It is aimed at relatively low data rate and/or infrequent data transmission.

Have a look into the BLE standard and what all the different terms mean. ([here is a tutorial from adafruit](https://learn.adafruit.com/introduction-to-bluetooth-low-energy/introduction))

Then the simplest example would be for you to use the gatt_server_service_table example in the idf.
This will set your ESP32 up as a peripheral for your phone to connect.

Hope this helps.

alexgrauer, 2017-07-02:
@lucazader, thanks for you answer.

What i need is to send a variable from the ESP32 to de mobile phone only when is requested by the phone.

Also, i need the phone to be able to send variables to the ESP32 whenever is needed.

Is that possible with gatt?

Sorry but im quite new to this, so i don't know much about bluetooth yet.
Thanks again!

lucazader, 2017-07-02:
Hi @alexgrauer,

Yes this is definitely possible. It is mostly what BLE and GATT were built for.

In this case you would need a gatt service, which has one characteristic. The gatt server runs on the esp32.
When the ESP32 needs to update the value it would change the value in the characteristic.

When the phone wants to know the variable it can request the value by reading the characteristic.
When the phone wants to send the variable it would write the value to the characteristic. This would then trigger a callback on the ESP32 which would let ESP32 do what it wants to with this variable.

Another advanced thing you can do is set up notifications/indications. This would mean that if the ESP32 alters the value in the characteristic, it would automatically send this change to the phone without the phone having to request the value change.

nkolban, 2017-07-02:
A new sample (not by Espressif) is now available as a video tutorial which sounds close to what you might need.... see:

https://www.youtube.com/watch?v=2_vlF_02VXk

Be aware that this uses C++ and also another open source wrapper on top of the underlying ESP-IDF SDK.

closed 2017-07-07 by FayeY

alexgrauer, 2017-07-17:
Hi all, 

I re-open this thread because im having problems to make this work.
Im using the gatt_server_service_table example, but im not being able to change the value of Heart Rate Control Point read.

Im doing the following:

- In the gatts_profile_event_handler i added:
case ESP_GATTS_READ_EVT:
            heart_ctrl_point[1] += 0x01;
            ESP_LOGI(GATTS_TABLE_TAG, "READ EVENT: %d\n",heart_ctrl_point[1]);
break;

- When i read the value using the nRF Connect App, i can see the READ EVENT text in the monitor, and the heart_ctrl_point is actually being incremented.
But in the App i still see 0x00 every time i read it.

Any clue?

Thanks in advance,
Alex

lucazader, 2017-07-17:
Hi @alexgrauer 
1. When you change a characteristic like that, you would need to use `esp_ble_gatts_set_attr_value` to actually get the altered characteristic value to update.

2. As for the ESP_LOGI not showing up. Check that you have your logging verbosity (in the menuconfig) is set to Info, Debug or Verbose. Try also changing it to ESP_LOGE as well.

Hope this helps.

HubbyGitter, 2017-07-17:
@alexgrauer See #796 for a potential memory leak in conjunction with repeated calls of `esp_ble_gatts_set_attr_value`.

alexgrauer, 2017-07-18:
Thanks all for the answers.
I still cant make it work, i still cant change heart_ctrl_point[1]'s value.
Is there any simpler example than the gatt_table_create_demo? I just need to read one characteristic, and be able to modify it from another part of the code.
Any other example with better documentation? Honestly im new to this and i find the code very complicated to understand.
Thanksss!!

nkolban, 2017-07-18:
Alex,
I'm working on some C++ classes to try and make BLE simpler ... this may be of no use to you at all ... but if it is, see the following ... https://esp32.com/viewtopic.php?f=13&t=2330

Because I am still looking for experience on how the classes are being perceived, I'll be delighted to work one on one with you as needed either through the forum or IRC (channel ESP32) to get you going.  Again, if what you are looking for should not include C++ or you want to go against the metal of ESP-IDF, please ignore this comment.

eldadwasserman, 2017-11-07:
Hi
maybe anyone can help me. 
I need to connect the esp32 to iPhone Android pc and Mac via Bluetooth and transmit 115200 bps of data. this bandwidth can't be supported by BLE. 

where can I find example c code for this kind of task ?

lucazader, 2017-11-07:
If you increase the MTU of the connection between the ESP32 and the client, you should be able to get approximately the same data throughput as 115200 baud. I have done some tests and have achieved ~11k bytes throughput.

Have a look at the SPP (serial port profile) examples in the IDF
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/ble_spp_server
That is a good example that has a Tx and an Rx characteristic that provides a nice interface to do serial comms over BLE.

eldadwasserman, 2017-11-07:
> bluetooth/ble_spp_server

thank you very much for your help.
I need to transmit in real time 80 bytes every 16.666 MSEC. exactly every
16.666 MSEC the Esp32 need to transmit this data. the data is time depended
data for time depending analysis calculation in the other side (the
receiver side). as my understanding after reading the BLE protocol it's
impossible to achieve this bandwidth.

please enlight me if I am wrong.

thank again.

Eldad

lucazader, 2017-11-07:
I am not sure if you can have a connection interval of exactly 16.666ms, You can setup a connection interval of 16.25 which is the closest approximation.  However not all platforms that you wish to support will support this connection interval as it is quite fast.

As for the data throughput, If you use a host with BLE4.2 you can increase the MTU of each packet. If you increase this to more than 80 you will be able to send 80 bytes per connection interval.

If you need to send exactly every 16.666 ms and your project and or code cant deal with data sent at any other frequency, then BLE is not for you.
If you can get around this by batching up connections or dealing with the slight offset that the 16.25ms connection interval introduces, then BLE should work fine.

eldadwasserman, 2017-11-07:
thank you very much.

maybe I can send half data every 8.88 MSEC, instead 16.666 ?

is it possible ?

eldadwasserman, 2017-11-07:
I mean 8.3333 Msec

בתאריך 7 בנוב׳ 2017 21:54,‏ "Eldad Wasserman" <eldad.wasserman@gmail.com>
כתב:

lucazader, 2017-11-07:
Unfortunately no, 
The BLE interval has to be a multiple of 1.25ms.

So in software if you set the interval as 6, it will equal 7.5ms.
If it is set to 7, it will have an interval of 8.75ms,
an interval of 8 would have an interval of 10ms
etc

eldadwasserman, 2017-11-07:
So, I believe that the best solution is classic BT and not BLE

lucazader, 2017-11-07:
Classic BT would allow such transmission yes.
Much higher power budget and slightly more involved connection procedure unfortunately.
But if you absolutely must send data with 1us accuracy then i would suggest wireless comms is probably not going to be accurate enough for you.

eldadwasserman, 2017-12-08:
Hi, lucazader.

eldadwasserman, 2017-12-08:
I simulate my output transmission with a different ration - 50 times per sec of 48 bytes each time. so actually i need to transmit 2400 bytes per second.

eldadwasserman, 2017-12-08:
I will be happy to get some clue what is the best example code of esp-idf for transmitting these 2400 bytes per sec to iphone, android, pc, mac, and also get some commands back to the ESP32. Thanks alot for any kind of advice 

Eldad

eldadwasserman, 2017-12-08:
so every 20MSEC i want to transmit 48 bytes

eldadwasserman, 2018-01-23:
Hi All,

Any chance to get help from anyone ?
Would like to transmit every 20msec 48 bytes to Android / IOS...

Eldad

chegewara, 2018-01-23:
@eldadwasserman Just change delay in loop and you have example
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_notify/BLE_notify.ino

eldadwasserman, 2018-01-23:
Hi Chegewara (nice name :) )

Thanks for your amazing fast reply. I am using ESP-IDF - I am not using Arduino IDE development environment. Can you send me example that related to ESP-IDF ?

Eldad

chegewara, 2018-01-23:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleNotify.cpp

eldadwasserman, 2018-01-23:
Sorry to annoying you... it looks like it's C++ example - do you have C example ?

Eldad

chegewara, 2018-01-23:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_server/main/gatts_demo.c

eldadwasserman, 2018-01-23:
Hi Chegewara,

Can you communicate with me directly ?

My email is eldad.wasserman@gmail.com

Eldad

chegewara, 2018-01-23:
Sure, or we can contact on IRC.

eldadwasserman, 2018-01-23:
Hi,

 

Actually the code of the gatt_server is very complicated and I am trying to move forward ASAP.

I will be happy to pay to someone that will prepare for me a working function that I can use.

I need to send 48 bytes X 50 times per second (every 20MSEC) total of 2400 bytes per second to mobile phone (Android or IOS).

The ESP32 is great MCU for us and working great in my project – now I am using OTG USB cable to communicate between Android and the MCU.

Before we have Arduino nano – and we migrate everything to ESP32 – and now we have much more CPU power and memory capacity to enhanced our product.

 

Please advise if I can get this kind of help to save some development time in the BLE issue.

 

Eldad