espressif/esp-idf #762: Arduino as IDF component : Struggling to get beyond examples

dmody, 2017-07-02:
Gang,
I know this will seem like I haven't read the manuals, but I think I have, and I'm missing some fundamental knowledge.  Could use a pointer or two (not the * type).

* I have the IDF setup (I think).
* I can make flash and monitor the hellow world
* I copied the blink example, and again I can make and flash that, and I can change the flashing speed.

*I'm trying to follow the instructions here:  https://github.com/espressif/arduino-esp32/tree/87093368d9b42d8a211d95ff54b8d3d2e26120d2#using-as-esp-idf-component

  * I took the copied blink example and followed this page's instructions to install the arduino 
  * But.. I can create a file called main.cpp and put the code in there, but if I put that file in the "main" folder, what gets compiled is the old blink example.  Question?  Where is it that I tell the IDF what file my main function resides in, or how do I tell the compiler to compile main.cpp instead of blink.c?
 * If I just copy the code from main.cpp example and put it into the blink.c file, a bunch of errors occur such as:
'
$ make flash
CC blink.o
C:/msys32/home/dave/soil_sensor/main/blink.c:3:8: error: expected identifier or '(' before string constant
 extern "C" void app_main()
        ^
make[1]: *** [/home/Dave/esp-idf/make/component_wrapper.mk:211: blink.o] Error 1
make: *** [C:/msys32/home/Dave/esp-idf/make/project.mk:386: main-build] Error 2
'

So it would seem that I can't actually compile arduino code, which makes me think I'm a) dumb, and b) have missed something in the instructions.  

Thanks for your help.

FHFS, 2017-07-04:
I think it just compiles all files in the main folder.
app_main() is your entry point(where your program starts), comment that out in your blink.c and make sure you have one in your main.cpp

> make menuconfig has some Arduino options
>     "Autostart Arduino setup and loop on boot"
>         If you enable this options, your main.cpp should be formated like any other sketch
> Else you need to implement app_main() and call initArduino(); in it.

inforaudio, 2017-08-18:
Is very dificult understand it. We have the same problem and we can advance with it.
All for only activate the watchdog on ESP32.
Do you find a very easy tutorial about it?

projectgus, 2019-02-25:
Hi @dmody ,

I know this issue has been open a for long time with no update, sorry about that. For future reference the best place to ask questions like this is the esp32.com forum.

For benefit of future readers:

> * But.. I can create a file called main.cpp and put the code in there, but if I put that file in the "main" > folder, what gets compiled is the old blink example.  Question?  Where is it that I tell [...]

By default IDF will compile all of the files in the source folders of the project's components. To build main.cpp instead of blink.c, delete blink.c and replace it with main.cpp.

> * If I just copy the code from main.cpp example and put it into the blink.c file, a bunch of errors occur such as: (error)

C++ is not the same language as C. IDF build system compiles .cpp files with a C++ compiler but C files with a C compiler. If you try to compile C++ as C, you'll get errors from any unsupported parts.

If you have further questions about using Arduino in IDF, please consider registering on https://esp32.com and posting questions there. Responses should be more forthcoming than they have been here.

closed 2019-02-25 by projectgus