espressif/esp-idf #764: BLE  GATT Client: More than 3 connected peripherals

lucazader, 2017-07-03:
When trying to setup the GATT_Client to have more than 3 connected peripherals, the error `ESP_GATT_NO_RESOURCES` reported back when trying to add the 4th profile.

```
esp_ble_gattc_app_register(PROFILE_A_APP_ID);
esp_ble_gattc_app_register(PROFILE_B_APP_ID);
esp_ble_gattc_app_register(PROFILE_C_APP_ID);
esp_ble_gattc_app_register(PROFILE_D_APP_ID);
```

I have had a brief look into this. It seems that the limitation is in the file `bta_gattc_int.h`.
In this file there are the definitions: 
```
#ifndef     BTA_GATTC_CL_MAX
#define     BTA_GATTC_CL_MAX    3 // 32
#endif
```
```
/* max known devices GATTC can support */
#ifndef     BTA_GATTC_KNOWN_SR_MAX
#define     BTA_GATTC_KNOWN_SR_MAX    3 // 10
#endif
```

I have found that increasing these to 7 (the max number of ACL connections that is configurable using make menuconfig) fixes the issue and allows the 4th profile to be added.
I have not done extensive testing with this yet, however it seems strange that the max ACL connections can be set to 7 if the gatt_client cant set up enough profile handlers to deal with them.

Please let me know if there is more information needed.

Weijian-Espressif, 2017-07-07:
@lucazader , the default maximum of register profile is 3, you can modify the num for register more profiles. By the way , register one profile is enough to handle multiple connections.

FayeY, 2017-07-13:
Hi @lucazader , did the above suggestion help you resolve the issue?

lucazader, 2017-07-15:
hi @FayeY 

Yes changing the value seems to have helped. I can now register more then 3.
I also see there is a commit for the v3.0 release that will address this issue.

@Weijian-Espressif 
I have not yet tried handling multiple connection using one profile. Can you explain how this can be achieved?

Weijian-Espressif, 2017-07-17:
@lucazader In one profile, different device have different conn_id. This method may be  complicated to deal with. if you have enouch ram, you can register multiple app_id . it is simple to handle the connection event.

lucazader, 2017-07-17:
Hi @Weijian-Espressif 

Thanks for the advice. I am currently already doing a similar method, using the gatt_if instead of the conn_id.
I had found that there was one instance where gatt_if was present but conn_id was not in a callback somewhere. I will have more of a look into this.

Thanks for the advice!

Weijian-Espressif, 2017-07-24:
@lucazader  you could get the conn_id in the open event.
/**
     * @brief ESP_GATTC_OPEN_EVT
     */
    struct gattc_open_evt_param {
        esp_gatt_status_t status;		/*!< Operation status */
        uint16_t conn_id;				/*!< Connection id */
        esp_bd_addr_t remote_bda;		/*!< Remote bluetooth device address */
        uint16_t mtu;					/*!< MTU size */
    } open;								/*!< Gatt client callback param of ESP_GATTC_OPEN_EVT */

lucazader, 2017-08-02:
Hi,

Yes this can be marked as solved for now.

closed 2017-08-03 by FayeY