espressif/esp-idf #765: mbedtls_pk_parse_key: `PK - Invalid key tag or value`

ThisNameIsNotAllowed, 2017-07-03:
Using the `mbedtls_pk_parse_key` method will provide the following error message:
`PK - Invalid key tag or value`.

From the changelog of mbedtls I got the info to use a null terminated string as private key.
(https://github.com/espressif/esp-idf/blob/release/v2.0/components/mbedtls/ChangeLog)

Using the `mbedtls_pk_parse_public_key` method, is working as expected.
I also checked the private and public keys. (Used `PEM` and `DER` format)

My keys are 2048 bit ones.
``` c

  static const uint8_t internal_rsa_key_der[] = {};// Not provided in issue
  static const uint8_t internal_rsa_priv_key_der[] = {};//Not provided in issue

esp_err_t internal_encryption_non_cert(uint8_t* data, uint8_t* result, size_t data_length, size_t* output_length){
  esp_err_t returnValue = ESP_FAIL;
  int error_code = 0;

  if(data_length > RSA_MAX_DATA_SIZE){
    #ifndef NO_ESP_LOGS_FLAG
      ESP_LOGE(MY_TAG, "GIVEN DATA LENGTH EXCEEDS MAX LENGTH OF: %i BYTES\n", RSA_MAX_DATA_SIZE);
    #endif
    return returnValue;
  }
    mbedtls_pk_context pk;
    mbedtls_pk_init(&pk);
    #ifndef NO_ESP_LOGS_FLAG
      ESP_LOGI(MY_TAG, "READING KEY TO VARIABLE\n");
    #endif
	
    //Reading rsa public key (Use DER-Encoded certificates only)
	
/*
	//Working
    error_code =  mbedtls_pk_parse_public_key(
		&pk,
		internal_rsa_key_der,
		sizeof(internal_rsa_key_der)
	);
*/	
	//Not working
    error_code = mbedtls_pk_parse_key(
      &pk,
      internal_rsa_priv_key_der,
      (sizeof(internal_rsa_priv_key_der)+1),
      NULL,
      0
    );

    if( error_code == 0){
      #ifndef NO_ESP_LOGS_FLAG
        ESP_LOGI(MY_TAG, "READING SUCCESSFULL\n");
      #endif
    } else {

      char error_output[500];
      mbedtls_strerror(error_code, &error_output, sizeof(error_output));
      #ifndef NO_ESP_LOGS_FLAG
        ESP_LOGE(MY_TAG, "%s\n", &error_output[0]);
      #endif

      #ifndef NO_ESP_LOGS_FLAG
        ESP_LOGE(MY_TAG, "ERROR READING KEY\n");
      #endif
      return returnValue;
    }

    mbedtls_entropy_context entropy;
    mbedtls_entropy_init( &entropy );
    const char* pers = "dh_genprime";

    //Generation of encrypted value
    mbedtls_ctr_drbg_context ctr_drbg;
    mbedtls_ctr_drbg_init( &ctr_drbg );
    #ifndef NO_ESP_LOGS_FLAG
      ESP_LOGI(MY_TAG, "INITIALIZING SEED\n");
    #endif
    error_code = mbedtls_ctr_drbg_seed(&ctr_drbg, mbedtls_entropy_func, &entropy, (const unsigned char*)pers, strlen(pers));
    if( error_code != 0){
      #ifndef NO_ESP_LOGS_FLAG
        ESP_LOGE(MY_TAG, "ERROR WHILE INITIALIZING SEED\n");
      #endif
      char error_output[500];
      mbedtls_strerror(error_code, &error_output, sizeof(error_output));
      #ifndef NO_ESP_LOGS_FLAG
        ESP_LOGE(MY_TAG, "%s\n", &error_output[0]);
      #endif
      return ESP_FAIL;
    }
    #ifndef NO_ESP_LOGS_FLAG
      ESP_LOGI(MY_TAG, "SEED INITIALIZED\n");
    #endif
    error_code = mbedtls_pk_encrypt(&pk, &data, data_length, result, &output_length, 256, mbedtls_ctr_drbg_random, &ctr_drbg);
    ESP_LOGI(MY_TAG, "REACH AFTER PK_ENCRYPT");
    if( error_code == 0){
      #ifndef NO_ESP_LOGS_FLAG
        for(int ii = 0 ; ii < output_length ; ii++){
          ESP_LOGI(MY_TAG, "%x", result[ii]);
        }
        ESP_LOGI(MY_TAG, "SIZE OF ENCRYPTED: %i\n", (int)output_length);
        ESP_LOGI(MY_TAG, "ENCRYPTING SUCCESSFULL\n");
      #endif
      returnValue = ESP_OK;
    } else {
      #ifndef NO_ESP_LOGS_FLAG
        ESP_LOGE(MY_TAG, "ERROR ENCRYPTING WITH PUBLIC KEY\n");
        ESP_LOGE(MY_TAG, "ERROR CODE: 0x%04x\n", error_code);
        char error_output[500];
        mbedtls_strerror(error_code, &error_output, sizeof(error_output));
        ESP_LOGE(MY_TAG, "%s\n", &error_output[0]);
      #endif
    }
  return returnValue;
}
```

closed 2017-07-07 by ThisNameIsNotAllowed

nkolban, 2018-08-25:
What was the solution?

sperrys, 2018-12-31:
Any update here?