espressif/esp-idf #766: esp_ble_gatts_set_attr_value truncates value

mh-dev, 2017-07-04:
It took me a few hours to figure out why corrupted/truncated data was transmitted to my client.

The root cause is in https://github.com/espressif/esp-idf/blob/c4edb2a4398aef32822b1f917157e402cc246e64/components/bt/bluedroid/btc/profile/std/gatt/btc_gatts.c#L128
This row performs a downcast from uint16_t to uint8_t. Depending on the passed set value length, the copied length is between 0 and 255 even if its above 255.
The fix is just replace uint8_t with uint16_t and you're good to go. This solved the problem for me.

negativekelvin, 2017-07-04:
I think you mean [https://github.com/espressif/esp-idf/blob/c4edb2a4398aef32822b1f917157e402cc246e64/components/bt/bluedroid/btc/profile/std/gatt/btc_gatts.c#L141](L141)?

mh-dev, 2017-07-04:
Yes I did, I updated the issue

Yulong-espressif, 2017-07-04:
@mh-dev Thank you very much for find the IDF bug to us. We will fixed this bug soon on the new version of SDK.

FayeY, 2017-08-09:
Hi @mh-dev , please have a try with the latest ESP-IDF. If the issue is still unsolved, please feel free to let us know. Thanks for your interest in ESP32!

closed 2017-08-09 by FayeY