espressif/esp-idf #768: emac: emac resetting error

mseerig, 2017-07-04:
I tried to setup an ethernet adapter with the example "ethernet" code. My hardware is an Olimex ESP32-EVB äquialent with the LAN8710A but it has an phy_en pin and my phy_addr is PHY0. As you see, I found out the right configurations for my board. ;)
After flashing the customized example code to my board, I got this:

> (2567) cpu_start: Pro cpu start user code
> (2932) cpu_start: Starting scheduler on PRO CPU.
> (3037) emac: mac version 1137a
> (3037) emac: emac start !!!
> (3037) emac: emac resetting ....
> (3037) emac: emac resetting ....
> (3037) emac: emac resetting ....
> (3037) emac: emac resetting ....
> (3047) emac: emac resetting ....
> (3047) emac: emac resetting ....
> (3057) emac: emac resetting ....
> (3057) emac: emac resetting ....
> (3067) emac: emac resetting ....
> (3067) emac: emac resetting ....
> (3067) emac: emac resetting ....
> (3077) emac: emac resetting ....
> (3077) emac: emac resetting ....

... and so on.
As you know, the LAN8710A has a clock input, so the documented reason for my issue doesn't fit.. 

```
void emac_reset(void) {
    REG_SET_BIT(EMAC_DMABUSMODE_REG, EMAC_SW_RST);
    while (REG_GET_BIT(EMAC_DMABUSMODE_REG, EMAC_SW_RST) == 1) {
        //nothing to do ,if stop here,maybe emac have not clk input.
        ESP_LOGI(TAG, "emac resetting ....");
    }
    ESP_LOGI(TAG, "emac reset done");
}
```

Is this a caused by my code, or is this maybe a idf software bug?
Thank you for your help! :)

RobinCutshaw, 2017-07-05:
This happened to me.  You have to route the PHY clock (50MHz) to GPIO0 (which is the same pin used to indicate program mode upon toggling the enable pin).  The tricky part is to prevent the PHY clock from being active during reset or you will have a 50/50 chance of going into program mode.

mseerig, 2017-07-05:
My PHY clock (50MHz) is routed to GPIO0. My problem is, that the clock is always high and not active. So far, I have not seen a 50 MHz clock on this line. All data pins and also the clock pin has a constant value (clearly high or low). 

Remember: I'm using the example code...

What is the function of the `EMAC_DMABUSMODE_REG`?
It seems, that here is the problem...

I have the ESP32D0WDQ6 (revision 0).

Are there additional changes to do, before I get a 50Mhz clock?

RobinCutshaw, 2017-07-05:
Remember that GPIO0 is an input for your external 50MHz oscillator.  What type of oscillator are you using?

hwmaier, 2017-07-06:
You have to check your hardware how the 50MHz is generated. You may have to enable the oscillator or the PHY if the PHY generates the clock, before you will see the 50 MHz on GPIO0. The esp-idf ethernet example supports a PHY_USE_POWER_PIN setting which you can use to configure a GPIO to enable either an oscillator or a PHY enable signal.

volca, 2017-07-06:
Happend to me also. I tried to edit the lines [idf_monitor.py](https://github.com/espressif/esp-idf/blob/master/tools/idf_monitor.py#L460) and it works again

Change 

```
    serial_instance.dtr = False
    serial_instance.rts = False
```
To

```
    serial_instance.dtr = True
    serial_instance.rts = True
```

My hardware is similar to devKitC

mseerig, 2017-07-07:
Thanks, it works! It was a mixture of the DTR and RTS thing and with the clock input to the GPIO0 Pin. It's very embarrassing, that I didn't see, that I have to add an external clock to GPIO0. However, shit happens. :/
Next time, I will check my circuit and the documentation more intensive!
To my excuse, the Olimex-EVB rev. A has also no external clock. That was the main reason why I thought that GPIO0 gives me the necessary clock..

Sorry for that, a thank you for your help!

closed 2017-07-07 by mseerig

volca, 2017-07-07:
@mseerig Did you mean you fix the issue just by adding an external clock to GPIO0?