espressif/esp-idf #770: GATT client connection loss

skateone, 2017-07-04:
i am attempting a GATT client on the esp32 connecting to a nordic nrf52832.
this is the example i am following :https://www.youtube.com/watch?v=brD3dYTK6Fg

i have managed to get the two devices communicating but for some reason i get a disconnect with an unknown reason code:
bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x003b.

is there anything i can do or is this a possible issue?
regards
Skate-one

heyinling, 2017-07-05:
@skateone what's the reproduce ratio of disconnect on your side? I also met this issue, but quite low ratio, say below 10% or 5%. want to confirm if it's a compatible issue with the nordic chip.

skateone, 2017-07-05:
100%. it happens every time i perform the connection.

Weijian-Espressif, 2017-07-10:
@skateone I am testing nordic ble to locate the point.

Weijian-Espressif, 2017-07-11:
@skateone  I have tested nrf51822 , everything is ok.  please open the bt log in bt_trace.h , give me more information.

skateone, 2017-07-11:
this is how we fixed it on the nordic.

looks like the NRF was closing the connection 

//PROBLEM
static void on_conn_params_evt(ble_conn_params_evt_t * p_evt) {
    uint32_t err_code;

    SEGGER_RTT_WriteString(0,"[MAIN]on_conn_params_evt\n\r");

    if (p_evt->evt_type == BLE_CONN_PARAMS_EVT_FAILED) {
        err_code = sd_ble_gap_disconnect(m_conn_handle, BLE_HCI_CONN_INTERVAL_UNACCEPTABLE);
        APP_ERROR_CHECK(err_code);
    }
}

//FIX
static void on_conn_params_evt(ble_conn_params_evt_t * p_evt) {
   // uint32_t err_code;

    SEGGER_RTT_WriteString(0,"[MAIN]on_conn_params_evt\n\r");

    if (p_evt->evt_type == BLE_CONN_PARAMS_EVT_FAILED) {
        //err_code = sd_ble_gap_disconnect(m_conn_handle, BLE_HCI_CONN_INTERVAL_UNACCEPTABLE);
       // APP_ERROR_CHECK(err_code);
    }
}

Weijian-Espressif, 2017-07-11:
@skateone  Can you give me your nordic51822 software and application hex?

joicetm, 2017-07-17:
@skateone   which is the ide you are you using to program nrf5238 ?.

Weijian-Espressif, 2017-07-20:
hi @skateone , we fix a bug for connection param update , please update to the latest idf and try. if any question , let me know.

skateone, 2017-07-20:
thank you for the swift reply... will try it and let you kow

skateone, 2017-07-20:
@joicetm we are using the SDK v11.0 on the nordic side for the NRF52832.

FayeY, 2017-08-01:
Hi @skateone , any update? Is this problem solved in the latest ESP-IDF?

skateone, 2017-08-01:
i will test the lasted IDF... we managed to get this working a few days ago and the BLE link is constant..

closed 2017-08-01 by skateone

imotion-software, 2017-10-06:
Hi @skateone,

we're using latest version of ESP-IDF (Sept 30th), and we're facing a very similar issue on Nordic NRF52832. This behavior happens half the time, and we thinks it's due to ESP32, because we can successfully connect to Nordic device from any other source (like iOS Apps and Android Apps).

During connect we get an error 133 due to device that sends a status 12 (Command Disallowed) to a HCI_BLE_READ_REMOTE_FEAT request.

We enabled BT logs. You can find them (for a failed connection) below:

```
I (19451) <BLTCOM>: Connecting to device: EF1050DFD1A3
E (19453) BT: btc_transfer_context msg 0 3 2 0x3ffeda78
E (19457) BT: btc_task msg 0 3 2 0x3ffd9af0
E (19467) BT: BTA got event 0x1f00
E (19473) BT: HCI Enqueue Comamnd opcode=0x200d
E (19477) BT: Receive packet event_code=0xf
E (19481) BT: *** Received Status: 0
E (19467) BT: btc_gattc_arg_deep_free Unhandled deep free 2
E (19485) BT: alarm_cbs_lookfor_available 3 0x3ffc9dc4
E (20086) BT: Receive packet event_code=0x3e
E (20092) BT: HCI Enqueue Comamnd opcode=0x2016 HCI_BLE_READ_REMOTE_FEAT
E (20170) BT: Receive packet event_code=0xf
E (20173) BT: *** Received Status: 12
E (20176) BT: Receive packet event_code=0x5
E (20180) BT: bta_gatts_conn_cback gatt_if=3 conn_id=3 connected=0 reason = 0x0062
E (20184) BT: bta_gatts_conn_cback  bda :ef-10-50-df-d1-a3 
E (20190) BT: btc_transfer_context msg 1 2 15 0x3ffe2210
E (20197) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x003e
E (20205) BT: p_conn is NULL in bta_gattc_conn_cback
E (20211) BT: BTA got event 0x1f11
E (20215) BT: btc_transfer_context msg 1 3 36 0x3ffe20d0
E (20197) BT: btc_task msg 1 2 15 0x3ffd9af0
E (20221) BT: btc_transfer_context msg 1 3 2 0x3ffe20b0
E (20232) BT: btc_task msg 1 3 36 0x3ffd958c
E (20237) BT: btc_task msg 1 3 2 0x3ffedd24
E (20242) <BLTCOM>: Connection failed, status 133
```

Do you know the reason?