espressif/esp-idf #771: Unable to use esp_ble_gatts_get_attr_value after BLE long write

mh-dev, 2017-07-05:
**Precondition**
esp_ble_gatts_create_attr_tab is used to setup my bluetooth service characteristics. 

**Bug**
My client performs a long write. When I try to read the attribute value in 'ESP_GATTS_EXEC_WRITE_EVT' with esp_ble_gatts_get_attr_value the following appears in my log:
'gatts_get_attribute_vaule failt:the value length is 0'

Further research resulted in a not set length attribute in the logic.
The row https://github.com/espressif/esp-idf/blob/master/components/bt/bluedroid/stack/gatt/gatt_sr.c#L405 does copy the value over but does not set the length attribute

Adding the following below this row fixes the problem. I can not tell if this has other side effects so I recommend to check this on your own.

`queue_data->p_attr->p_value->attr_val.attr_len = queue_data->p_attr->p_value->attr_val.attr_len + queue_data->len;`

mh-dev, 2017-07-14:
Any update on this?
The fix is not really complicated.

mh-dev, 2017-08-11:
I did further testing, the solution mentioned above is only partially correct.
The value needs to be set to zero before it gets set the first time in the while loop.

Yulong-espressif, 2017-09-28:
@mh-dev where is this code ? queue_data->p_attr->p_value->attr_val.attr_len = queue_data->p_attr->p_value->attr_val.attr_len + queue_data->len. I can't find it in the bluedroid.

mh-dev, 2017-09-28:
Its expected that you can't find the code, since its not existence is the actual problem.

The current problem location is https://github.com/espressif/esp-idf/blob/master/components/bt/bluedroid/stack/gatt/gatt_sr.c#L423

This row copies the value but not its length.

My local code looks like the following (is not fully on the same version as the one on github)

```
uint8_t length_initialized = 0;
//dequeue prepare write data
while(GKI_getfirst(&(prepare_record->queue))) {
	queue_data = GKI_dequeue(&(prepare_record->queue));
	if (is_prepare_write_valid){
		if((queue_data->p_attr->p_value != NULL) && (queue_data->p_attr->p_value->attr_val.attr_val != NULL)){
			memcpy(queue_data->p_attr->p_value->attr_val.attr_val+queue_data->offset, queue_data->value, queue_data->len);
			if(length_initialized == 0) {
				queue_data->p_attr->p_value->attr_val.attr_len = 0;
				length_initialized = 1;
			}
			queue_data->p_attr->p_value->attr_val.attr_len = queue_data->p_attr->p_value->attr_val.attr_len + queue_data->len;
		}
	}
	GKI_freebuf(queue_data);
} 
```

Yulong-espressif, 2017-10-11:
@mb-dev thank you so so much, it is really a bug in our stack. we have change it now. Thank you for your interesting in the ESP-IDF.

closed 2017-11-21 by FayeY

ujeenator, 2017-11-23:
@Yulong-espressif Hello, in which version it was fixed? 

I don't see linked commit, and still experience this in v2.1

ujeenator, 2017-11-23:
@Yulong-espressif thank you, I take latest available ESP-IDF v2.1.1 and applied your fix manually, everything work like a charm

jfdevmoto, 2018-02-11:
I have a problem after long writes. It looks like the lenght value is doubled or gets corrupted.
I have a 159 bytes long characteristic. Then my client writes 159 bytes. Suddenly during the last GATT_WRITE_EVT callback processing, the lenght value becomes 318. Then I receive the ESP_GATTS_EXEC_WRITE_EVT.
I think there is a regression in the code as latest code varies significantly from what is written above.
line 425 may have to be changed to 

queue_data->p_attr->p_value->attr_val.attr_len = queue_data->len;
 instead of 
queue_data->p_attr->p_value->attr_val.attr_len += queue_data->len;

Can you confirm ? I don't have much time to look deeper and validate. Would be great for you to confirm before I come back from holliday.

If this is not the issue, please give me some other help.
I am working with esp-idf 3.1 dev168
Thanks