espressif/esp-idf #773: build projiect problem

noticeable, 2017-07-05:
hello ,i have seen a problem when i hit build project ,this is the console give to me:
make all 
WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-59-g8d95cad
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 4.8.5
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-59-g8d95cad
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 4.8.5
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC src/bootloader_random.o
In file included from /cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:20:0,
                 from /cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/wdev_reg.h:17,
                 from /cygdrive/d/eclipse/esp-idf/components/bootloader_support/src/bootloader_random.c:16:
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h: In function 'PIN_PULLUP_DIS':
/cygdrive/d/eclipse/esp-idf/components/esp32/include/esp_assert.h:23:28: error: first argument to '__builtin_choose_expr' not a constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:169:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_CLR_BIT);                                                                    \
             ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:96:5: note: in expansion of macro 'REG_CLR_BIT'
     REG_CLR_BIT(PIN_NAME, FUN_PU);
     ^
/cygdrive/d/eclipse/esp-idf/components/esp32/include/esp_assert.h:23:28: error: expression in static assertion is not constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:169:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_CLR_BIT);                                                                    \
             ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:96:5: note: in expansion of macro 'REG_CLR_BIT'
     REG_CLR_BIT(PIN_NAME, FUN_PU);
     ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h: In function 'PIN_PULLUP_EN':
/cygdrive/d/eclipse/esp-idf/components/esp32/include/esp_assert.h:23:28: error: first argument to '__builtin_choose_expr' not a constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \
             ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:101:5: note: in expansion of macro 'REG_SET_BIT'
     REG_SET_BIT(PIN_NAME, FUN_PU);
     ^
/cygdrive/d/eclipse/esp-idf/components/esp32/include/esp_assert.h:23:28: error: expression in static assertion is not constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \
             ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:101:5: note: in expansion of macro 'REG_SET_BIT'
     REG_SET_BIT(PIN_NAME, FUN_PU);
     ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h: In function 'PIN_PULLDWN_DIS':
/cygdrive/d/eclipse/esp-idf/components/esp32/include/esp_assert.h:23:28: error: first argument to '__builtin_choose_expr' not a constant
[... 25 lines cut ...]
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \
             ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:111:5: note: in expansion of macro 'REG_SET_BIT'
     REG_SET_BIT(PIN_NAME, FUN_PD);
     ^
/cygdrive/d/eclipse/esp-idf/components/esp32/include/esp_assert.h:23:28: error: expression in static assertion is not constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \
             ^
/cygdrive/d/eclipse/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:111:5: note: in expansion of macro 'REG_SET_BIT'
     REG_SET_BIT(PIN_NAME, FUN_PD);
     ^
make[1]: *** [/cygdrive/d/eclipse/esp-idf/make/component_wrapper.mk:211: src/bootloader_random.o] Error 1
make: *** [/cygdrive/d/eclipse/esp-idf/make/project.mk:387: bootloader_support-build] Error 2

projectgus, 2017-07-05:
Thanks for reporting these errors. I have a couple of questions:

* How did you configure your computer for ESP-IDF? I noticed you are using Cygwin, this is different to our [supported Windows environment which is documented here](http://esp-idf.readthedocs.io/en/latest/get-started/windows-setup.html). Are you using a different setup? Do you have a link we can follow?

noticeable, 2017-07-05:
different？i learn the setup from the bbs http://bbs.elecfans.com/jishu_946533_1_1.html  step by step ,i can get the right reply at first ,but when i get the last step ,the errors come.

igrr, 2017-07-23:
Please use the official setup guide: http://esp-idf.readthedocs.io/en/latest/get-started/windows-setup.html
As @projectgus has mentioned above, Cygwin is not supported by our build system.

closed 2017-08-17 by igrr