espressif/esp-idf #774: Stack overflow in IDLE task.

jaracil, 2017-07-05:
When I switch my project form Debug to Release mode, stack overflow in task IDLE is raised and there is no way in menuconfig to changle idle task stack size.

`***ERROR*** A stack overflow in task IDLE has been detected.
abort() was called at PC 0x400867d4 on core 1
0x400867d4: vApplicationStackOverflowHook at /home/pepe/esp32/esp-idf/components/esp32/./panic.c:118

Backtrace: 0x4008678f:0x3ffb5620 0x400867bb:0x3ffb5640 0x400867d4:0x3ffb5660 0x400850c8:0x3ffb5680 0x400864d0:0x3ffb56a0 0x4008648a:0xa5a5a5a5
0x4008678f: invoke_abort at /home/pepe/esp32/esp-idf/components/esp32/./panic.c:138

0x400867bb: abort at /home/pepe/esp32/esp-idf/components/esp32/./panic.c:147

0x400867d4: vApplicationStackOverflowHook at /home/pepe/esp32/esp-idf/components/esp32/./panic.c:118

0x400850c8: vTaskSwitchContext at /home/pepe/esp32/esp-idf/components/freertos/./tasks.c:4476

0x400864d0: _frxt_dispatch at ??:?

0x4008648a: _frxt_int_exit at ??:?`

Thanks.

HubbyGitter, 2017-07-06:
Which stack overflow detection method did you enable? If it's by canary bytes then there could be a chance that the reason is not really the idle task (but there it is noticed, e.g. if the idle task has to kick the watchdog).

If your control flow returns from a task's executed function then that's also a case where all bets are off.

Did you also change the optimisation level?

The reported error is so unlikely that you generally want to think out of the box.

jaracil, 2017-07-07:
I have canary bytes enabled. If I change it to "check by stack pointer value" it works Ok in release mode (-Os).

Why canary bytes are smashed in release mode?

Thanks.

HubbyGitter, 2017-07-07:
Sounds like you have something messing up memory area which is used by a task's stack. Check whether you hand out references to automatic variables which are going to be used out of that variable's scope. Also check all pointer arithmetics.

projectgus, 2017-07-07:
Hi @jaracil ,

To 100% rule out an actual idle task stack overflow, could you please edit the stack size manually in components/freertos/tasks.c:117 and recompile?

The other thing to do is enable "Set a debug watchpoint as a stack overflow check" which is under the normal stack checker mode. This is very useful check, as any stack overflow will cause an immediate debug exception and the backtrace will show you exactly where the overflow happened - unlike the other check types which only fire during a context switch.

If the debug watchpoint doesn't fire (ie you can run "check by stack pointer" + debug watchpoint and nothing crashes), then odds are that this is memory corruption from somewhere else which happens to clobber the idle task stack (but not while the idle task is running). There are some things you can try to help triangulate this, but I'll wait until you get some more information first.

jaracil, 2017-07-07:
Hi @projectgus 

This is the backtrace with "Set a debug watchpoint as a stack overflow check" enabled:

`I (1663) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1973) phy: phy_version: 355.1, 59464c5, Jun 14 2017, 20:25:06, 0, 0
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (IDLE) 
Register dump:
PC      : 0x4000bff0  PS      : 0x00060e36  A0      : 0x80085599  A1      : 0x3ffb56a0  
A2      : 0x00000000  A3      : 0x00060e23  A4      : 0x00060e20  A5      : 0x00000005  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x80084c82  A9      : 0x3ffb5740  
A10     : 0x00000003  A11     : 0x00060e23  A12     : 0x00060e23  A13     : 0x00000001  
A14     : 0x00000001  A15     : 0x3ffb6700  SAR     : 0x00000000  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x4000bff0:0x3ffb56a0 0x40085596:0x3ffb5770 0x400d38c7:0x3ffb5790 0x400d38ff:0x3ffb57c0 0x400d39ea:0x3ffb57e0 0x400859f4:0x3ffb5800
0x40085596: vTaskExitCritical at /home/pepe/esp32/esp-idf/components/freertos/./tasks.c:4476

0x400d38c7: esp_task_wdt_feed at /home/pepe/esp32/esp-idf/components/esp32/./task_wdt.c:172

0x400d38ff: idle_hook at /home/pepe/esp32/esp-idf/components/esp32/./task_wdt.c:172

0x400d39ea: esp_vApplicationIdleHook at /home/pepe/esp32/esp-idf/components/esp32/./freertos_hooks.c:46

0x400859f4: prvIdleTask at /home/pepe/esp32/esp-idf/components/freertos/./tasks.c:4476

`

jaracil, 2017-07-07:
@projectgus 

In components/freertos/tasks.c:117

#define tskIDLE_STACK_SIZE      512  // Stack Overflow
#define tskIDLE_STACK_SIZE     1024 // Works OK

igrr, 2017-10-13:
@jaracil Idle stack size has been increased in https://github.com/espressif/esp-idf/commit/3ad163e644ae8aa232cd9a761d54698b396454b4.

closed 2017-10-13 by igrr