espressif/esp-idf #775: WL_Flash 读取state结构体内容但没使用.

nickfox-taterli, 2017-07-06:
components/wear_levelling/WL_Flash.cpp:

Line:180 result = this->flash_drv->read(this->addr_state2, &this->state, sizeof(wl_state_t));

&this->state前面代码写入后未改变过,原样读出,后续也没使用过,就算state2所在的sector是坏的,那么只比较version变量也是不管用的啊.我删除后发现工作正常(已经模拟出错情况),不知道是什么原因?

noticeable, 2017-07-06:
同道中人？？建议用英语再发一贴

FayeY, 2017-07-07:
顶楼上！建议用英语再发一遍
答题的都是外国工程师，也蛮考验我国工程师的英文水平哟。

bvernoux, 2017-07-10:
Does it is possible to keep all Issues in english (as global rules and especially because it is public issues) ? as we are a lot to do not understand mandarin chinese

FayeY, 2017-07-11:
@nickfox-taterli 
I tried to translate your question into English, please correct me if I translated it wrong.
The question is:

> 
> components/wear_levelling/WL_Flash.cpp:
> 
> Line:180 result = this->flash_drv->read(this->addr_state2, &this->state, sizeof(wl_state_t));
> 
> Does the data read into "&this->state" be used anywhere?
> 
> It seems that we can NOT know whether the flash sector (which the "addr_state2" located) is broke by only checking the variable "version".

closed 2017-11-21 by FayeY