espressif/esp-idf #778: wifi: ampdu, m eb failed

Nicholas3388, 2017-07-07:
Hi,
I have created custom http webserver with Wi-Fi in station mode. and my PC which is in same network as my ESP32 is sending http request to ESP32. 

It works fine for some http requests. but after some responses it stops responding and shows follwing error on console:
W (416235) wifi: ampdu, m eb failed

The following list the log info of the issue:
I (102409) wifi: bcn_timout,ap_probe_send_start
W (102419) wifi: alloc eb len=76 type=2 fail

W (102419) wifi: m f probe req l=0

I (103289) led: action[0], pin[27]
W (103419) wifi: alloc eb len=76 type=2 fail

W (103419) wifi: m f probe req l=0

W (104409) wifi: alloc eb len=76 type=2 fail

W (104409) wifi: m f probe req l=0

I (104909) wifi: ap_probe_send over, resett wifi status to disassoc
I (104909) wifi: state: run -> init (1)
I (104919) wifi: n:7 0, o:7 2, ap:7 2, sta:7 2, prof:7

W (116939) wifi: ampdu, m eb failed
W (117739) wifi: ampdu, m eb failed
W (118119) wifi: ampdu, m eb failed
W (118389) wifi: ampdu, m eb failed
W (119279) wifi: ampdu, m eb failed
W (120119) wifi: ampdu, m eb failed
W (120129) wifi: ampdu, m eb failed
W (122429) wifi: ampdu, m eb failed
I (122529) wifi: bcn_timout,ap_probe_send_start
W (123389) wifi: ampdu, m eb failed
W (124549) wifi: ampdu, m eb failed
I (125029) wifi: ap_probe_send over, resett wifi status to disassoc
I (125029) wifi: state: run -> init (1)
I (125029) wifi: n:7 0, o:7 2, ap:7 2, sta:7 2, prof:7
W (125139) wifi: ampdu, m eb failed
W (130149) wifi: ampdu, m eb failed
W (133149) wifi: ampdu, m eb failed
W (135129) wifi: ampdu, m eb failed
W (135129) wifi: ampdu, m eb failed
W (146139) wifi: ampdu, m eb failed
W (154699) wifi: ampdu, m eb failed

I can't even establish a connection to the device any more. What could be causing this problem?

Nicholas3388, 2017-07-15:
solved. close issue

closed 2017-07-15 by Nicholas3388

Designer-Rick, 2017-12-07:
I too have this issue how was it solved?

snahmad, 2018-09-05:
I am getting similar error
wifi: alloc eb len=192 type=2 fail

I am using WIFI with external RAM.

Internal Heap free: 41120
Heap free: 4195984:4233744

Plenty of external ram and some internal RAM.

Does in default settings WIFI only uses internal ram.