espressif/esp-idf #782: Bluetooth HID implementation progress?

huming2207, 2017-07-09:
Hi all,

I'm considering to build a bluetooth keyboard by myself later on with ESP32 as its microcontroller. But I can't find much details or documentations about those HID APIs. I did [found a topic](https://www.esp32.com/viewtopic.php?t=377) on Espressif forum saying "will implement it eventually". So...did you finish implementing those stuff?

Regards,
Jackson

FayeY, 2017-07-11:
We are working on the Bluetooth HID now, it will be released in the ESP-IDF V3.0. Thanks for your interest in ESP32!

huming2207, 2017-07-11:
Okay thanks. By the way, will it be released before November, 2017?

huming2207, 2017-07-11:
Okay got it, thanks. By the way, will it be released before November, 2017? @FayeY

Yulong-espressif, 2017-07-11:
@huming2207 do you means HID device or HID host, we will add the HID device soon, but we don't have plan to do the HID host now, Thanks, any other question please let me now.

huming2207, 2017-07-11:
@Yulong-espressif Hmm...should be HID device I think. I just would like to design a bluetooth mechanical keyboard by myself.

Yulong-espressif, 2017-07-11:
okey, we have the HID keyboard demo now, it will be very easy used for you. Thanks.

huming2207, 2017-07-11:
@Yulong-espressif Do you mean you've got a hardware HID keyboard demo, or just some code/libraries/documents etc?

Yulong-espressif, 2017-07-11:
@huming2207 we have the hardware HID keyboard demo now, not just the documents etc. Thanks.

huming2207, 2017-07-11:
@Yulong-espressif Ahhh nice...guess you will beat some HID chips like BCM20730 lol... By the way is there any links available? I've googled it and I can't find any further details about this hardware demo kit.

6enno, 2017-07-22:
@Yulong-espressif I am also in the process of developing an HID device using the ESP32. I can't seem to find any HID keyboard demo in the repository yet but I would be very happy if one was available. Is it possible to get a link or nudge in the right direction? The other examples have been extremely helpful.

neilmendoza, 2017-08-05:
@Yulong-espressif Where would I find the keyboard demo code? Do you have any plans to create an example of a HID MIDI device? Thanks!

Yulong-espressif, 2017-08-06:
@neilmendoza We will pull the BLE keyboard demo to the github in the middle of this month.

JeremyProffitt, 2017-08-23:
@Yulong-espressif  just curious if there are any updates.  Thanks!

jujucables, 2017-09-07:
I would really really like an update for this! I so want to get a pcb with ESP32 inbuilt for a custom mechanical keyboard as well! 

THANKS SO MUCH FOR YOUR EFFORTS! :)

huming2207, 2017-09-07:
@Yulong-espressif Hi, any updates?

cartufer, 2017-09-13:
I too am waiting on "keyboard demo" as a key part of a project.

mamert, 2017-09-20:
@Yulong-espressif Hi, did you forget to push the keyboard demo?

TianHao-Espressif, 2017-09-25:
@mamert , sorry for HID demo delay. Because of the work priority, the HID demo is ready, but is still not in ESP-IDF.

huming2207, 2017-09-25:
@TianHao-Espressif @Yulong-espressif so...will it be available in December?

mringwal, 2017-09-25:
@huming2207 @mamert @cartufer Hi! You can run BTstack instead of Bluedroid on the ESP32 (https://github.com/bluekitchen/btstack/tree/master/port/esp32) and it comes with a Classic HID Keyboard demo. There's also an HID-over-GATT/BLE demo in the develop branch (and mouse demos).

6enno, 2017-10-31:
Why is this closed @FayeY? Did I miss an update? Is the bluetooth HID demo available for ESP32 _without_ using bluekitchen? I am already using BK as an interim solution however it is slow (and a bit crashy for my particular operation) I am reluctant to spend time fixing it because I am waiting for this version to come through. My product is intended to be commercially available (eventually) and therefore the licencing for BK is not ideal.

cartufer, 2017-10-31:
I too am still waiting.

FayeY, 2017-11-01:
Sorry, my mistake. We will provide a patch recently. Sorry for the inconvenience.

mringwal, 2017-11-01:
Hi @6enno Could you tell us a bit more about your experience with BTstack. It should neither be slow nor crashy. 

By default, full logging, incl. HCI data is activated and this will slow down everything, but it's there for development only. You can disable logging in app_main() by commenting the call to hci_dump_open().

BTstack is not thread-safe. If you call it from a different thread, it will not work. If you have all Bluetooth code on the main thread, it should not crash. Please provide some test code for further investigation.

6enno, 2017-11-01:
Hi @mringwal 
Reading back my message, it sounded harsh and I should apologize. I know that Bluekitchen/BTstack is great solution and commend you guys on your innovative and active development.

I didn't mean to suggest that BTStack itself was crashy and I am aware of the synchronization limits however my particular application involves a lot of high frequency interrupts and context switching between several threads. This will require me to time particular activities and make very specific modifications to the BT code. 

Because this product is being developed with the view of going commercial, it will be difficult for me to navigate the licencing which is not free for commercial use as is the Apache licence for the overall esp-idf.

Also, thanks for the tip on disabling the HCI data, that should speed things up a little in the mean time however I am still reluctant to spend much time on a module that will need to be replaced.

Thanks for everyone's quick responses and help!

Yulong-espressif, 2017-11-03:
@huming2207 @6enno The BLE HID device is ready now, But will be release in the next version, if you are in a hurry, I can send you the demo.

jmarcelino, 2017-11-03:
@yulong-espressif can you send me demo too jmarcelino@pycom.io or post it somewhere. Thanks :)

6enno, 2017-11-03:
Thanks @Yulong-espressif ! that would be fantastic. em: ben.norquay@gmail.com

vikoneal, 2017-11-03:
It would be great to have it since I am currently developing a project that involves HID and I am really close to the deadline. My email is deletsidi@gmail.com. Thanks a lot!!

zoutepopcorn, 2017-11-05:
@Yulong-espressif
can you please send me the example?

zoutepopcorn
@
hotmail
.com

Thanks

Yulong-espressif, 2017-11-06:
@zoutepopcorn @vikoneal @6enno @jmarcelino Have sent the HID demo to your e-mail, please have a check, any question, don't hesitate to let me know.

kat-kai, 2017-11-06:
@Yulong-espressif Could you send me the demo?
My email is kat.kai.kat@gmail.com. Thanks!!

7evin, 2017-11-06:
@Yulong-espressif 
I'm late to the party, but i'd love to check this out too. 7evinmcg@gmail.com

JeremyProffitt, 2017-11-06:
Can someone send it to me as well?  proffitt.jeremy@gmail.com  Thanks!!!

huming2207, 2017-11-07:
@Yulong-espressif  Sorry mate I'm a bit busy these days. Could you please send the email to huming2207@gmail.com? Thanks!

Yulong-espressif, 2017-11-07:
[ble_hid_device_demo.zip](https://github.com/espressif/esp-idf/files/1449197/ble_hid_device_demo.zip)

Yulong-espressif, 2017-11-07:
@huming2207 @JeremyProffitt Please download it directly.

kat-kai, 2017-11-07:
Hi @Yulong-espressif, thank you very much for your quick response!

mringwal, 2017-11-08:
Hi @benjaminaigner. BTstack has added HOG demos for keyboard and mouse a while ago, feel free to give them a try.

Yulong-espressif, 2017-11-08:
@benjaminaigner LOG you provide from the point of view, did not find any problems. So what equipment do you use to scan?  Is it a smart phone?

benjaminaigner, 2017-11-08:
@Yulong-espressif : 
I used a Smartphone with nrfconnect app (much more sophisticated data view).

benjaminaigner, 2017-11-13:
@Yulong-espressif 
It was a defective Sparkfun ESP32 :-(... Everything OK, but it seems to be a faulty RF frontend (nothing received)

@mringwal 
Thanks for the hint, I give BTSTack always a try!

FilbertWong, 2017-11-16:
@Yulong-espressif 
First of all, thanks for your sharing demo. I have run the demo and try connecting to my android phone and basically it can control my phone's speaker volume. Then, I tried modified the demo to make it becomes a ble hid mouse function. Currently, I can control the mouse pointer on my phone but my Windows 10 PC cannot (of course, my PC has got built-in Bluetooth 4.0 le). The situation is that the message shows "driver error" when I tried pairing to PC. Do you have any ideas or suggestions for the issue I'm facing? Any third party app or somethings I can install to my PC to solve this issue?

Yulong-espressif, 2017-11-16:
@FilbertWong I have't used it to control the PC directly, when you want to control the PC, I think you should buy a BLE dongle. For example, TI's CC2540 dongle can be controlled by the BLE PC keyboard and mouse functions.

benjaminaigner, 2017-11-16:
@FilbertWong 
I use this demo controlling an Android 7 phone as well as a Linux PC (BT Dongle: CSR8510 A10, no built-in BLE). It works for mouse and keyboard on both devices.
I will release an extended demo soon, as replacment for all of the BT HID modules (like Adafruit EZ-Keys), and post the link here.
Including text to keystroke translation with ~20 different keyboard layouts (ASCII and Unicode/UTF8) :-)

@Yulong-espressif 

There is still one problem with this demo:
If a device disconnects (out of range, BT switched off,...), I receive:
```
E (3184811) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x001f
E (3184811) BT: hidd_event_callback(), ESP_HIDD_EVENT_BLE_DISCONNECT
E (3184821) BT: hidd_event_callback(), ESP_HIDD_EVENT_BLE_DISCONNECT
```
But there is no ESP_HIDD_EVENT_BLE_CONNECT fired, even if the host is connected again -> no HID reports are sent.
Do you have any suggestion? Maybe a missing event handler?

Thank you very much for this demo, this was a major step for me!

benjaminaigner, 2017-11-16:
I pushed my project to this repo:
https://github.com/asterics/esp32_mouse_keyboard

FilbertWong, 2017-11-17:
@benjaminaigner : Thanks a lots for sharing. So I have to install the dongle's driver and plug it into PC to use, so it's no any different from another Bluetooth/Wireless mice on the market right? I just wonder if there is any ways to salvage the built-in BLE to cut down the 3rd party device :)

FilbertWong, 2017-11-17:
@benjaminaigner : Fortunately, I update the bluetooth driver of my PC, and it can pair with my "esp32 mouse" successfully now.

FilbertWong, 2017-11-20:
Hi All, anyone tried running Bluetooth and Wifi functions in the same time? I have a "esp32_mouse" as Bluetooth LE HID's one and I use Wifi to do file transfer system. And, I can run successfully one by one on my esp32 wrover kit. But when I tried running them in parallel, the board will reboot. Any ideas about this issue?

chegewara, 2017-11-20:
I can guess its low heap amount problem. My advice is to change settings in menuconfig->wifi to lowest, mbedtls from 16kB to 512B, and add esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT); in app_main, if you are using only BLE. Thats some start.

FilbertWong, 2017-11-20:
@chegewara : I used esp_idf 2.0 to compile my project but I couldn't find the function esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT); Then I switch to esp_if_master one to use this function but my BLE HID mouse doesn't work for me any more. It keeps show the error (please see the attached) 
[image]

chegewara, 2017-11-20:
Yours factory partition is too small. Look at message at 533 ms.

FilbertWong, 2017-11-21:
@chegewara : Can I expend the size of this partition?

chegewara, 2017-11-21:
@FilbertWong Of course you can. Are you using arduino ide or esp-idf?

FilbertWong, 2017-11-21:
@chegewara : I use esp-idf. Could you show me how to expand it?

chegewara, 2017-11-21:
You need to change settings in menuconfig. And create csv file, by default its partitions.csv.
This is example partitions.csv content:
```
# Espressif ESP32 Partition Table
# Name,   Type, SubType, Offset,  Size
nvs,      data, nvs,     0x9000,  0x6000
phy_init, data, phy,     0xf000,  0x1000
factory,  app,  factory, 0x10000, 1M

```

You need only change size of factory partition, lets say 2M will be good.

Now options:
from cli run `make menuconfig` and change this option:
`Partition Table->Custom`
All other option in this submenu can be default.

FilbertWong, 2017-11-21:
@chegewara : thank you so much :)

closed 2017-11-21 by FayeY

Sagigamil, 2017-11-25:
I'm looking for a HID host for a few days, someone have an example code to show me?

mringwal, 2017-11-27:
@Sagigamil The HID Host need to find the HID Device somehow (with Classic, doing an Inquiry and looking for devices) and then do an SDP query to get the L2CAP PSMs for HID Interrupt & Control and then open these. HID Reports can be received then (more or less).

For LE, it's similar, just using GATT operations instead of the SDP query.
We don't have ready code for BTstack for this either yet.

mringwal, 2017-11-27:
@Sagigamil I've added an HID Host example to BTstack's develop branch that connects to a known device, gets the PSMs from SDP and opens the HID channels. You're receiving HID reports now. To properly use them, you'd need an HID parser still, but maybe the current version might already help.

https://github.com/bluekitchen/btstack/blob/develop/example/hid_host_demo.c

[11 more replies]